dresden-elektronik/deconz-rest-plugin #82: What does the colors mean in the deconz ?

donnib, 2017-08-03:
[image]

Kontor Spot X are 5 ikea GU 10 spots sitting within 3 meters. Why are different colors for them ?

Kontor is a IKEA Remote, why does it not have a link to the gateway ? It work because i can see events in websockets. Kontor temp is a hue sensor, the same question goes here. Is it because they are not routers ?

The manual says :

> The links between the nodes visualize the single hop neighborhood. The color of a link represents the Link Quality Indicator (LQI) value between two nodes. The color changes from green (good signal quality) to yellowish/red (weak signal quality).

Is this the explanation ? That it's the linq quality ?

Also perhaps we could make an discord room for question so we don't have to raise issue ? Would that be a good idea ?

manup, 2017-08-04:
Colors represent:
* *blue* coordinator
* *yellow* routers (often lights)
* *grey* end devices (often battery powered)

Your end devices are likely connected to one of the spots, due problems with IKEA too large responses deCONZ won't show the connections between IKEA nodes for now, this will be re-enabled in future release.

ebaauw, 2017-08-04:
What do the colours of the little left circle (status light, almost) on the node mean?  It seems to blink blue on traffic, solid red on error (?), but sometimes it blinks yellow and sometimes it's solid yellow (when the light seems unresponsive?).

manup, 2017-08-04:
Yellow state means a command is end but not yet finished (or ACKed).

donnib, 2017-08-07:
@manup The colors you mention above are the device colors but what about the lines, are those link quality ? Does IKEA mesh with Hue lights or other zigbee lights ?

manup, 2017-08-07:
Yes the lines represent roughly the Link Quality Indicator (LQI) between nodes.
It seems that IKEA and OSRAM behave strange in regards to mesh network. It's not that they don't have the feature but it's working different than Philips hue. Currently I think they use some kind of tree routing, but that needs more investigation.

donnib, 2017-08-07:
@manup thank you for the explanation, do they suppose to do mesh now when using deconz ?

ebaauw, 2017-10-08:
I think we can close this issue.

manup, 2018-09-17:
Closing the oldest issues for know to tidy up the tracker and duplicates in newer issues.

closed 2018-09-17 by manup

IIIdefconIII, 2019-12-11:
What about orange? non-connected?

[image]

Smanar, 2019-12-11:
Lol, for me it's still yellow but darker ^^.
But yes, they are not connected, more visible than the "orange" part is the name (grey istead of dark), all the none is grayed.

Smanar, 2019-12-11:
Lol, for me it's still yellow, but after looking more closely, yes it's more darker.

But yes, they are not connected, more visible than the "orange" part is the name (grey istead of black), all the none is grayed.

apper13, 2021-05-28:
Hi, I found the meaning of most of the colors, but what is missing is the color "green" for the status indicator. Normally it's grey and @manup wrote that yellow means "Yellow state means a command is sent but not yet finished". I see a lot of blue after the communication, so this seems to mean "communication done". But what means green? I often see a green circle.

Smanar, 2021-05-28:
Green is polling.

HemiBob, 2022-02-17:
> What about orange? non-connected?
For colourblind people, deConZ is a nightmare!
In the image shown, I can see one blue, one grey and ALL the rest are orange, am I missing a yellow on?

Smanar, 2022-02-17:
>In the image shown, I can see one blue, one grey and ALL the rest are orange, am I missing a yellow one?

Lool, I m seing same, and I m not colourblind
To found not connected, I look the name, disconnected device have grayed name, 2 first are connected, 2 next no.