dresden-elektronik/deconz-rest-plugin #84: state.bri_inc not yet implemented?

ebaauw, 2017-08-04:
When I try and PUT /api/username/lights/1/state with a body of `{"bri_inc":254}` (or any other value), deCONZ returns nothing and the light's brightness doesn't change.  PUTting `{"ct_inc":512}` (or any other value) works as expected.  
Same for group requests.

manup, 2017-08-04:
Correct, `ct_inc` was implemented to support IKEA tradfri remote color temperature buttons. The `bri_inc` is on the todo list.

... need better error messages for yet unsupported features.

wvuyk, 2017-08-04:
same for hue_inc and sat_inc? Both were used in my plugin as well

manup, 2017-08-04:
Yes currently only `ct_inc` is implemented, the others will follow but there is no ETA yet.

ebaauw, 2017-08-14:
I started sniffing the commands the Hue bridge sends on PUTting these attributes, see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Philips-Hue-sniffing.