dresden-elektronik/deconz-rest-plugin #86: colormode is missing from group doc

donnib, 2017-08-04:
Hi,
I can see that the REST API returns a `colormode` but i can't set it or there is no doc about this.

ebaauw, 2017-08-04:
`colormode` is read-only.  It reflects how the light or group colour was last set: using `ct`, using `hue` and `sat` (`hs`) or using `xy`.  Indeed it's missing from the documentation for /groups, but it's documented for /lights.

donnib, 2017-08-04:
Yes i know but it's not shown for `groups` and also does it make sense ? Since i can use `ct` only. I don't know how this works when you have lights that support different `colormode` hmmm

manup, 2017-08-04:
In my opinion it doesn't make sense, since all lights in the group need to be of same type and all set to equally color (and color mode), which is rarely the case.

manup, 2017-08-18:
Closing this for now.
The lights state `colormode` is documented here: http://dresden-elektronik.github.io/deconz-rest-doc/lights/#getstate

As mentioned for groups colormode won't be implemented.

closed 2017-08-18 by manup