dresden-elektronik/deconz-rest-plugin #87: Incorrect behaviour when turning group off

donnib, 2017-08-04:
State of group before doing anything :

```
{
    "action": {
        "bri": 255,
        "colormode": "ct",
        "ct": 500,
        "effect": "none",
        "hue": 0,
        "on": false,
        "sat": 127,
        "xy": [
            0,
            0
        ]
    },
    "devicemembership": [],
    "etag": "d3150d5b7a80666f016a14f49b17c0d3",
    "hidden": false,
    "id": "5",
    "lights": [
        "4",
        "6",
        "5",
        "2",
        "3"
    ],
    "lightsequence": [],
    "multideviceids": [],
    "name": "Office Spots",
    "scenes": [],
    "state": {
        "any_on": false
    },
    "type": "LightGroup"
}
```
Issue command `{"on": true,"bri":255, "ct":500}`

Events:
```
{"e":"changed","id":"4","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"ct":500},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"ct":500},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"ct":500},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"ct":500},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"ct":500},"t":"event"}
{"e":"changed","id":"3","r":"groups","state":{"any_on":true},"t":"event"}
{"e":"changed","id":"5","r":"groups","state":{"any_on":true},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"ct":454},"t":"event"}
```

State of group :

```
{
    "action": {
        "bri": 255,
        "colormode": "ct",
        "ct": 500,
        "effect": "none",
        "hue": 0,
        "on": true,
        "sat": 127,
        "xy": [
            0,
            0
        ]
    },
    "devicemembership": [],
    "etag": "7f88f978123ba376d4d81af775ccf48d",
    "hidden": false,
    "id": "5",
    "lights": [
        "4",
        "6",
        "5",
        "2",
        "3"
    ],
    "lightsequence": [],
    "multideviceids": [],
    "name": "Office Spots",
    "scenes": [],
    "state": {
        "any_on": true
    },
    "type": "LightGroup"
}
```
Issue command : `{"on": false,"bri":255, "ct":500}`

Events:
```
{"e":"changed","id":"4","r":"lights","state":{"on":false},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"on":false},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"ct":500},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"on":false},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"on":false},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"on":false},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"bri":255},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"ct":500},"t":"event"}
{"e":"changed","id":"3","r":"groups","state":{"any_on":false},"t":"event"}
{"e":"changed","id":"5","r":"groups","state":{"any_on":false},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"5","r":"groups","state":{"any_on":true},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"3","r":"groups","state":{"any_on":true},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"bri":254},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"ct":454},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"ct":454},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"ct":454},"t":"event"}
```
All lights should be off now but they are not. Here is the state of group now :

```
{
    "action": {
        "bri": 255,
        "colormode": "ct",
        "ct": 500,
        "effect": "none",
        "hue": 0,
        "on": false,
        "sat": 127,
        "xy": [
            0,
            0
        ]
    },
    "devicemembership": [],
    "etag": "c5d0f82b0abc884971290945d197271a",
    "hidden": false,
    "id": "5",
    "lights": [
        "4",
        "6",
        "5",
        "2",
        "3"
    ],
    "lightsequence": [],
    "multideviceids": [],
    "name": "Office Spots",
    "scenes": [],
    "state": {
        "any_on": true
    },
    "type": "LightGroup"
}
```
Same goes for one light 2 (just an example the others have same state):
```

{
    "etag": "d70cd72b5ae2c1c53ee7660ce535d401",
    "hascolor": true,
    "manufacturer": "IKEA",
    "modelid": "TRADFRI bulb GU10 WS 400lm",
    "name": "Kontor Spot 2",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "ct",
        "ct": 454,
        "on": true,
        "reachable": true
    },
    "swversion": "1.1.1.1-5.7.2.0",
    "type": "Color temperature light",
    "uniqueid": "00:0b:57:ff:fe:24:d5:98-01"
}
```

This may be related to : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/76

ebaauw, 2017-08-04:
You shouldn't specify any other attributes (except `transitiontime`) when setting `"on": false`.  Normally, the gateway returns an error 201 `parameter, /lights/1/bri, is not modifiable. Device is set to off.` when you try and change an attribute while the light is off.  It looks like the combined request first sends a command to turn the lights off, and then a command to change the brightness, causing the light to ignore (or counter?) the first command.

Note the difference between notifications caused by sending commands to the lights vs notifications as result from attribute reporting by the light.  All lights support `bri` values from 1 of 254 and the Ikea tunable whites support `ct` values from 153 to 454.  When sending a command with an out-of-range value, you first get an event for the value sent, then one for the actual value of the light.

donnib, 2017-08-04:
Well i don't see in the documentation an explnation from this besides this the api does not give me an error for doing this. I see in the doc that bri is 1-255 and not 254 and ct 150-500 but that is different for some lights ? How does one know the supported values ? 

No matter what the behavior i believe it's not handled correctly.  Btw lights support this somehow. If you set a light like this it will not get into the problem above but it will set the bri to 0 on off no matter what you set it to then it will set it to 1 which somehow also weird.

manup, 2017-08-04:
In ZigBee Light Link (ZLL) brightness values 0 and 255 are special, 255 will be clamped to 254, value 0 doesn't exist but is used when command `Move to Level (with on/off)` is send. Lights will then always report `bri` 1, which is correct within the ZLL standard.

If you wan't to turn of lights or groups use either `{"on":false}` or `{ "on": false, "bri": 0, "transitiontime": xxx}`. *Only* the first one will preserve the current brightness value the second one allows to slowly fade down but will set the brightness to 1.