dresden-elektronik/deconz-rest-plugin #88: Weird behavior with IKEA lights

donnib, 2017-08-04:
Hi,
So i have these lights and groups as you can see here :

```
{
    "config": {
        "UTC": "2017-08-04T21:07:53",
        "apiversion": "1.0.1",
        "backup": {
            "errorcode": 0,
            "status": "idle"
        },
        "bridgeid": "00212EFFFF010284",
        "datastoreversion": "60",
        "dhcp": true,
        "factorynew": false,
        "gateway": "127.0.0.1",
        "internetservices": {
            "remoteaccess": "disconnected"
        },
        "ipaddress": "127.0.0.1",
        "linkbutton": false,
        "localtime": "2017-08-04T23:07:53",
        "mac": "38:60:77:7c:53:18",
        "modelid": "deCONZ",
        "name": "RaspBee-GW",
        "netmask": "255.0.0.0",
        "networkopenduration": 60,
        "panid": 11100,
        "portalconnection": "disconnected",
        "portalservices": false,
        "portalstate": {
            "communication": "disconnected",
            "incoming": false,
            "outgoing": false,
            "signedon": false
        },
        "proxyaddress": "none",
        "proxyport": 0,
        "replacesbridgeid": null,
        "starterkitid": "",
        "swupdate": {
            "checkforupdate": false,
            "devicetypes": {
                "bridge": false,
                "lights": [],
                "sensors": []
            },
            "notify": false,
            "text": "",
            "updatestate": 0,
            "url": ""
        },
        "swversion": "2.04.60",
        "timeformat": "24h",
        "timezone": "Europe/Copenhagen",
        "uuid": "c80af094-2a9a-4ad0-8eed-2fb9b626b8f7",
        "websocketport": 443,
        "whitelist": {
            "06B513B2AC": {
                "create date": "2017-07-27T13:02:14",
                "last use date": "2017-07-27T13:52:28",
                "name": "deCONZ WebApp"
            },
            "4A09426CBD": {
                "create date": "2017-07-27T20:59:00",
                "last use date": "2017-08-04T21:07:53",
                "name": "my application"
            },
            "52EF08F0AD": {
[... 437 lines cut ...]
                "lastupdated": "2017-08-04T21:06:43"
            },
            "swversion": "1.1.1.1-5.7.2.0",
            "type": "ZHASwitch",
            "uniqueid": "00:0b:57:ff:fe:2f:e6:6f-01-1000"
        }
    }
}
```
I have made a bash script that does following for test purposes :
```
#!/bin/bash

curl -H 'Content-Type: application/json' -X PUT -d '{ "on": true}' http://10.10.XX.XX:80/api/XX/groups/5/action

while :
do
    curl -H 'Content-Type: application/json' -X PUT -d '{ "on": false}' http://10.10.XX.XX:80/api/XX/groups/5/action
    sleep 2
    curl -H 'Content-Type: application/json' -X PUT -d '{ "on": true}' http://10.10.XX.XX:80/api/XX/groups/5/action
done
```
And works perfectly fine. Now if i click on the main button on the remote with id 1 it messes the lights up so some are off the on, everything goes wrong. As you can see i don't use the remote in the groups. If i click the remote while the lights are on and i don't run the bash it doesn't do anything. It's hard to explain but i have a feeling it's interacting with the lights. I have not paired this remote with any of those lights. If it were paired i would have expected it to be usable also while the lights are on but no only when the bash is running and the group is turning off/on every 2 sec and i use the remote.

Any explanation to this, it's driving me crazy.

manup, 2017-08-04:
Simple guess, you're flooding the network and broadcast queues on the lamps fill up. In a ZigBee sniffer your setup will look like crazy, every groupcast is repeated 3 times from every light. And each groupcast is stored in all lights for a few seconds to prevent repetition, but memory is limited.

So don't do group- or broadcasts in that frequency in a loop, if you raise from 2s to 10s and more it will improve I guess.

Even if the remotes are not paired to the lights, the lights will repeat the group casts from the remote through the network. In *normal* usage this is no problem but if a loop bombards the network it is ;) Usually lights are able store minimum of 8 groupcasts in parallel.

donnib, 2017-08-04:
Ok i understand, the problem started when i was testing the remote thru HASS to turn on/off the group and i was doing this like every second by hand just to stress test but yes that is not a usecase. I guess this problem probably also is the case by using the ikea hub.

runningman84, 2017-08-05:
Do you have a working HASS component for this dongle?

donnib, 2017-08-05:
@runningman84 yes i have developed one or rather still works on it, it's very rough and ugly.

closed 2017-08-08 by donnib