dresden-elektronik/deconz-rest-plugin #89: Support for Xiaomi button (lumi.sensor_switch)

snozzlebert, 2017-08-05:
I was able to let the Xiaomi button join the network. It appears in deCONZ (2.04.61), but not in the REST api.
In deCONZ I see the following:

Model identifier: lumi.sensor_switch
IEEE: 0x00158d000152d709
Basic Cluster 0000
On/Off Cluster 0006

Running deCONZ with dbg-aps/dbg-info it shows the following when pressing and releasing the button (same messages):

Button down:
`11:53:53:750 APS-DATA.indication srcAddr: 0x00158d000152d709, dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -47`

Button release:
`11:53:53:959 APS-DATA.indication srcAddr: 0x00158d000152d709, dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -46`

There is also no difference in messages when the button is hold down for a longer time, except  a longer time between the first and second message.
Is it possible to integrate this button in the REST api?
If you need any additional information to integrate the button in the REST api, please let me know.

ebaauw, 2017-08-05:
Interesting, it behaves exactly like the Aqara door/window sensor.  It's more a stateful contact sensor than a stateless switch, like the Hue dimmer or the Trådfri remote.  Would you want this supported as a ZHASwitch or a ZHAOpenClose sensor?

As ZHASwitch, it would probably only support two values for `state.buttonevent`: `1000` (press) and `1002` (release).  As ZHAOpenClose, it would show `state.open` as `false` while the button is pressed and `true` while it's not.

snozzlebert, 2017-08-05:
Behaviour of the button indeed looks the same as the Xiaomi door/window sensor.
ZHASwitch seems more appropriate to me, because it is a button.
I will probably use only the websocket events to trigger events in my home automation setup.
The device itn't really useful compared to the Hue dimmer switches and the Hue Tap. I bought them for fun.

manup, 2017-08-05:
>As ZHASwitch, it would probably only support two values for state.buttonevent: 1000 (press) and 1002 (release).

The ZHASwitch is also my favourite, I'll put something together in the next release based on the ZHAOpenClose code.

ebaauw, 2017-08-07:
> I bought them for fun.

That's the most compelling argument!  And less than EUR 7 on GearBest.  I couldn't resist...  It should arrive in 20 days or so.

manup, 2017-08-07:
Beta version 2.04.62 now includes experimental support, need to be tested. Based on some Xiaomi Youtube video I figure that the switch also could support long press event in future.

https://github.com/dresden-elektronik/deconz-rest-plugin/commit/b9cf4bc890c5fdfd00cd53b2f3a2a7a3c32f5a67

snozzlebert, 2017-08-08:
Thanks, it appears in the REST api:
```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "ffca99bfb5034fbbc971ae3f660ce253",
    "mode": 1,
    "modelid": "lumi.sensor_switch",
    "name": "lumi.sensor_switch 7",
    "state": {
        "buttonevent": 1000,
        "lastupdated": "2017-08-07T21:40:02"
    },
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:01:##:##:##-01-0006"
}
```
I also get events when connected to the websocket.
However, when I push the button down I get:
`{"e":"changed","id":"7","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2017-08-08T08:35:13"},"t":"event"}`
and after releasing I get:
`{"e":"changed","id":"7","r":"sensors","state":{"buttonevent":1000,"lastupdated":"2017-08-08T08:35:15"},"t":"event"}`
I would expect 1000 (initial press) before 1002 (released). And 1002 as last state of the sensor.

ebaauw, 2017-08-08:
> I would expect 1000 (initial press) before 1002 (released). And 1002 as last state of the sensor.

Indeed, so would I.

> it appears in the REST api

No `manufacturername`?

snozzlebert, 2017-08-08:
It took a while to get the manufacturername, I had the same with my door sensor.

```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "34303a632ba1ebb6e8242fbccfdf644f",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.sensor_switch",
    "name": "lumi.sensor_switch 7",
    "state": {
        "buttonevent": 1000,
        "lastupdated": "2017-08-08T11:15:36"
    },
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:01:##:##:##-01-0006"
}
```

manup, 2017-08-09:
The events are reversed now: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/f801bd65d7f01d1aa3da6501aa595f1b20d43ee0

ebaauw, 2017-08-09:
homebridge-hue should now support this switch as well, but I haven't been able to test it yet...

manup, 2017-08-10:
I've also ordered the Xiaomi magic cube, looks like a fun device.

ebaauw, 2017-08-10:
It does!

Might be challenging to define meaningful buttonevents for it, though.  Maybe use different "buttons" for the different gestures: tap, shake, move, rotate, turn; and distinguish between 90° and 180° turn by short release vs long release button events?

HomeKit only knows _Single Press_, _Double Press_, and _Long Press_.  I don't support _Double Press_ from homebridge-hue, as I cannot detect that reliably through polling.  With deCONZ event notifications I probably could, however.  _Single Press_ is issued on the short release buttonevent; _Long Press_ on the first hold or long release buttonevent.

snozzlebert, 2017-08-11:
Events are in the right order now. Should I close the issue or leave it open for magic cube?

manup, 2017-08-12:
I'll close it for now, it's best to open a new issue for the cube to keep headlines organized :)

closed 2017-08-12 by manup

rtenklooster, 2017-08-30:
The button works fine.
It's only reporting on / off.
double click, short click and long click events are not supported right now?
Any plans on supporting those events? I can see long / short clicks in the mi home app, so the switch supports those events.