dresden-elektronik/deconz-rest-plugin #90: Sensor config.on not implemented?

ebaauw, 2017-08-05:
When I set a sensor's `config.on` attribute to false, it's state continues to be updated.  On the Hue bridge, setting `config.on` to false blocks updates to the sensor state.  This is used by the Hue app to disable the Hue motion sensor.

manup, 2017-08-05:
Added to the todo list. Currently this is only possible by disabling the rules. The `config.on` makes sense and simplifies the task.

ebaauw, 2018-11-23:
Still on the list, I hope.

manup, 2018-11-23:
I thought this is already working :) need to check, so basically rules shouldn't be processed when the sensor is off. Should web socket events also be stopped?

ebaauw, 2018-11-23:
If memory serves, the Hue bridge sets the `state` attributes to `0` or `null` or `"none"` or something and no longer accepts PUTs to CLIP sensors `state`.

manup, 2018-11-23:
Ok guess this should be compared in more detail, can be handy to disable some automations without messing too much with the existing rules.

ebaauw, 2020-08-03:
Not solved.

ebaauw, 2024-09-04:
I suppose we just need to document that setting `config/on` only disables firing of rules.

closed 2024-09-04 by ebaauw