dresden-elektronik/deconz-rest-plugin #91: Docker/Hass.io addon?

lajo-osd, 2017-08-07:
Hi!

I have an install on a Ubuntu 17.04 install on a virtual machine with the latest hass.io install which uses docker and also has support for add-ons, also via docker.

Are there any plans for creating a docker package for deCONZ? And if so any plans for creating and add-on for hass.io?

Some info can be found here: https://home-assistant.io/hassio/installing_third_party_addons/

I don't yet have the knowledge for this myself so if anyone can contribute to this it would be great!

Thanks!

donnib, 2017-08-07:
Hej, @lajo-osd i am working on a HASS component that i have working at the moment using in my house with IKEA Lights GU10, E27, E14, Remote Control 5 button, Motion Sensor and Philips Hue motion. I am still developing on it and it needs a major overhaul code wise since i just hacked it together but it seems to work for the moment.

lajo-osd, 2017-08-07:
Ok, that sounds really good!
Can I try it or when do you expect to have it more ready?

I just got my Conbee and Trådfri lights, remote and  sensor so I am eager to get them up and running :)

Will the component work for Conbee and Ubuntu 17.04?

donnib, 2017-08-07:
I am using it with RaspBee but i see no reason why not since it's still runs deConz. I have no experience with Hass.io how you install components, it seems components are docker containers and this one is not so i don't know. My component is work in progress, i find new bugs everyday which i fix but mostly it works :)

lajo-osd, 2017-08-07:
Yes, they are docker containers and then some additional info is needed to add it as an add-on.

There are several instructions on how to create a docker container online but since I have not even installed deCONZ yet I am not ready to look at creating a docker container for it :)

Would you be able to outline the install steps for a headless setup? I have seen a few examples online and none are the same it seems.

donnib, 2017-08-07:
@lajo-osd i am using a Rasberry PI without a monitor and which uses the official image from Dresden Elektronik and that starts deconz automatically and it works ok. What are you running your Hass.io on ? in a vm ?

lajo-osd, 2017-08-07:
@donnib Ok, so you did not have to modify anything for X11 and systemd? 

Yeah, I am running Ubuntu 17.04 in a VM with the latest Hass.io install with a few add-one.
It seems to run great and update fine both for Hass, the supervisor and the add-ons.
But I am just getting started with Hass now and plan to migrate my scenes from my Vera to Hass and add IKEA and Xiaomi via deCONZ if all goes well.

donnib, 2017-08-07:
Nope did not do any changes to the rPi image, i just plugged in the module (Raspbee) and i was running. 

I believe it should be possible quite easy to use my component in your install as long as you have access to /config where the custom_components folder must be placed. I think you should start getting deConz to work with Conbee then we can talk. Also we should probably continue this talk in HASS community https://community.home-assistant.io/t/request-for-deconz-hass-io-add-on/23312 ;)

manup, 2017-08-07:
One side note, a new beta release for Ubuntu will be released this week. It contains the recent improvements for IKEA and Xiaomi sensors. The current Ubuntu version 2.04.35 doesn't support these yet.

donnib, 2017-08-07:
@manup but you can build them yourself no ?

manup, 2017-08-07:
Yes but only the REST plugin. The application core and device firmware need to be updated via the package.

donnib, 2017-08-07:
Ahh i see thx.

lajo-osd, 2017-08-07:
Yes, let's discuss in my thread in the Hass community. I'll get it up and running first on Ubuntu and then once the beta is released I'll upgrade to that. I might to an install on a separate VM to see if I can figure out to create a docker container for this.

closed 2017-08-07 by lajo-osd

manup, 2017-08-07:
Ubuntu beta version 2.04.62 is noch available.
https://www.dresden-elektronik.de/deconz/ubuntu/beta/

lajo-osd, 2017-08-08:
Thank you very much for that!

lajo-osd, 2017-08-08:
@manup I cant install the beta using the install instructions here. It fails on the "qmake && make -j3" command. Perhaps because the is no V2_04_62 to be checked ut in this command: git checkout -b mybranch V2_04_52.

I followed the instructions from here: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/README.md

**Can you please outline the install instructions for all deconz components for Ubuntu and also to get headless working?**

Thanks!

rtenklooster, 2017-08-30:
@donnib are you willing to share your code/ component with me?
I want to connect HA to the REST Api and since you already created something I would be helped a lot!

donnib, 2017-09-01:
Yes I'd like to share it when I fell comfortable sharing it since I am not happy with it. I am in vacation at the moment so when I am back and I have time I'd like to put it out

sveip, 2017-12-29:
Any progress on this @donnib?