dresden-elektronik/deconz-rest-plugin #92: Groups events are not received

donnib, 2017-08-07:
Hi,
I am pretty sure i have seen an event for group `state` before but it doesn't work now.

I do following : `PUT 10.10.0.XX:80/api/XX/groups/3/action`
 `{"on": true}`

I get response :
```
[
    {
        "success": {
            "/groups/3/action/on": true
        }
    }
]
```
and events :
```
{"e":"changed","id":"2","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"3","r":"groups","state":{"any_on":true},"t":"event"}
{"e":"changed","id":"5","r":"groups","state":{"any_on":true},"t":"event"}
```
but as you see no group events on the state. 

Here is the group :

```
{
    "action": {
        "bri": 127,
        "colormode": "hs",
        "ct": 0,
        "effect": "none",
        "hue": 0,
        "on": true,
        "sat": 127,
        "xy": [
            0,
            0
        ]
    },
    "devicemembership": [],
    "etag": "6dcc20881eb7a2a47f04d4beeb72f05f",
    "hidden": false,
    "id": "3",
    "lights": [
        "2",
        "4"
    ],
    "lightsequence": [],
    "multideviceids": [],
    "name": "Kontor Every Second",
    "scenes": [],
    "state": {
        "any_on": true
    },
    "type": "LightGroup"
}
```

ebaauw, 2017-08-07:
But you do see group state events: lines 3 and 4 (apparently (at least) one of the lights is also in /groups/5).  No that the group state only contains a single attribute: `any_on`.

There's no events for the group `action` object.  I'm not even sure if they would make sense, as `action` holds "The last action which was send to the group" (see http://dresden-elektronik.github.io/deconz-rest-doc/groups/#getattr), not the actual state of the lights in the group.  This is different from the Hue bridge, where the group `action` holds "The light state of one of the lamps in the group" (see https://developers.meethue.com/documentation/groups-api, section 2.3).

donnib, 2017-08-07:
So i must have been dreaming. So i am suppose to make a rest call on the group to check state in action or that is not useable? What exactly should I use for group state?

ebaauw, 2017-08-07:
Not sure I understand your question.  How would you want to use group state?

There is no such thing as a group state in ZigBee.  Each light has a state, and a light can belong to a number of groups.  In essence, a ZigBee group is only a multicast address, to reach all its member lights with a single ZigBee message.  The group action is used to to set the state of its member lights.  At best a group state would be an aggregate of the states of its member lights.  Currently this is only implemented for the light `state.on`, resulting in a group `state.any_on` (and, on the Hue bridge,`state.all_on`).  I could imagine a `state.avg_bri`, `state.min_bri`, `state.max_bri` for a group, based on the `state.bri` of its member lights, but I have no idea how to use a `state.avg_xy`.  This is probably too expensive to compute, anyways, as both DE and Philips took a different approach for reporting group action.

donnib, 2017-08-07:
Well i show the group in HASS as one entity that is either on/off and all the lights separate. I think what i am after is `state.all_on` since that's essentially stating all lights in the group is on everything else is half group if you can say so.