dresden-elektronik/deconz-rest-plugin #93: Problems 10 ikea GU 10 spots

donnib, 2017-08-07:
Hi,
I have just added my kitchen GU10 spots Trådfri to deConz and added them to a group and they behave weird. The pairing went kind of ok, it took some time before deconz showed line between gw and the spots and i still have spots which do not have lines but do blink once in a while with blue. Also i have a spot which is paired but i can't control it from the webui. I have tried to turn off/on 6 times then open the network again in hope it will work but nope. I have atm 3 spots on while deconz thinks they are off, toggling them does nothing but it does issue events as it was done. What can i do ?

donnib, 2017-08-07:
After few resets of the lights (6 times on/off) now everything is screwed up. I also did few reboot from the UI of rPi. All devices show yellow small indicator and no lines between gateway. I don't know what happend. How can i get back live again ?

donnib, 2017-08-08:
This morning i looked at deConz again, other lamps started to work but i still could not get the 10 x GU10 spots to work. I have previously paired 5 spots the one on the left but that was with another firmware. I got erratic behavior like :

- Turn off all lights but only few turn off
- Turn on all lights but only few turn on
- Not all lights had lines between them
- I had one light which i never succeeded to turn on/off, it just stayed on but it was shown in deConz
- I had lights disappearing then showing up if i took the power and turned them on again.

All above after i paired them and also this morning when i tried again to have a look with fresh eyes. 

Did i do something not allowed when pairing them ? I have no idea how to get them in deConz in a working state. The only thing i did comparing to last time i paired is that after pairing i used identify "Blink" to identify spots then name them using REST API.

After turning them off they are shown as red in deconz atm.

[image]

ebaauw, 2017-08-08:
Might be related to #33?

donnib, 2017-08-08:
Yes definitely, i did read your post about that. It's driving me crazy, have no idea how to solve the problem and i have very few nodes in total i would say. I had in mind adding much more. It's disappointing :( I hope i can find a fix.

donnib, 2017-08-08:
@manup i would appreciate some ideas how i can overcome this, any ideas ?

donnib, 2017-08-08:
Really no idea what to do ? Should i pair all lamps again ? Should i delete them first from deConz ? Is it ok to pair all of them same time ? (removing all 10 from the spots in the ceiling is a hassle for me). Can i get more info by debugging ? Right now i have 10 spots that are pretty much unusable in deconz :(

ebaauw, 2017-08-08:
> After few resets of the lights (6 times on/off)

After that, you would again need to pair the light with deCONZ.  Did you restart deCONZ after resetting the lights?  They shouldn't show in the GUI anymore after that.

To ditch lights from deCONZ, best:
1. Remove the resource from `/lights` in the REST API;
2. Delete the node from deCONZ (clicking it and pressing Del);
3. Reset the light (if needed).

If the lights are still paired, but showing red (as in the picture), you can try:
- Clicking on the node and pressing `0`; or
- Powering the light off and back on; or
- Restarting deCONZ.  In my experience, deCONZ reconfigures the lights after restart.

donnib, 2017-08-08:
Ok i will try your suggestions. I did actually restart deConz many times yesterday. And i reset the lights many times (6 times on/off). 

Should deConz be able to pair all of them once or do i really have to disconnect all of them then connect each of them at a time and open network each time ?

ebaauw, 2017-08-08:
> I did actually restart deConz many times yesterday.

If the lights re-appear after restarting deCONZ, they're still paired to the gateway (i.e. still connected to the ZigBee network created by the gateway).  They should come on yellow in the deCONZ GUI after a start.  You might want to read the basic cluster attributes of each light, so the manufacturer, model, etc are known.  Then open the network (in the web app) and the lights should appear in the REST API.

donnib, 2017-08-08:
Hmm that is weird because after the lights are reset they all flash once i am pretty sure they are reset but even in that case they are obviously still paired because they did show up. I will do your 3 points to ditch the light from deConz and then hope i can pair the lights without doing it one by one but all at the same time.

Does 
>Remove the resource from /lights in the REST API

and
 
> Delete the node from deCONZ (clicking it and pressing Del)

not do the same ?

donnib, 2017-08-08:
So i removed the lights from REST API, deleted them from deconz, reset the lights, opened the network again and still very weirdly some have lines, some don't. Out of 10, 1 is not found at all, 2 are shown but without lines and i can't turn them off.

manup, 2017-08-08:
Don't give too much attention to the lines. If lights are already in the network reset them won't do any good, I figure the main challenge here is that your network consists of IKEA only lights.

I've figured out something in the sniffer, I wan't to try to adjust the RaspBee firmware, not sure yet if it helps but it will improve discovery of IKEA lights. They send fairly large neighbour table frames and deCONZ firmware ignores them, this is a bug in the RaspBee/ConBee firmware.

donnib, 2017-08-08:
Yes I have only IKEA lights actually all devices are IKEA besides two devices which are Phillips hue motion sensor and a Phillips dim switch. Is there a bigger chance to pair them if I shut off all other lights than these 10 spots which I can only shut off all at the time?

donnib, 2017-08-09:
So i succeded to pair all the spots, after pairing all shown fine in deConz, i added all of them to the group but then on / off on the group only turned off few spots so i did a reboot of the rPi and then thr group on/off works hopefully it stays like that.