dresden-elektronik/deconz-rest-plugin #94: update firmware of conbee is not working with latest x64 build

holli73, 2017-08-07:
hello,

i'm on 2.04.62 with my x64 linux box and the conbee - on the mainscreen (lights - webapp) i have notification that a new firmware is available and once i click on the settings and update the firmware i can see on the UI (vnc session) that the firmware is loaded and deconz is restarted but after some time the update firmware info is coming back/shown again.

[image]

any idea?
thanks
holli

manup, 2017-08-08:
From the `/tty/USB4` I reckon you have multiple USB (FTDI?) devices attached to the computer? The firmware updated isn't very at at the moment and might reject to update if it's not clear what to update. This will improve over time.

In the meantime you can try to manually updatewhile deCONZ isn't running.

1) check which USB device one is the ConBee

    `$ sudo GCFFlasher_internal -l`

2) update firmware (here -d 0 is used for first USB device)

    `$ sudo GCFFlasher_internal -d 0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26110500.bin.GCF`

holli73, 2017-08-09:
hello,

if i run the command i get:

```
holli@obelix:~/BUILD/deconz$ sudo GCFFlasher_internal -l
rmmod: ERROR: Module ftdi_sio is in use
rmmod: ERROR: Module usbserial is in use by: option usb_wwan ftdi_sio
GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26

       5  FTDI devices found

 device | vendor | product | serial  | description
--------|--------|---------|---------|----------------------
failed to open FTDI device (0) status = 3

failed to open FTDI device (1) status = 3

failed to open FTDI device (2) status = 3

failed to open FTDI device (3) status = 3

failed to open FTDI device (4) status = 3

```
yes i do have some more ftdi devices attached and ttyUSB4 is the conbee - but why can't the flasher open the device for update?

any idea?
thanks
holli

manup, 2017-08-09:
Status 3 is from FTDI library and means FT_DEVICE_NOT_OPENED.
The problem seems to be that the FTDI kernel modules can't be unloaded because they are used from the other devices. The FTDI library is troublesome because it's not working when running in parallel with other FTDI libs.

Therefore to keep it simple I would recommend to update the ConBee on your Rapberry Pi.

ebaauw, 2017-08-09:
I was able to flash the BitCatcher firmware to the ConBee on my Ubuntu virtual machine.  I don't have any other USB nor serial devices in that VM, however.

ooii, 2018-03-14:
Hi,

I'm running deCONZ under Ubuntu and would like to update my ConBee firmware. Should I use the latest of `Ubuntu Linux` availble [here](https://www.dresden-elektronik.de/funktechnik/service/download/driver/?L=1#c3944)? Or can I use a `RaspBee Firmware`?

Thanks.

manup, 2018-03-14:
Please use latest version from here:

https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.12-qt5.deb

Installation: 

```bash
$ sudo dpkg -i deconz-2.05.12-qt5.deb
$ sudo apt install -f
```
This includes also latest firmware for ConBee.

You can start deCONZ from Ubuntu applications menu.

ooii, 2018-03-14:
Thanks for your answer. But `/usr/share/deCONZ/firmware` contains only `deCONZ_Rpi_*` files. Is that safe to install one of them while I'm running deCONZ on ubuntu?

manup, 2018-03-14:
>Thanks for your answer. But /usr/share/deCONZ/firmware contains only deCONZ_Rpi_* files. Is that safe to install one of them while I'm running deCONZ on ubuntu?

Yes RaspBee and ConBee use the same firmware.

Manual firmware update:

   
```bash
    # deCONZ must be closed first
    $ sudo systemctl stop deconz

    # update firmware
    $ sudo GCFFlasher_internal -d 0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261e0500.bin.GCF
```

> And I don't have any Ubuntu application menu since I'm running a headless server.

We have not adapted the systemd script for headless Ubuntu yet (coming soon).
The file `/etc/systemd/system/deconz.service` was made for headless Raspberry Pi, you may edit it to use you Linux username instead of *pi*.

Then start deCONZ headless service by
`$ sudo systemctl start deconz`

ooii, 2018-03-14:
Thank you very much. 
BTW, I have already adapted the systemd script to my needs.

closed 2019-03-30 by stale[bot]