dresden-elektronik/deconz-rest-plugin #100: deCONZ implements `state.alert` differently from the Hue bridge.

ebaauw, 2017-08-08:
deCONZ implements a light `state.alert` (or a group `action.alert`) differently from the Hue bridge.

Philips Hue lights don't react to`{"alert": "select"}` from deCONZ.  When issuing the _Identify_ command from the deCONZ GUI with an _Identify Time_ of 1, the Philips lights don't react either.  
With an _Identify Time_ of 2 or larger, the light reacts, and indeed they react to `{"alert": "lselect"}`.  I notice, however, that the light blinks in power on default state.  If memory serves, with the Hue bridge they would blink using the current state.

The deCONZ `{"alert": "select"}` works with IKEA, OSRAM, and ubisys lights, and with the innr RB 162.  The innr UC 110 don't react, but, on the good part, they don't lock up as they do after a `{"alert": "select"}` from the Hue bridge.

According the Hue documentation: 
> The alert effect, is a temporary change to the bulb’s state, and has one of the following values:
“none” – The light is not performing an alert effect.
“select” – The light is performing one breathe cycle.
“lselect” – The light is performing breathe cycles for 15 seconds or until an "alert": "none" command is received.

> Note that this contains the last alert sent to the light and not its current state. i.e. After the breathe cycle has finished the bridge does not reset the alert to "none".

> breathe cycle | The light or lights do one smooth transition from the current state to a higher brightness in the current color to a lower brightness in the current color and back to the original state.

Indeed, the Hue alert looks like a `Breathe` _Trigger Effect_, but then stopped after a second (for `select`) or 15 seconds (for `lselect`).  I'd love to sniff the commands sent by the Hue bridge, but I would need to recover the network key created by the Hue bridge before BitCatcher can decrypt them.  How would I do that?

ebaauw, 2017-08-08:
> I would need to recover the network key created by the Hue bridge before BitCatcher can decrypt them. How would I do that?

I managed to do that for my OSRAM gateway, I think through the following steps:
1. Enter the ZLL link key;
2. Sniff while pairing a new device.  You should than see a decrypted message where the bridge sends the network key to the new device;
3. Enter that key and... voilà, the entire traffic is decrypted.

ebaauw, 2017-08-14:
> I'd love to sniff the commands sent by the Hue bridge

On the Hue bridge, setting `state.alert` to `select` results in two _Trigger effect_ (0x0003/0x40) commands.  The first with parameters _effectIdentifier_ `Breathe` (0x01) and _effectVariant_ `Default` (0x00).  The second with `Finish effect` (0xFE) and, again, `Default` (0x00).  The second command is sent immediately (30ms) after the first.  Setting `state.alert` to `lselect` results in only the first command, which causes the light to breathe for 15s.  Setting `state.alert` to `none` results in only the second command.

manup, 2017-08-14:
Hmm tricky I'm afraid not all lights do support these commands well, should be testet with cluster info panel. And as a fallback for non supporting lights and group casts I would suggest the plain old identify command.

ebaauw, 2017-08-14:
>Hmm tricky I'm afraid not all lights do support these commands well, should be testet with cluster info panel.

Indeed, the innr PL 110 and UC 110 lights hang after receiving these commands and require a power cycle.  I do prefer the Breathe over Identify for Hue lights, as Identify uses the unholy power-on default state, whereas Breathe uses the current state.  This would be yet another attribute to keep in the whitelist :-(

> And as a fallback for non supporting lights and group casts I would suggest the plain old identify command.

For non-supporting lights: yes.  For groups: can we at least change the _Identify Time_ to 2?  The Philips Hue lights simple don't react when _Idenfity Time_ is 1.  The Hue bridge sends _Idenfity Time_ 2 to the Hue motion sensor when `config.alert` is set to `select`.  
Incidentally, the innr lights seem fine when setting `action.alert` to `select`.  I still need to sniff whether that's the same command with different timing (the Hue bridge allegedly throttles group commands to one per second), or a different command.

ebaauw, 2017-08-20:
> can we at least change the Identify Time to 2

PR #128.  Also sets time for `lselect` to 15 (in line with Hue bridge).  Implementing _Breathe_ for Hue lights is still on the TODO list ;-)

ebaauw, 2017-09-10:
Added new values for `state.alert` (and `action.alert`) to issue the corresponding Trigger Effect command "blink", "breathe", "okay", "channelchange", "finish", and "stop".  This is available in 2.04.76.

To emulate the Hue bridge "select", issue a "breathe" followed by a "finish".  For "lselect", simply issue a "breathe".

closed 2017-09-10 by ebaauw

manup, 2017-09-10:
Good work, this can become very handy in some situations to provide a certain feedback to users. For example lights may trigger *okay* after a switch is configured/bound.