dresden-elektronik/deconz-rest-plugin #102: Feature request: support power-on defaults for OSRAM lights

ebaauw, 2017-08-08:
Could we support setting the power-on defaults of an OSRAM light from the deCONZ REST API?  I'd imagine a syntax like PUT /lights/xx/config/poweron with a body of `{"bri":xxx, "ct":xxx}`.

Sniffing around the OSRAM gateway, it looks like it issues a command to the 0xFC0F cluster to store the power-on defaults (packet 691).  Before that, it issues an On command to the 0x0006 cluster (packet 683), and a 0x0A command to 0x0300, with the current colour temperature as data (packet 687).  The same sequence in 311, 315, 317.

ebaauw, 2018-11-23:
Please keep this on the list.  It will be very relevant this fall, with the expected new firmware for the Hue bulbs finally allowing a power-on setting.

manup, 2018-11-23:
Level control cluster attributes suggest that when I recall correctly, but I haven't tested if it works.

ebaauw, 2018-11-23:
There's two different issues (I had them confused as well):
- How the light behaves when turning it on (while it has remained powered).  This is set through the _On Level_ and _OnOff Transition TIme_ attributes of the Level cluster.  The Hue lights don't support these, and behave as if they're set to 255 (previous level) and 4 (0.4 sec transition time).  Most other lights (OSRAM, IKEA, ubisys, innr) support these, so you want to set them to the Hue values.  They're stored in volatile memory, so this settings needs to be applied regularly, or at least after a device announcement.
- How the lights behaves when powered on.  Currently, the Hue lights come on at full intensity and some unholy colour temperature.  OSRAM allows changing this through a manufacturer specific cluster.  I've not seen anything like this for other lights.   Hue have announced they will support setting another power-on default, see https://github.com/ebaauw/homebridge-hue/issues/41#issuecomment-436936220.  I'm assuming it will be through an API enhancement, probably writing some manufacturer specific attribute.

ebaauw, 2018-11-24:
> hopefully Philips will go the standard route here.

I'm confident they will, the issue is more: which standard?  ;-)

manup, 2018-11-26:
Side note: the new hue developer docs describe the `startup` attribute:

https://developers.meethue.com/develop/hue-api/supported-devices/

ebaauw, 2018-11-26:
Interesting.  They don't mention it on the [Lights API](https://developers.meethue.com/develop/hue-api/lights-api/) page.  It is mentioned in the [changelog](https://developers.meethue.com/develop/hue-api/api-documentation-changelog/).

Last weekend, I updated my Hue bridge to the latest firmware (API version 1.28.0), and paired a Hue LCT003 extended color spot (to see if the firmware for the light would already be available - not).  It didn't show a config.startup (my sorting):
```json
{
  "capabilities": {
    "certified": true,
    "control": {
      "colorgamut": [
        [
          0.675,
          0.322
        ],
        [
          0.409,
          0.518
        ],
        [
          0.167,
          0.04
        ]
      ],
      "colorgamuttype": "B",
      "ct": {
        "max": 500,
        "min": 153
      },
      "maxlumen": 250,
      "mindimlevel": 5000
    },
    "streaming": {
      "proxy": true,
      "renderer": true
    }
  },
  "config": {
    "archetype": "spotbulb",
    "direction": "downwards",
    "function": "mixed"
  },
  "manufacturername": "Philips",
  "modelid": "LCT003",
  "name": "Hue color spot 1",
  "productname": "Hue color spot",
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "ct",
    "ct": 366,
    "effect": "none",
    "hue": 14988,
    "mode": "homeautomation",
    "on": true,
    "reachable": true,
    "sat": 141,
    "xy": [
      0.4575,
      0.4101
    ]
  },
  "swupdate": {
    "lastinstall": null,
    "state": "noupdates"
  },
  "swversion": "5.105.0.21536",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:10:3c:9d:0a-0b"
}
```

closed 2019-04-02 by stale[bot]

intruder7777, 2019-07-03:
i would like reopen this request because power-on defaults for osram lights actually already not implemented

ebaauw, 2019-09-08:
The `general.xml` from the above commit adds deCONZ GUI support, see #1833.

Luckily, the bulb seems happy with a standard (non-manufacturer-specific) 0x01 command to the manufacturer-specific 0xfc0f cluster.  Note the mismatch between the manufacturer code in the node descriptor (0xbbaa) vs that in the sniffed command (0x110c).

7wells, 2021-03-13:
Erm, sorry to ask, but it remains unclear to me after reading all this:
How would I set values (light temperature and brightness) for an OSRAM Lightify/Smart+ E27 bulb to be used when electricity went off and comes back on again? I assume I have to do this in deconz (not in phoscon) but am still not clear where exactly in deconz this is set. Of course I do not expect a full guidance, but a hint which tab to visit and which code to use would be great. Thank you! 👍

7wells, 2021-03-13:
In deconz, I see for one of my "OSRAM Extended color lights" in the "Node Info" tab a section "Power Descriptor":
Power Mode: On When Idle
Power Source: Mains
Power Level: 100%

It seems that I can overwrite e.g. the 100% by 90% but cannot check its effect (as I am away). Would this be the way to go? And how about the light temperature? I do not see anything there (despite the light is a "tunable white" bulb). And I have not understood what is meant by "proprietary command on the 0xFC0F cluster to store the current state as power-on state". Once more, thanks for your help! :)

ebaauw, 2021-03-13:
Set the current state however you want: in Phoscons, in deCONZ, through your favourite API client.  Then in the GUI open the _Cluster Info_ panel and select the FC0F cluster on the OSRAM bulb, and press the button to execute the _Store Power-On Defaults_ command.

7wells, 2021-03-24:
@ebaauw 
Very good - thanks for your helpful advice. 😄 First I did not know that I first have to click on the right one of the two radio buttons of the device in order to see something on the _Cluster Info_ panel. 😊 The execution of e.g. the Osram mini garden pole was successful, same as for a night light. There is no (future) possibility to get this done directly in phoscon? Because it is manufacturer-specific?

ebaauw, 2021-03-24:
Because it isn't exposed by the REST API.  Neither is the (more or less?) standard setting.

Typically you do these settings only once, after pairing the light, so very low priority for API support.  I have no compassion for the ~~fools~~ folks who still think it's a good idea to run deCONZ without the GUI.

7wells, 2021-03-25:
Erm, thanks for having struck out the word "fools" in your post (though it sticks). It was just a (kindly meant) question from me as basic user of a Dreden-Elektronik's RaspBee II. But thanks for the explanation. And I do use deconz with its GUI (else, I would not be able to set the power stage). I agree with you that these settings are rather not done multiple times.

ebaauw, 2021-03-25:
> though it sticks
> And I do use deconz with its GUI

Sorry, I wasn’t referring to you, as, obviously, you are indeed running the GUI.

ebaauw, 2023-04-29:
> However, due to some updates in the meantime (I guess), the settings seems to be different/in another place.

That could only happen is OSRAM changed that in the light firmware, which seems highly unlikely, as they exited this business.

OSRAM doesn't support any attributes for setting the power-on state.  You need to set the desired state on the light, and then issue the _Store Power-On Defaults_ command on the 0xFC0F cluster.