dresden-elektronik/deconz-rest-plugin #97: /lights resources missing after deCONZ restart

ebaauw, 2017-08-08:
When I restart deCONZ, the resources under `/lights` are missing in the REST API, until the corresponding nodes have been discovered in the deCONZ GUI.  As it takes several minutes for the deCONZ GUI to discover all my lights (maybe related to #33 ?), this is becoming a real nuisance.  The lights are saved alright in the database.

Would it be possible to apply the same logic to `/lights` resources as for `/sensors` resources?  deCONZ shows all `/sensors` resources after restart, but with `config.reachable` false.  When the corresponding ZigBee node is discovered in the deCONZ GUI, `config.reachable` is marked true.

Not sure if it's related: the resource for my ubisys D1 dimmer sometimes shows manufacturer `Philips` after deCONZ restarts.  When I check the basic cluster in the deCONZ GUI, it hasn't been read.  When I read the attributes manually, the resource gets updated.  Again, the record in the database holds the correct values.  Note: the ubisys dimmer has an empty SW Build ID attribute.

donnib, 2017-08-08:
I agree this would be nice. I have same issue. I need to time the reboot/restart of HASS after deConz has found all lights

ebaauw, 2017-08-11:
> Not sure if it's related: the resource for my ubisys D1 dimmer sometimes shows manufacturer Philips after deCONZ restarts

This is a bug: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L1118

manup, 2017-08-13:
This will be improved with the release of the new app but can't be changed till then.

The reason is it's not possible to delete lights from the current web app. We have many installations out there dating back to 2012, there are many lights in the databases which weren't powered on for years. It would be scary if they would just pop up again :)

With the new release *fresh* lights will always be shown immediately, were fresh means they were seen by an recent deCONZ version, this will be possible by extending the database to record last seen by version/date fields.

ErnstEeldert, 2018-02-05:
@manup, any updates with regards to the roadmap of improving this?

FlippAre, 2018-08-06:
This is a bit of a pain currently. Using something that relies on the api, such as @ebaauw amazing homebridge-hue, and a reboot (due to power outage or upgrade) will in my case Apple HomeKit to forget the connected devices. Even after they eventually return, they are no longer apart of any automations and must manually be re-added and set up. The later is of course more of HomeKit issue, but the root cause is Deconz. I have some what midigated the problem by adding a delay to the launch of Homebridge after an reboot, but it feels very fragile and doesn’t work 100%. Any update on how this issue is coming along? Thanks!

ebaauw, 2018-08-06:
There's a better workaround for this behaviour, see https://github.com/ebaauw/homebridge-hue/issues/246.

ebaauw, 2018-09-07:
Fixed in v2.05.37.

closed 2018-09-07 by ebaauw