dresden-elektronik/deconz-rest-plugin #99: Feature request: configure attribute reporting through API.

ebaauw, 2017-08-08:
Now that I've updated homebridge-hue with support for deCONZ web socket events, I can monitor changes in real-time in HomeKit.  Übercool!

However, I see that deCONZ typically configures attribute reporting (for lights) with _Min Report Interval_ set to 5, _Max Report Interval_ to 180, and _Reportable Change_ to 5.  I'd like to use 1, 300, 1 instead.  I can change this manually form the deCONZ GUI, but it looks like deCONZ overwrites this on restart (except for the ubisys dimmer, but that might be related to not reading the Basic cluster correctly on discovery).

_Min Report Interval_ of 5 causes some weird user experiences when controlling lights with the _On/Off_, _DimUp_, or _DimDown_ buttons of a Hue dimmer or Trådfri remote switch (where deCONZ cannot predict the light state): with multiple button presses in short succession, the first change is reflected immediately, but subsequent changes are delayed.  E.g. see the event log when pressing the _On/Off_ button twice:
```
Tue Aug 08 2017 12:43:19: /sensors/106/state: {"buttonevent":1002,"lastupdated":"2017-08-08T10:43:19"}
Tue Aug 08 2017 12:43:19: /lights/101/state: {"on":true}
Tue Aug 08 2017 12:43:19: /groups/2097/state: {"any_on":true}
Tue Aug 08 2017 12:43:19: /groups/49285/state: {"any_on":true}
Tue Aug 08 2017 12:43:20: /sensors/106/state: {"buttonevent":1002,"lastupdated":"2017-08-08T10:43:20"}
Tue Aug 08 2017 12:43:24: /lights/101/state: {"on":false}
Tue Aug 08 2017 12:43:24: /groups/2097/state: {"any_on":false}
Tue Aug 08 2017 12:43:24: /groups/49285/state: {"any_on":false}
```

As mentioned earlier, I'd like to receive an event notification for each attribute reporting message, even when the value hasn't changed.  This way, I can build up a history log (e.g. of temperature) at the rate set by _Max Report Interval_.

I appreciate there might not be a one-size-fits-all answer to the attribute reporting configuration.  Would it be an idea to support setting this through the API, incl. persisting the configuration in the database?

ebaauw, 2017-08-08:
Incidentally, not all attribute reporting is equal.  When I set `{"bri": 254, "transitiontime": 6000}` while the light has `bri` of 1, the ubisys dimmer sends a report for each change during the next 10 minutes, where the Trådfri and OSRAM don't.  The light state is updated only on polling the light, as with Philips lights that don't support attribute reporting.

manup, 2017-08-11:
> However, I see that deCONZ typically configures attribute reporting (for lights) with Min Report Interval set to 5, Max Report Interval to 180, and Reportable Change to 5. I'd like to use 1, 300, 1 [...]

It should be ok to adjust the default values to 1, 300, 1 when they work better in general. The 5, 180, 5 was more a random pick rather than based on experience :)

> As mentioned earlier, I'd like to receive an event notification for each attribute reporting message, even when the value hasn't changed. This way, I can build up a history log (e.g. of temperature) at the rate set by Max Report Interval.

That's fine too, the event message can be extended to reflect the fact that the value has been received via attribute reporting `{ ... "via": "report" }` or something similar.

> As mentioned earlier, I'd like to receive an event notification for each attribute reporting message, even when the value hasn't changed. This way, I can build up a history log (e.g. of temperature) at the rate set by Max Report Interval.

While possible I would prefer finding good, maybe device specific, default values &mdash; since most applications likely won't set this configuration.

closed 2018-09-17 by manup