dresden-elektronik/deconz-rest-plugin #103: colorloop works in sync for every light?

wvuyk, 2017-08-09:
When I send a colorloop command through the REST API {"on":true,"effect":"colorloop"}
to light 1 and I wait a random time and send another colorloop command to light 2, both lights synchronize their colors. 

I would expect they would not be at the same color because of the differrence in time in starting it. Is this done delibarately or a miss in deCONZ?

Wim

manup, 2017-08-09:
They can't synchronize their colorloops unless you send the command to a group.
The colorloop on a light starts when command is received.

wvuyk, 2017-08-10:
Manup, 
Rechecked here and I noticed my app was not honouring the wait between the commands. Sorry, it was my bad!

Wim

closed 2017-08-10 by wvuyk