dresden-elektronik/deconz-rest-plugin #105: What is the correct way to set the color of this lamp ?

donnib, 2017-08-09:
```
{
    "etag": "f4e45d32ef08d01c50d567b88bf09af5",
    "hascolor": true,
    "manufacturer": "OSRAM",
    "modelid": "Gardenpole RGBW-Lightify",
    "name": "Bedlamper",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "xy",
        "ct": 370,
        "effect": "none",
        "hue": 43970,
        "on": true,
        "reachable": true,
        "sat": 251,
        "xy": [
            0.162594,
            0.073791
        ]
    },
    "swversion": "V1.04.93",
    "type": "Extended color light",
    "uniqueid": "7c:b0:3e:aa:00:a2:4d:04-03"
}
```
Can i just set `bri`,`ct`, `sat` and `xy` or do i have to set the `hue` as well or is it either `hue` or `xy` ?

manup, 2017-08-09:
For OSRAM lights I would recommend `hue` and `sat`.

* hue and sat
* xy
* ct

Don't mix `ct` with the first two - do either color or color temperature.

donnib, 2017-08-10:
@manup what about xy? Should i use that with hue sat ?

manup, 2017-08-10:
Xy is another more precise way to specify the color (CIE 1931 XY).

If you specify hue, sat and xy, API will pick just xy.

donnib, 2017-08-10:
Ok i understand, thank you ;)

donnib, 2017-08-10:
I might be missing something here if i set the lamp to this : `{"xy": [0.173987,0.733593]}` then i get a response from REST API :
```
[
    {
        "success": {
            "/lights/22/state/xy": [
                0.173987,
                0.733593
            ]
        }
    }
]
```
why do i get events : 
```
{"e":"changed","id":"22","r":"lights","state":{"x":11357},"t":"event"}
{"e":"changed","id":"22","r":"lights","state":{"y":47888},"t":"event"}
```

ebaauw, 2017-08-10:
The events for `xy` don't work yet, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/32#issuecomment-317168199.

donnib, 2017-08-10:
got it thx,

closed 2018-09-17 by manup