dresden-elektronik/deconz-rest-plugin #108: Question: new features or attributes for lights, sensors etc. in actual version 2.0.64

dghh, 2017-08-12:
I wonder whether i have to pair my lights, sensors etc. newly to get all the changes in the configurations ? Or will the config be updated automatically?

dghh, 2017-08-12:
Is there anywhere an overview which, beside reading all the comments here, functions are available or can be set for lights, sensors e.g. "state.dark"

manup, 2017-08-12:
>I wonder whether i have to pair my lights, sensors etc. newly to get all the changes in the configurations ? Or will the config be updated automatically?

For lights it should just work, formerly unsupported sensors and switches will be created when they rejoin the network while sensor search is active (web app > settings > open network). For that the sensors should be reset to factory state while network is open.

>Is there anywhere an overview which, beside reading all the comments here, functions are available or can be set for lights, sensors e.g. "state.dark"

The documentation lacks behind and needs to be updated especially for sensors, rules and websocket events, it will take some time though.

Till then it's best to query the state for lights and sensors and see what's there, I recommend a helper tool like Postman for that. https://www.getpostman.com/

ebaauw, 2017-08-12:
@manup, Would it be an idea to place (a copy of) the API documentation on the WiKi for this repository, so we can update it as well.

@dghh, I just implemented some of the attributes already available on the Hue API.  That's documented here: https://developers.meethue.com/philips-hue-api.

manup, 2017-08-12:
>@manup, Would it be an idea to place (a copy of) the API documentation on the WiKi for this repository, so we can update it as well.

I'll open the gh-pages repo to support pull requests to the documentation, I'd like keep the documentation concentrated to gh-pages. It's still on an old version of Jekyll but works solid.

manup, 2017-08-12:
Followup, the gh-pages docs are now a public repository.

https://github.com/dresden-elektronik/deconz-rest-doc

closed 2018-09-17 by manup