dresden-elektronik/deconz-rest-plugin #112: Philips HUE DS

donnib, 2017-08-12:


ebaauw, 2017-08-12:
In my experience, you need to
- delete the corresponding resource in `/sensors` through the REST API;
- restart deCONZ;
- Reset the dimmer;
- Open the network and make sure a new `/sensors` resource is re-created.

When factory reset, the dimmer pairs with a ight when holding the _On_ button.  This shouldn't happen once the dimmer has been paired with the gateway, but sometimes it does, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/20#issuecomment-297907098.

donnib, 2017-08-12:
@ebaauw  thanx, still it pairs fine, it's shown in deconz with a line and blinks yellow once i push it but no events are shown. The resource is in sensors. 

[image]

ebaauw, 2017-08-12:
Yeah, that looks familiar.  The deCONZ REST API plugin should setup the bindings automatically on pairing (and, I think, on restart), but you can try and do that manually in the deCONZ GUI. In the _Bind Dropbox_ create the following bindings:
- From the dimmer's FC00 cluster to the gateway (this should result in buttonevents);
- From the 0006 cluster to the group linked to the dimmer;
- From the 0008 cluster to the group linked to the dimmer;

The last two are needed so the dimmer interacts directly with lights in the linked group, even when the gateway is down.  I suppose there should be a binding from the 0005 cluster to the group as well, but I don't think it's used.  Also, two of my Hue dimmer switches, with older firmware than the other five, lack this cluster (they report themselves as non-color controller instead of non-color scene controller).  Not sure if this is just the firmware, or also some hardware limitation.

donnib, 2017-08-12:
Never tried binding manually, do i choose de00 cluster on the gateway as destination ?

ebaauw, 2017-08-12:
No, the ZHA endpoint:
[image]

donnib, 2017-08-12:
Hmm, i must be stupid, can't get it to work. I had to press on on the remote before it said binding success but still no events. Is the factory defaults the pin button on the back of the remote or the holding the 4 buttons down ?

donnib, 2017-08-12:
I did it again, held on to long then the remote didn't work anymore, very annoying, anything i can do to stop that from happening ?

ebaauw, 2017-08-12:
Factory reset is the pin button on the back for 10s.  Holding all four buttons is just re-establishing the network connection (or something like that).

You really need to delete the /sensors resource to get a "clean" pairing with deCONZ.  Somehow only that tells the remote to stop trying to form a network by itself.

donnib, 2017-08-12:
I did DELETE the resource AND also deleted in deconz ui. I got it back but it's unreliable, What does red led mean when i push ? Failed ?

ebaauw, 2017-08-12:
> I did it again, held on to long

Same thing happened to me just now (while sniffing the dimmer buttons): held down _On_ and the nearest light starts to blink.  Apparently I let go of the button just in time, for the dimmer still works.  I did see some network rejoin requests and now it has a new NWK address.

I think @LunaSela is right in that an additional setting might be needed.

ebaauw, 2017-08-12:
> What does red led mean when i push ? Failed ?

No contact with the network.  Can you see from the deCONZ GUI whether it connects to the RaspBee / ConBee or to one of the IKEA lights?  Do you get events when you power off the IKEA lights (forcing the dimmer to connect to the RaspBee/ConBee directly)?

manup, 2017-08-13:
Instead of reset, if the remote (or sensor) is already in the network I recommend:

* pull out the battery for a few seconds
* Web App > Settings > Open Network
* put battery it in again

deCONZ will then recreate the bindings if necessary.

The same goes for lights, turn off for a few seconds and turn on again is much easier than reset and has basically the same effect.

Resets — should always be the last resort if the device is already in the network.

donnib, 2017-08-13:
Sure but it seems very unstable, it comes and goes. I tried your suggestion without it helped.

donnib, 2017-08-13:
I must admit that i can't figure out how anybody would find these Dim Switches reliable, so goes for the IKEA ones. They work as the wind blows. The IKEA switch at least stays paired but the Philips is horrible on that. Even when it's paired i push the button and no events, i push it again then event, it's just not usable. Is it really so for users using this original gateway and hue lamps. Again same goes for the IKEA one. Horrible experience. The lights work perfectly fine when i turn them off from HASS or thru the HomeKit plugin. I have a hard time believing people pay for these in this condition.

manup, 2017-08-13:
Because usually they are very reliable. The behavior is not normal, the problem could be elsewhere, maybe the gateway is in bad position or Wifi/Bluetooth interference. There are apps which show how much specific 2.4 GHz channels are occupied, that can be used to figure out which ZigBee channel is more appropriate.

>  I have a hard time believing people pay for these in this condition.

Again, I don't think this is device specific, wireless can be tricky if the environment is difficult.

donnib, 2017-08-13:
I guess if the problem was interference it would also be shown on the lights which have no problems, same goes for the ikea motion sensor and philips motion sensor, they all work well. I am trying to use the remote very close to the gateway but i can see from the lines in deconz it meshes thru some lights.

manup, 2017-08-13:
Or the lights to something strange, in this case they have to forward the commands. As much as I'd like to blame Philips their devices were always the most reliable ones and sticking to the standard. It might be a problem with the remote but from my experience it's very unlikely.

Anyway to test if the lights are the problem: If the remote is near the gateway you can force it to connect directly to it by power off the lights until the remote connects to the gateway.

donnib, 2017-08-13:
I just took a brand new dim switch out of the box, opened network, pulled the battery plastic out and the remote turned on and paired with the gateway but even in this case the remote does not work, no events from the remote. Maybe the new mesh capability does something here ? I'll try your suggestion.

donnib, 2017-08-13:
btw, how do i know if the binding are made ? Is there a way to check ?

donnib, 2017-08-13:
> Anyway to test if the lights are the problem: If the remote is near the gateway you can force it to connect directly to it by power off the lights until the remote connects to the gateway.

So that did indeed help, i turned all of the lights nearby (not all that were further away) then i repaired the remotes and it works fine now so the lights (IKEA) are doing something funny

donnib, 2017-08-13:
I did some stress test of both dim switch remotes at the same time and they seemed perfectly fine, even if i hold the on button down they work great fine now.

donnib, 2017-08-13:
As soon as i turned on the lights one of the remotes started to not work anymore, I see the blue light blink in deconz when i push the buttons but no events.

manup, 2017-08-13:
Ah I see, the commit from earlier today might fix it. At least the problem looks the same: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/faccb2c6ad1fab4fdc8e0b0852e9fd9323b91bc0

donnib, 2017-08-13:
Unfortunately i have that already, i am on 2fb7016

manup, 2017-08-13:
Hmmm damn, tricky.

Can you please provide debug output then commands are received from the remote and no events are fired? Since the node is blinking blue at least something must be received.

`$ deCONZ --dbg-aps=1 --dbg-info=1`

manup, 2017-08-13:
I forgot: since you are using official image you may have to disable watchdog before stopping deCONZ, otherwise system will reboot.

`$ sudo systemctl stop watchdog`

donnib, 2017-08-13:
Sure ill do that in the morning. My deconz starts automatically so how do i lunch the command ? Which folder is it located in? It's good you mentioned the watchdog because I had been wondering why my pi was rebooting when I quit deconz.

donnib, 2017-08-14:
@manup here is the log output https://gist.github.com/donnib/9d358e71dbe57fc478ce74a709f54df5  

ID 11 in the rest api is the one that does not work, ID 12 is the remote that works. I can see in deconz that exactly this ID 11 is one that is connected through a light were the other one thru the gateway.

```
{
    "config": {
        "battery": 100,
        "group": "10879",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "bb7414211157d7698899c50874880fcc",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "RWL021 11",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2017-08-13T16:36:13"
    },
    "swversion": "5.45.1.17846",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:02:19:c6:23-02-fc00"
}
```
```
{
    "config": {
        "battery": 100,
        "group": "9482",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "b4f677280ada79bebc33a4ddad96a0be",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "RWL021 12",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2017-08-14T14:50:08"
    },
    "swversion": "5.45.1.17846",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:02:e2:68:15-02-fc00"
}
```

donnib, 2017-08-14:
So the switch have a line green to one of the IKEA GU10 spots i have which runs latest version 1.2.217.

1. I tried this and it didn't help.
2. I turned off the GU10 spots, the line was still green on the remote to the GU10, i guess it takes sometime before they are shown as not reachable. I tried pushing the buttons on the remote many times nothing changed. 

Here is an updated log file, continued from the one earlier so it's big. https://gist.github.com/donnib/9d358e71dbe57fc478ce74a709f54df5/060ff81d3076e4c9684314460fadb56bc8a8c970

I'll update if anything changes when the lights (the GU10 spots the remote think it's connected to) turn red in deconz

manup, 2017-08-14:
Hmm very strange, I'll test same scenario tomorrow with an dimmer and IKEA GU10 would be odd if IKEA lights can't forward unicast end device commands.

meanwhile.. I'll try to test with another end device now

donnib, 2017-08-14:
@manup Great, looking forward to hear from you, hope we can nail it down.

donnib, 2017-08-14:
So after 30min the GU10 lights finally became red in deconz. The line between the dim switch and the GU10 disappeared but after pressing many times on the remote it has not reestablished a new line to another light and it's still doesn't work (do not get events)

manup, 2017-08-14:
By looking in the sniffer I see the gateway has an issue, in my setup currently it doesn't allow to rejoin the hue motion sensor over an IKEA E27 light into the network. I need to dig deeper into the firmware to check whats going on.

donnib, 2017-08-14:
Interesting, yesterday when i had trouble i am pretty sure then it was connected to the E27 ikea lamp

manup, 2017-08-14:
Something is really strange here.

* Hue sensor rejoin via Hue E27 OK
* Hue sensor rejoin via dresden elektronik FLS OK

ebaauw, 2017-08-14:
> By looking in the sniffer I see the gateway has an issue, in my setup currently it doesn't allow to rejoin the hue motion sensor over an IKEA E27 light into the network. I need to dig deeper into the firmware to check whats going on.

I have a feeling that the gateway has indeed an issue, but I cannot quite put my finger on it.  Two of my motion sensors somehow dropped of the network, I had to restart them (short press on the pin hole) twice, before they worked again.  Once to join the network, once again for attribute reporting to be setup.  In both cases, while the network was open.

I've seen twice now in de deCONZ GUI that all the status dots on the gateway and router nodes are solid yellow, and the gateway doesn't seem to send any command to the lights.  It is still updating it's state from attribute reports though.  The only remedy is a power cycle of the RaspBee - rebooting the Raspberry Pi isn't enough.

> Hmm very strange, I'll test same scenario tomorrow with an dimmer and IKEA GU10 would be odd if IKEA lights can't forward unicast end device commands.

I paired the IKEA bulb with the Hue bridge (just to see what that looked like), and then rejoined it to the deCONZ network.  Now, I cannot get it to react to the remote's _Previous_ and _Next_ buttons, even though the remote's bindings haven't changed.  I did notice the remote was now attached to one of my Philips lights, causing me to think that maybe they don't forward the (non-standard) scene commands.  However, now I have the remote connected to the IKEA bulb (according to the deCONZ GUI) and it still won't work.  I definitely need to sniff some more...

manup, 2017-08-14:
> I have a feeling that the gateway has indeed an issue, but I cannot quite put my finger on it. Two of my motion sensors somehow dropped of the network, I had to restart them (short press on the pin [...]

A sniffer log would be helpful here, in my above test the gateway wasn't involved and in fact turned off to check if the motion sensor rejoins to a new parent.

ebaauw, 2017-08-14:
> A sniffer log would be helpful here

Of what (or rather: when)?  Shall I just leave BitCatcher running overnight?  My "production" network is 61 nodes (1 gateway, 39 routers, 21 end devices), creating a lot of noise.

I do see that the sniffer regularly misses some frames, especially unicasts.  For the Hue bridge sniffing, I placed the bridge next to the ConBee in my MacBook, but the Raspberry Pi with the ConBee is tucked away nicely - I cannot put my laptop next to it.

I also see an occasional frame from another PanID.  Could this be corrupted frames (due to collisions?) or are my neighbours also on channel 25?

manup, 2017-08-14:
The interesting part is then the sensors lose connection for some reason, usually they try to rejoin the network. This can be analysed later by filter the sniffer log for sensor NWK address.

In my tests from today hue motion sensor rejoin fails if the chosen parent is a IKEA light, it works with all other brands, albeit the IKEA light does the rejoin response right &mdash; or I miss something here.

In the worst case scenario the hue sensor has a mac address filter, I don't think they do that but I will check anyway :D

>I also see an occasional frame from another PanID. Could this be corrupted frames (due to collisions?) or are my neighbours also on channel 25?

Either that or a orphan device somewhere, in my home I see another nearby hue network too.

manup, 2017-08-15:
Its still not clear to me what happens exactly, but the setup IKEA + Hue motion sensor mesh works now in office. The new deCONZ version *2.04.65* contains a new firmware version *0x26130500* which fixes an issue related to network join/rejoin. It may help here but I'm not sure if it's the root cause.

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.65-qt5.deb

donnib, 2017-08-15:
@manup thx, i have done the upgrade remotely, when i get home i'll see if the remote works without doing anything else, however after upgrade and restart of deconz i see the remote ID 11 after some time making a green line between one of the E14 ikea lights and if i try to read attributes on the basic cluster nothing happens. I can't try physically the remote since i am not atm. What should i try now ? Reboot rPI ? Total power off->on of the rPi ? Reset remote ? Take battery out then put it back in and open network ?

manup, 2017-08-15:
The remote sleeps, you need to press the physical buttons in order to wake it up and see if it works.

Please **don't** reset the remote or  reboot the RPi.

The only helpful thing (if it is not working already) is pull out battery and put in again, as well as turn off lights and turn on again from mains power.

donnib, 2017-08-15:
So i came home and the only thing that i did was turn off/on the lights and the remote started to work, it's connected (indicating with the a green line) to a IKEA E14 light, next thing i did was to turn off the lights (IKEA E14) to see what then would happen but the remote kept working, the line still indicates that it has a connection to E14 which obviously is not possible so it must mesh thru something else which i can't see, is there a way to see that?

ebaauw, 2017-08-16:
v2.04.65 seems more resilient: one of my motion sensors was "lost" again last night.  I left deCONZ running, not touching the sensor, and it found the sensor and re-established the attribute reporting bindings.

manup, 2017-08-18:
@donnib  are there any news on the problem, did the 2.04.65 version fix the lost hue dimmer problem?

> [...] the line still indicates that it has a connection to E14 which obviously is not possible so it must mesh thru something else which i can't see, is there a way to see that?

After some time the old lines will go red and disappear. The lines between nodes are based on neighbor table entries which are queried from the lights periodically, it's not a realtime view but will always adapt through time.

ebaauw, 2017-08-18:
> it's not a realtime view

Which would also explain that sometimes the GUI shows two connections for an end device:
[image]

donnib, 2017-08-19:
@manup yes i believe this specific problem did get resolved with 2.04.65 so i will close this one

closed 2017-08-19 by donnib

donnib, 2017-08-21:
@manup So i am not sure the problem is related to same thing but same decice Hue Dim Switch decided to forget its name so it went back to its default name it had when it was joined. Idk if this is related somehow that maybe it rejooned and got reset ?