dresden-elektronik/deconz-rest-plugin #114: Support for Netvox Z716A

Hector47, 2017-08-13:
Hello, how deCONZ can support the Netvox Z716A  Humidity/Temperature Sensor ?

You can find the cluster list in the documentation but they doesn't  appear in deCONZ.
[Z716A_UserManual_ENG_WEB.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1220662/Z716A_UserManual_ENG_WEB.pdf)

Do I need to update the ZCLDB ?

Can it appear in the sensors part of the rest plugin ?

Thanks

manup, 2017-08-16:
Hi Hector,
from the documentation it looks like the device can be added with little afford, I'll put something together for the next release.

The clusters described are already in the ZCLDB, but a few things must be added to the REST API to support the sensor.

manup, 2017-09-10:
Hello hector can you please provide a screenshot of deCONZ *Node Info* tab with the sensor selected?

Hector47, 2017-09-11:
[image]
Do you need more info ?

manup, 2017-09-11:
The sensor is added to the supported devices, you may compile the plugin with latest commits or wait till 2.04.77 is available.

In order to create the REST API resources do:
* *WebApp > Menu > Settings > Open Network*
* reset the sensor (since it is already visible in deCONZ a simple power cycle, battery out for few seconds and in again, should be sufficient.

The API should then create two sensors.