dresden-elektronik/deconz-rest-plugin #118: Battery reading from IKEA remotes maybe not right ?

donnib, 2017-08-16:
Hey,
Looking at all remotes i get a reading of 30 - 50% which i thought was fine until i changed the battery on the one that reported 30% then the remote reported 50%...hmmmm so maybe a divide by two error somewhere ? So maybe 50% is 100% ?

donnib, 2017-08-17:
@ebaauw have you seen your remote from ikea report more than 50% battery ?

ebaauw, 2017-08-17:
No, deCONZ supports `config.battery` for the remote only after I updated its firmware.

The ZigBee standards uses attribute 0x0021 of the _Power configuration_ cluster (0x0001) to hold the battery charge in half percentages (i.e. from 0 to 200).  All my Hue motion sensors and dimmers still report value 200 for this attribute.  I think I might have seen values over 100 for the remote in the deCONZ GUI, but I have no record for that, so I'm not sure.  I'll check once I'll have replaced the remote's battery.  It's not unprecedented that IKEA take a liberal interpretation of the ZigBee standard...  Does the IKEA app hub show the battery percentage when the remote is connected to the IKEA hub?

I understand from @manup that tracking the remaining charge for button-cell batteries is notoriously unreliable, so the value reported should be taken with some scepticism.  I had one of my Hue dimmer switches report 87% or so for a while (when it was still connected to the Hue bridge), but it reverted to 100% spontaneously.

What did you use as replacement battery?  There are huge price differences between the IKEA branded CR2035: EUR 1,27 for an eight pack, vs C2035 batteries from regular brands (Duracell, Varta, Panasonic): EUR 3,75 a piece.  They're all listed as 3V Lithium cells, but I cannot believe they would have the same capacity.

donnib, 2017-08-17:
I don't recall seeing battery in the IKEA app using the Hub.

I replaced the battery with a VARTA Pro CR2032 battery so the quality should be top notch.

manup, 2017-08-17:
I've seen low battery warning once in the IKEA app but no state so far. As @ebaauw  mentions the ZigBee standard defines the value as 0..200, maybe IKEA has another interpretation of the definition, maybe the value is just not very exact or maybe only the very low voltage warning is reliable.

I'm not sure how IKEA calculates the battery state but from our electric engineer I know it is *very* tricky to get reliable results for these kind of batteries.

ebaauw, 2017-08-18:
I went to IKEA to get a pack of PLATTBOJ CR2032 batteries.  Over there, I heard a little voice whispering: "Buy me!"  I'm weak, so now I'm in the happy possession of a Trådfri motion sensor  ;-)

Just installed the two CR2032 batteries that came with the sensor: _Battery Percentage Remaining_ (0x0001/0x0021) reads 100 (and, consequently, the rest API shows 50%).  _Battery Voltage_ (0x0001/0x0020) reads 30, 3.0V, so apparently the batteries are wired in parallel.  
The sensor is still on the old firmware - I'll report back after it's updated.

After admitting to myself that I'm weak, I figured I might as well read the manual.  It reads that "A red LED indicator will shine steadily when it's time to replace the batteries," which should last ~2 years "when the unit is used regularly and as intended."  I don't suppose that means regular firmware updates ;-)  
Double checking the remote's manual (online), it reads: "When it’s time to replace the battery, a red LED indicator will flash when you press any button on the remote control."

ebaauw, 2017-08-18:
After the firmware update, the motion sensor reports 87, which deCONZ translates to 43%, and still 30 (3.0V).

My remote was reporting 47 and 28 (2.8V).  I put in a fresh PLATTBOJ, and now it reports 100 and 31 (3.1V).  But back in the old battery, it reported 60 for a short while, then went back to 47.

> maybe a divide by two error somewhere ? So maybe 50% is 100% ?
> maybe IKEA has another interpretation of the definition

I'd say yes (although it's rather a "forgot to multiple by two" error on IKEA's side).  I'll see if I can create a PR to workaround this.

> so apparently the batteries are wired in parallel.

Indeed, the motion sensor works with only a single battery installed.

manup, 2017-08-19:
My IKEA motion sensor reports 60 after 2&ndash;3 months. I think for now it's safe to assume that the value for IKEA battery status shouldn't be divided. Happy to see an PR :)

ebaauw, 2017-08-20:
> Happy to see an PR :)

#126

manup, 2017-08-20:
Thanks it works, the API now reports correct value of 60 for my IKEA sensor.
I close the issue for now.

closed 2017-08-20 by manup