dresden-elektronik/deconz-rest-plugin #119: deConz has lights as red but i can still control them

donnib, 2017-08-16:
Hi,
I have seen few times some lights go red in deConz, i tried 0 on the keyboard and nothing happened. In REST API the lights is `"reachable": false` but i can still control the group the light is in and it turns off/on without issues. 

Maybe connected to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/33 ?

ebaauw, 2017-08-16:
I find that hitting 0 works on: Hue bulbs and spots, OSRAM bulbs and plug/in unit, ubisys dimmer, and Trådfri bulb, but not on: Hue Blooms,  Hue Lightstrips, nor innr lights.

I experience exactly the same symptoms.  It's seems a bit better with 2.04.65, fewer red nodes, but sometimes most of the lines on the deCONZ GUI have disappeared.  I find that lights that support attribute reporting (OSRAM, ubisys, IKEA), as well as the Hue bulbs and spots and the most consistent.  Not sure if that's related to the reacting to 0, or just coincidence.

donnib, 2017-08-17:
So the lights that were red were IKEA and 0 did not do anything. It really annoying. Not sure what causes this. 
@manup is this 799abab503bd44e629428cdf803297f6a97d2ea4 and 3fe405450b353d3782651c21de86d3f7c24e620b related to this ?

manup, 2017-08-17:
No the commits are related to just the sensors. The next versions will improve in regards to red nodes. I have done some sniffing and found a few issues which can be improved but need some testing.

manup, 2017-08-19:
There is a new version 2.04.66 which should improve the red nodes problem.

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.66-qt5.deb

donnib, 2017-08-21:
Today i see again 3 nodes going red, i am not home to see if i can still control them but never the less they are red, one of them is within 3m from the gateway, 2 of the nodes has no lines, one of them has a line but when i move the node in deconz the line does not follow along, drawing issue or something else. All 3 nodes beeing IKEA Lights.

donnib, 2017-08-21:
So it seems i can control the lights fine even though they were not even shown in deconz. I did a power off/on after i did see i could control them then they showed up in deconz ui.

manup, 2017-08-21:
I've compiled a experimental firmware for RaspBee/ConBee which has `many-to-one routing` also called source routing enabled. This routing type is used by IKEA and OSRAM lights, maybe it works better than classic mesh routing for these lights.

It's still untested but you may give it a shot:

http://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x26140500.bin.GCF

**Installion RaspBee**
```
1) disable watchdog (if dresden elektronik sd-card image)

  $ sudo systemctl stop watchdog

2) close deCONZ

3) update RaspBee firmware

  $ sudo GCFFlasher_internal -f deCONZ_Rpi_0x26140500.bin.GCF
```

manup, 2017-08-21:
Hmm rollback, it seems that the lights work better but somehow routing for end devices connected directly to the RaspBee is messed up.

donnib, 2017-08-22:
@manup that's a shame :(

manup, 2017-08-22:
Yes but at least I think it's the way to go, the source routing seems to work well in my setup. I'm checking now what's going on with the end devices. This stuff must be ironed out.

donnib, 2017-08-22:
Sounds great, i have 99% ikea lights and one osram light, what would happen when/if i add philips lights which as you say use different routing ? Would it be able to coexist and work together ?

manup, 2017-08-24:
@donnib please test the new version 2.04.68, after installation the web app will ask to update firmware as well (firmware version 0x26160500). This version has source routing enabled (for IKEA and OSRAM).

After firmware update is finished it may take a few minutes to establish routes, red nodes should go away afterwards.

`$ sudo dpkg -i deconz-2.04.68-qt5.deb`

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.68-qt5.deb

donnib, 2017-08-24:
@manup great, i will give it a go and let you know how it performs. Do you have a comment for my question here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/119#issuecomment-323958800 ?

manup, 2017-08-24:
>Sounds great, i have 99% ikea lights and one osram light, what would happen when/if i add philips lights which as you say use different routing ? Would it be able to coexist and work together ?

This works fine, I have also a mixed network with both mesh and source routing devices.

donnib, 2017-08-24:
@manup Great, i did the upgrade now, i had few nodes that were lost in deConz prior to upgrade (e.g they disappeared from deConz completely not even red nodes) should i try power on/off then leave them to see how stable they are ? 

Should this release help that problem as well that nodes disappear and don't reappear again unless you cycle power on them ?

manup, 2017-08-24:
>  should i try power on/off then leave them to see how stable they are ?

To see if it figures our everthing on its own you may let it just run for an hour or so.
If it doesn't work our you may power cycle the related lights.

>Should this release help that problem as well that nodes disappear and don't reappear again unless you cycle power on them ?

I think the release should improve this case but needs some more testing, I hadn't had this particual issue in my setup yet.

snozzlebert, 2017-08-30:
Since 2.04.69 / 0x26160500 (skipped .68) I have a problem with my three oldest bulbs (1st gen Philips A19 color). They appear red (with black dot) in the deCONZ gui after starting deCONZ. After power cycle them they become yellow again.
Other Philips bulbs and OSRAM plug almost immediately appear yellow in the gui.
With versions/firmware before 0x26160500 all lights appear almost immediately yellow after starting deCONZ.
Is this related to the new routing mechanism?

Another issue related to firmware updates:
After upgrading to a new deCONZ version (e.g. from 2.04.69 to 2.04.70) I get a notification of a firmware update while there is actually no update (my ConBee was already on 0x26160500). The rest api returned:
```
fwversion: 0x00000000
fwversionupdate: 0x26160500
```
This probably triggers the notification. Is this a known issue? Anyone having the same experience?

manup, 2017-08-30:
> Since 2.04.69 / 0x26160500 (skipped .68) I have a problem with my three oldest bulbs (1st gen Philips A19 color). They appear red (with black dot) in the deCONZ gui after starting deCONZ. After power cycle them they become yellow again.

It's related to deCONZ 2.04.69.

There were some 'ongoing' changes which may cause this behavior, however deCONZ should detect the lights by it's own, but it may take a while (depending on network size a few minutes).

> Another issue related to firmware updates:
After upgrading to a new deCONZ version (e.g. from 2.04.69 to 2.04.70) I get a notification of a firmware update while there is actually no update (my ConBee was already on 0x26160500). The rest api returned:

Have you by any chance updated Raspbian to latest stretch? The GCFFlasher which is used for fw updates doesn't yet work with stretch and hence causes exactly this error. We're on it to provide an update.

snozzlebert, 2017-08-30:
> Have you by any chance updated Raspbian to latest stretch?

No, I'm still on jessie
```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
Linux 4.9.35-v7+ armv7l
```
I had this after all deCONZ updates since I start using my ConBee for the first time using deCONZ 2.04.54. May be it is related to using ConBee (usb) instead of the RaspBee which most people use in combination with the Raspberry Pi.

manup, 2017-08-30:
Ok I'll check this. Albeit the ConBee firmware update should work on Raspbian, it was mostly testet on Windows and Ubuntu. Should be fixed during the next days.

snozzlebert, 2017-08-30:
The firmware update **does** work. However deCONZ doen't report the right firmware version to the rest api after installing a new version of the deCONZ software (not firmware). The about box in the GUI does show the right version after updating the software (e.g. 0x26160500). The rest api doesn't (0x00000000). After clicking "update firmware" in the web application, deCONZ starts flashing the firmware (while it is already on that firmware). After flashing the rest api shows the right version again.

ebaauw, 2017-08-30:
Where (what resource/attribute) does the REST api report the firmware version?

snozzlebert, 2017-08-30:
Resource: ```config```
You need to add the 'secret' Accept header with value "vnd.ddel.v1" to your request. The deCONZ Web App does the same. You will get more attributes than a regular request.
Attributes:
```
    "fwneedupdate": false,
    "fwupdatestate": "idle",
    "fwversion": "0x26160500",
```
Attribute ```fwversionupdate``` is available if there is an update.

snozzlebert, 2017-09-02:
My problems mentioned in this issue are solved in 2.04.73 Very fast initialisation of the network! I only have 14 nodes at the moment, however compared to previous version it is much faster.
Also no notification for firmware update.

posixx, 2017-09-05:
Yes i can confirm; version 73 is much faster in initialisation of the network after a power cycle. In my network with 70+ devices it took about 20 minutes before, now it takes about 5 with 95% of the network available after 1 minute! good work!

closed 2018-09-17 by manup