dresden-elektronik/deconz-rest-plugin #120: ZigBee attribute reading and reporting

ebaauw, 2017-08-17:
I'm trying to get the deCONZ REST API plugin to support the additional attributes for the Hue motion sensor (see issue #20).  I run into some issues which I think are directly related to the deCONZ software architecture, and are way beyond my current understanding (and ambition) to solve.  @manup, I would love to hear your view on this.

I apologise if I'm too forthcoming (I'm Dutch).  I really appreciate deCONZ and the fact the you've open sourced the REST API plugin.  I hope my analysis below can help to improve the product.

1. The C++ deCONZ plugin API notifies a plugin when data in a cluster has been updated, through a ZCL _Read attributes_ command or a ZCL _Report attributes_ command (class `deCONZ::NodeEVent` in `/usr/include/deconz/node_event.h`).  However, it doesn't indicate what data has been updated;
2. Of course, a plugin can keep a copy in memory of the cluster data and compare the cached values with the actual values to determine which values have actually changed (I use the same pattern in homebridge-hue when polling the Hue bridge or deCONZ gateway).  The REST API plugin does so in class `RestNodeBase` and in `DeRestPluginPrivate::updateSensorNode()`.
3. However, there's no way for a plugin to tell the difference between an unchanged attribute that was included in the _Read attributes_ or _Report attributes_ command and an attribute that wasn't included in the command in the first place.
4. This isn't an issue when all attributes (of interest) are included in every single ZCL command.  I suppose this was the case when the deCONZ REST API was designed, but there is one exception: the ZCL attributes corresponding to a ZHAPresence sensor's `state.presence` (0x0000) and `config.duration` (0x0010) are both attributes of the _Occupancy sensing_ cluster (0x0406).  However, they're polled separately (in `DeRestPluginPrivate::processZclAttributes()`), resulting in separate _Read attribute_ commands.  Also, attribute reporting is only setup for `state.presence`.
5. After starting deCONZ, the first `NodeEvent` notification for the _Occupancy sensing_ cluster is for 0x0000 only.  When `DeRestPluginPrivate::updateSensorNode()` processes this event, the cache copy of 0x0010 hasn't yet been initialised with the actual ZCL attribute.  `DeRestPluginPrivate::updateSensorNode()` wrongly assumes that the ZCL attribute differs from `config.duration` (restored from the database) and sends a _Write attribute_ command to the sensor.
6. When `DeRestPluginPrivate::updateSensorNode()`processes the `NodeEvent` notification for the _Occupancy sensing_ cluster due to polling 0x0010, it wrongly updates `state.lastupdated` and issues a state changed web socket notification for `state.presence`.  This is causing the two notifications per five minutes for `state.presence`, where `state.lightlevel` and `state.temperature` are only notified once per five minutes.
7. These issues might be solved, by combining both attributes in a single _Read attributes_ and _Configure Reporting_ command (effectively deprecating the need for polling 0x0010 and a separate `READ_OCCUPANCY_CONFIG` flag in `DeRestPlugin::idleTimerFired()`).  I haven't tested yet whether this works.  This would require refactoring `DeRestPluginPrivate::sendConfigureReportingRequest()` to include multiple attributes.
8. We might need a similar approach to support attribute reporting for a Color light or Extended color light, needing multiple attributes from the _Color control_ cluster (0x0300) besides `state.ct`.  Incidentally, from the ZigBee Light Link Standard (Version 1.0), section 6, ZCL usage and enhancements: _"Attribute reporting shall not be used in this profile."_  Maybe IKEA (and OSRAM for their US firmware) had a good reason to use ZHA instead of ZLL?  And (I hate to admin) Philips has a valid reason not to support attribute reporting for their ZLL lights.
9. This approach will _not_ work for `config.sensitivity` and `config.sensitivitymax`.  These are manufacturer-specific attributes (0x0030 and 0x0031) in the _Occupancy sensing_ cluster and cannot be combined in a single command with standard attributes (I did try this).  Also, it doesn't make sense to setup attribute reporting for `config.sensitivitymax`, as it's constant (except maybe for firmware updates).  I'm not even sure if the Hue motion sensor would even support it.
10. This approach will _not_ work work `config.ledindication` and `config.usertest` either.  These are manufacturer-specific attributes (0x0033 and 0x0032) in the _Basic_ cluster (0x0000).  As no attribute reporting for the other attributes in this cluster is used, we probably could update 0x0033 and 0x0032 only from a `NodeEvent` caused by _Report attributes_.

manup, 2017-08-17:
*First things first* :)

The polling and configuration of sleeping end devices is currently not very well implemented, it will be more solid soon, once the polling is refactored and works more reactive for end devices. With the current implementation, sending commands to sleeping end devices may just lock up the sending queues and result in more red zombie nodes. The handling of `unoccupied to occupied delay` is available for Philips Hue motion sensor by accident, its origin is a mains powered motion sensor which doesn't sleep.

Therefore I think it's best to put more (automatic) polling on hold until refactoring is done. And just implement the rx handlers for the attributes. I'll add functions so that cluster info reading of manufacturer specific attributes will be possible.

> 1. The C++ deCONZ plugin API notifies a plugin when data in a cluster has been updated, through a ZCL Read attributes command or a ZCL Report attributes command (class deCONZ::NodeEVent in /usr/include/deconz/node_event.h). However, it doesn't indicate what data has been updated;

Currently the info on what has changed is only available for the cluster but not which attribute(s), I think it makes sense to include the related attribute ids in the event object.

The NodeEvent is just *sugar* to keep it simpler, *all* received packets, including read and report attribute frames, are also forwarded to `apsdeDataIndication()`.

> 3. However, there's no way for a plugin to tell the difference between an unchanged attribute that was included in the Read attributes or Report attributes command and an attribute that wasn't included in the command in the first place.

As mentioned above, extending the NodeEvent should solve the issue, albeit it would also be possible to dissect the reports entirely in the REST plugin.

> 5. After starting deCONZ, the first NodeEvent notification for the Occupancy sensing cluster is for 0x0000 only. When DeRestPluginPrivate::updateSensorNode() processes this event, the cache copy of [...]

> 6. When DeRestPluginPrivate::updateSensorNode()processes the NodeEvent notification for the Occupancy sensing cluster due to polling 0x0010, it wrongly updates state.lastupdated and issues a state [...]

Should also be solved by extended NodeEvent object. Also the presence event should only fire if the actual state has changed.

ebaauw, 2017-08-17:
> _First things first :)_

Thanks, @manup.  I appreciate your input!  With this, I'm confident we can bring the Hue motion sensor support in deCONZ at the same level as the Hue bridge.   Actually at a higher level due to `config.duration`!

> Therefore I think it's best to put more (automatic) polling on hold until refactoring is done

Agreed.

> I'll add functions so that cluster info reading of manufacturer specific attributes will be possible.

I already added an optional parameter `manufacturerCode` to `DeRestPluginPrivate::readAttributes()` and `DeRestPluginPrivate::writeAttribute()`, see  https://github.com/ebaauw/deconz-rest-plugin/commit/6e1723960031a0fd746bd2e1ee85c162a9dc4b6d.  I'm happy to create a PR, but I'm not sure if you can just pick a single commit from that.

> [1] Currently the info on what has changed is only available for the cluster but not which attribute(s), I think it makes sense to include the related attribute ids in the event object.

Cool.  It would also make `DeRestPluginPrivate::updateSensorNode()` simpler and more efficient, as we can just loop over the attributes in the NodeEvent, instead of the attributes in de RestNodeBase.

> The NodeEvent is just sugar to keep it simpler, all received packets, including read and report attribute frames, are also forwarded to `apsdeDataIndication()`.

Cool. I hadn't found that.

> [3] As mentioned above, extending the NodeEvent should solve the issue, albeit it would also be possible to dissect the reports entirely in the REST plugin.

Please do extend NodeEvent - dissecting ZigBee frames goes way too far for a hobby  ;-)

> [5 and 6] Should also be solved by extended NodeEvent object. Also the presence event should only fire if the actual state has changed.

The Hue bridge does update `state.lastupdated` when it receives a _Report attributes_ command (i.e. when `presence` is changed, or every five minutes).  Same for `lightlevel` and `temperature`.  I agree, that you might want to omit `presence` when it hasn't changed, but I would still like a web socket event for `lastupdated`.  Maybe this discussion should be a separate issue?

I'll be happy to prepare these changes, once the (specs for the) extended NodeEvent is available.

> And just implement the rx handlers for the attributes.

Shall I also try to change `DeRestPluginPrivate::sendConfigureReportingRequest()` and combine 0x0406/0x0000 and 0x006/0x0010 in one _Configure Reporting_ command?  And use `manufacturerCode` there as well?

ebaauw, 2017-08-20:
> 7. These issues might be solved, by combining both attributes in a single Read attributes and Configure Reporting command.

Except that the Hue motion sensor doesn't see, to support attribute reporting on 0x0406/0x0010.

closed 2017-09-10 by ebaauw