dresden-elektronik/deconz-rest-plugin #121: Hue Tap buttonevent wrong value?

posixx, 2017-08-17:
I get some strange buttonvalues coming in through the websocket when pressing hue tab buttons from different hue taps instantly after eachother. See these websocket responses (log from JowHue Beta, which uses websocket):

 17:39:30.1786181 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:39:34"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:39:33.5290475 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:39:37"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:39:35.3430579 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:39:39"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:39:37.3192824 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:39:41"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:39:45.4675485 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":100,"lastupdated":"2017-08-17T15:39:49"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:39:45.6276855 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":101,"lastupdated":"2017-08-17T15:39:50"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:39:46.9490611 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":100,"lastupdated":"2017-08-17T15:39:51"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:39:47.1716388 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":101,"lastupdated":"2017-08-17T15:39:51"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:39:49.2186424 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:39:53"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:44:34.4360596 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:44:38"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:17.6591647 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:45:22"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:25.8462069 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:45:30"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:27.3051321 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:45:31"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:29.2869826 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:45:33"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:31.1882809 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":100,"lastupdated":"2017-08-17T15:45:35"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:31.3143270 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":101,"lastupdated":"2017-08-17T15:45:35"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:33.2378790 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:45:37"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:34.6082073 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:45:39"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:36.3650363 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:45:40"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:37.8266131 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:45:42"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:39.4803412 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":100,"lastupdated":"2017-08-17T15:45:43"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:39.7133597 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":101,"lastupdated":"2017-08-17T15:45:44"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:45:41.4770063 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:45:45"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:57:19.2470707 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:57:23"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:28.5592654 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:59:32"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:45.4246295 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:59:49"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:47.2810348 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:59:51"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:50.0245947 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:59:54"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:51.6989723 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:59:56"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:53.9011900 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T15:59:58"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:55.0584684 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T15:59:59"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:56.6768783 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T16:00:01"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:58.0524879 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T16:00:02"},"t":"event"} new queue size is 0 on 00212EFFFF010390
17:59:59.3531392 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T16:00:03"},"t":"event"} new queue size is 0 on 00212EFFFF010390
18:00:02.5566910 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T16:00:06"},"t":"event"} new queue size is 0 on 00212EFFFF010390
18:00:11.6791113 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-08-17T16:00:16"},"t":"event"} new queue size is 0 on 00212EFFFF010390
[... cut ...]
18:00:45.4503556 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":99,"lastupdated":"2017-08-17T16:00:49"},"t":"event"} new queue size is 0 on 00212EFFFF010390
18:02:52.6640410 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-08-17T16:02:57"},"t":"event"} new queue size is 0 on 00212EFFFF010390

Apart form the expected values we get 100, 101, 98 and 99. Is this a bug in the deCONZ software?

manup, 2017-08-17:
Should be valid, the received button press events are just forwarded for ZGP devices.

**Meaning**
* 98 Press 1 of 2
* 99 Release 1 of 2
* 100 Press 2 of 2
* 101 Release 2 of 2

The rf module in the Tap has more capabilities than the Philips case and physical buttons are made for, maybe you hit into them.

posixx, 2017-08-17:
I don't understand, what you mean by 'Press 1 of 2' ? I don't press them at once, just right after eachother..

manup, 2017-08-17:
If you pull out the module from the Tap and press two buttons at the same time (top or bottom row) it will send these commands.

Details:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L595

[image]

ebaauw, 2017-08-17:
Cool!  I managed to get 98 and 99 by pressing/releasing 3 and 4 simultaneously, without disassembling the tap.  Let's see how the Hue bridge reacts to that next time I connect the tap to it...

posixx, 2017-08-17:
Well, that's it! I use Gira enOcean switches (i replaced the enOcean module with the hue Tap module) and i can confirm this works perfect.: When simultaneously press the upper flips i get 100 on press, 101 on release. When i press both bottom flips i get 98 and 99!

[image]

closed 2017-09-04 by posixx