dresden-elektronik/deconz-rest-plugin #122: some problems/questions with 2.04.65

dghh, 2017-08-17:
Since update to 2.04.65 "Save Configuration" brings up an error:
"The file /deCONZ.tar.gz couldn't be found."

I thought that i have seen in the last version under "menu" an additional entry. Can´t remember the name.... Now it is missing. Maybe it was an experimental setting/feature ?

Still no sensors are shown in Web App.

The other things are running fine so far.

manup, 2017-08-17:
> Since update to 2.04.65 "Save Configuration" brings up an error:
"The file /deCONZ.tar.gz couldn't be found."

Sorry for that, we messed around with the http server and somehow broke this one, will be fixed in the next version.

> Still no sensors are shown in Web App.

These will be visible in the new web app, we have still to fix a few issues before the private beta will be available.

manup, 2017-08-19:
> Since update to 2.04.65 "Save Configuration" brings up an error:
"The file /deCONZ.tar.gz couldn't be found."

Fixed in version 2.04.66, *save configuration* should work again.

dghh, 2017-08-22:
Thanks. "Save Configuration" works now with 2.04.67

Where can i subscribe to support testing for the private beta of the web app ?

closed 2017-09-02 by dghh

drbytes, 2018-01-04:
2.04.99 : This bug is back.

manup, 2018-01-04:
Which platform are you using? RPi, Ubuntu or Windows?

drbytes, 2018-01-04:
Currently it's running n windows 10.
When it's stable it'll move to ubuntu (my production system). I was hoping that the export would help move things along quickly.

drbytes, 2018-01-04:
I've managed to snagg a backup. 
On the Win10 machine where I had everything configured the 7za.exe was unable to execute. I replaced it with a 7z.exe I knew worked on the machine and I managed to get a xxxxxxxxxxxxxxxxx.dat generated and sent via the browser. I renamed it to tar.gz and I got an archine with 3 files in it.
The Win10 runs on a Minix Z64.