dresden-elektronik/deconz-rest-plugin #124: v2.04.66: Hue dimmer switch config.reachable remains false

ebaauw, 2017-08-19:
Just installed deCONZ v2.04.66 on my Raspberry Pi with RaspBee firmware 0x26130500.

All my six Hue dimmer switches report `config.reachable` as `false`.  They show as normal grey nodes in the deCONZ GUI.  They seem to be fully functional.  When I press a button, the switch becomes reachable for a short while, but then reverts to unreachable.  My other light and sensor resources all show reachable true.

Here's the web socket event log:
```
Sat Aug 19 2017 23:51:24: /sensors/226/config: {"reachable":true}
Sat Aug 19 2017 23:51:24: /sensors/226/state: {"buttonevent":1000,"lastupdated":"2017-08-19T21:51:24"}
Sat Aug 19 2017 23:51:24: /lights/226/state: {"on":true}
Sat Aug 19 2017 23:51:24: /lights/223/state: {"on":true}
Sat Aug 19 2017 23:51:24: /lights/222/state: {"on":true}
Sat Aug 19 2017 23:51:24: /lights/221/state: {"on":true}
Sat Aug 19 2017 23:51:24: /lights/225/state: {"on":true}
Sat Aug 19 2017 23:51:24: /lights/224/state: {"on":true}
Sat Aug 19 2017 23:51:24: /groups/220/state: {"any_on":true}
Sat Aug 19 2017 23:51:24: /groups/20059/state: {"any_on":true}
Sat Aug 19 2017 23:51:24: /sensors/226/state: {"buttonevent":1002,"lastupdated":"2017-08-19T21:51:24"}
Sat Aug 19 2017 23:51:24: /sensors/220/state: {"flag":true,"lastupdated":"2017-08-19T21:51:24"}
Sat Aug 19 2017 23:51:24: /lights/227/state: {"on":true}
Sat Aug 19 2017 23:51:24: /lights/228/state: {"on":true}
Sat Aug 19 2017 23:51:25: /groups/220/scenes/1
Sat Aug 19 2017 23:51:33: /sensors/226/config: {"reachable":false}
```

Here's the `deCONZ --dbg-info=2` log:
```
23:51:24:101 update ZCL value 0xFC00/0x0000 for 0x0017880110xxxxxx after 23 s
23:51:24:101 button 1000 initial press
23:51:24:111 fast rule check 57 (Living Room Dimmer Off press)
23:51:24:116 fast rule check 58 (Living Room Dimmer Off hold 1/2)
23:51:24:122 no button handler for: RWL021 cl: 0x0006 cmd: 0x01 pl[0]: 000
23:51:24:127 fast rule check 59 (Living Room Dimmer Off hold 2/2)
23:51:24:133 fast rule check 60 (Living Room Dimmer On press)
23:51:24:139 fast rule check 61 (Living Room Dimmer On hold)
23:51:24:211 update ZCL value 0xFC00/0x0000 for 0x0017880110xxxxxx after 0 s
23:51:24:211 button 1002 short release
23:51:24:219 fast rule check 57 (Living Room Dimmer Off press)
23:51:24:224 fast rule check 58 (Living Room Dimmer Off hold 1/2)
23:51:24:230 fast rule check 59 (Living Room Dimmer Off hold 2/2)
23:51:24:235 fast rule check 60 (Living Room Dimmer On press)
23:51:24:235 trigger rule 60 - Living Room Dimmer On press
23:51:24:241 fast rule check 61 (Living Room Dimmer On hold)
23:51:24:246 fast rule check 48 (Living Room off)
23:51:24:261 fast rule check 49 (Living Room on, daylight (1/2))
23:51:24:262 Node 0x0017880110yyyyyy is known by 1 neighbors, last seen 24 s
23:51:24:267 fast rule check 50 (Living Room on, daylight (2/2))
23:51:24:273 fast rule check 51 (Living Room on, dark, day (1/3))
23:51:24:273 trigger rule 51 - Living Room on, dark, day (1/3)
23:51:24:280 fast rule check 52 (Living Room on, dark, day (2/3))
23:51:24:286 fast rule check 53 (Living Room on, dark, day (3/3))
23:51:24:291 fast rule check 54 (Living Room on, dark, night (1/3))
23:51:24:297 fast rule check 55 (Living Room on, dark, night (2/3))
23:51:24:304 fast rule check 56 (Living Room on, dark, night (3/3))
23:51:24:309 fast rule check 44 (Living Room Status 3 (1/2))
23:51:24:315 fast rule check 42 (Living Room Status 0)
23:51:24:736 Node 0x0017880102yyyyyy is known by 3 neighbors, last seen 32 s
23:51:25:042 Erase task zclSequenceNumber: 205 send time 0
23:51:25:216 Node 0x0017880102yyyyyy is known by 1 neighbors, last seen 34 s
23:51:25:695 Node 0x000B57FFFEyyyyyy is known by 1 neighbors, last seen 53 s
23:51:26:175 Node 0x001FEE0000yyyyyy is known by 32 neighbors, last seen 100 s
23:51:26:655 Node 0x0017880102yyyyyy is known by 0 neighbors, last seen 225 s
23:51:27:136 Node 0x0017880102yyyyyy is known by 1 neighbors, last seen 222 s
23:51:27:615 Node 0x0017880110yyyyyy is known by 3 neighbors, last seen 295 s
23:51:28:096 Node 0x0017880102yyyyyy is known by 3 neighbors, last seen 105 s
23:51:28:600 Node 0x00158D0000yyyyyy is known by 26 neighbors, last seen 102 s
23:51:29:055 Node 0x0017880100yyyyyy is known by 19 neighbors, last seen 86 s
23:51:29:536 Node 0x0017880102yyyyyy is known by 3 neighbors, last seen 111 s
23:51:30:016 Node 0x0017880110yyyyyy is known by 3 neighbors, last seen 1225 s
23:51:30:496 Node 0x0017880100yyyyyy is known by 17 neighbors, last seen 106 s
23:51:30:975 Node 0x0017880110yyyyyy is known by 1 neighbors, last seen 115 s
23:51:31:455 Node 0x0017880102yyyyyy is known by 2 neighbors, last seen 160 s
23:51:31:936 Node 0x0017880110yyyyyy is known by 20 neighbors, last seen 64 s
23:51:32:416 Node 0x8418260000yyyyyy is known by 9 neighbors, last seen 51 s
23:51:32:895 Node 0x0017880110yyyyyy is known by 2 neighbors, last seen 142 s
23:51:33:376 Node 0x8418260000yyyyyy is known by 7 neighbors, last seen 46 s
23:51:33:475 Node data 0x0017880110yyyyyy profileId: 0x0104, clusterId: 0x0001
23:51:33:475 SensorNode id: 226 (Living Room Dimmer) no longer available
```

snozzlebert, 2017-08-20:
Same here for my dimmer switches. 
My Xiaomi door sensor and button are also had config.reachable: false. They didn't get state updates either in rest api and websocket. Had to reset them and add them again.

ebaauw, 2017-08-20:
I find that if you just briefly press the reset button on a Xiaomi sensor, they report all their attributes (incl. the Basic cluster), after which, deCOZN finds it.  No need for reset/repair.

manup, 2017-08-20:
> I find that if you just briefly press the reset button on a Xiaomi sensor, they report all their attributes (incl. the Basic cluster), after which, deCOZN finds it. No need for reset/repair.

Also if devices don't have buttons, it's recommend to try pull out the battery and put it in again instead of reset. Resets can mess up all sorts of things and should be treated only as last resort.

manup, 2017-08-20:
> All my seven Hue dimmer switches report config.reachable as false. They show as normal grey nodes in the deCONZ GUI. They seem to be fully functional. When I press a button, the switch becomes reachable for a short while, but then reverts to unreachable. My other light and sensor resources all show reachable true.

This should be fixed now in 2.04.67, end devices will only be set to reachable false if there was no communication for 24 hours.

ebaauw, 2017-08-20:
Fixed in v2.04.67.  Thanks!

closed 2017-08-20 by ebaauw