dresden-elektronik/deconz-rest-plugin #125: v2.04.66 breaks light state for (extended) color lights

ebaauw, 2017-08-19:
With v2.04.66, the `state` for (extended) color lights is broken:
- The  `state.ct`, `state.hue`, `state.sat` attributes of are no longer updated when `state.colormode` is `xy` (i.e. when color was set using `xy`).  The corresponding ZCL attributes in the deCONZ GUI are correct, but the REST API plugin seems to ignore them;
- Similarly, `state.ct` and `state.xy` are no longer updated when `state.colormode` is `hs`;
- Similarly, `state.hue`, `state.sat`, and `state`xy` are no longer updated when `state.colormode` is `ct`.

This cause homebridge-hue, and I guess every other application that follows the Hue guidelines for colour conversion between CIE 1931 and RGB, to display the wrong colour, when lights are updated using `ct` or `hue` and `sat` (e.g. from scenes, rules, or from other apps).

manup, 2017-08-20:
Hmm damn, I think this issue as well as the hue dimmer problem has to do with the latest ZCL attribute "improvements". I'm on it, fix should arrive soon.

snozzlebert, 2017-08-20:
I can confirm this. The missing color updates cause a lot of trouble. 
Not only the state isn't updated, also ```modelid``` and ```swversion``` aren't updated in the REST API. I've added two color temperature lights after upgrading to v2.04.66 and ```modelid```is empty in the REST API while deCONZ GUI shows LTW012. And ```swversion``` is ```01000B1A``` in the REST API while deCONZ GUI shows 1.15.2_r19181

Is it possible/save to downgrade to 2.04.65

ebaauw, 2017-08-20:
> Is it possible/save to downgrade to 2.04.65

In my experience: yes.

manup, 2017-08-20:
Version 2.04.67 is online, it should fix the issues of 2.04.66

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.67-qt5.deb

snozzlebert, 2017-08-20:
It does. Thanks.

ebaauw, 2017-08-20:
Indeed.  Thanks!

closed 2017-08-20 by ebaauw