dresden-elektronik/deconz-rest-plugin #129: Support for Bi-directional Green Power Commissioning

shayanhaider, 2017-08-21:
is there any support planned for bi-directional GP devices in the upcoming release?

manup, 2017-08-22:
There are no plans for that, according to my knowledge there are no bi-directional GP end-devices on the market. Or do you mean commissioning GP proxy lights?

shayanhaider, 2017-08-22:
There are no productized bi-directional GP end-devices in the market. I wanted to commission one of the development sample device. Are you planning to enable commissioning of GP proxy lights?. As per my understanding GP devices like Hue Tap currently communicates directly with the deconz gateway without the involvement of any proxy nodes. In case the GP device is out-of-range of
the gateway, proxy node functionality in lights will be required to relay messages.

manup, 2017-08-22:
>There are no productized bi-directional GP end-devices in the market. I wanted to commission one of the development sample device.

I'm afraid official support will be implemented after devices appear on market or we develop some on our own.

> Are you planning to enable commissioning of GP proxy lights?. As per my understanding GP devices like Hue Tap currently communicates directly with the deconz gateway without the involvement of any proxy nodes. In case the GP device is out-of-range of
the gateway, proxy node functionality in lights will be required to relay messages.

Indeed, this will happen sooner, there was some activity on the topic recently: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/117

I think the proxy support can be implemented with little afford, there is no ETA yet but expect it in the next months.

closed 2017-08-22 by shayanhaider