dresden-elektronik/deconz-rest-plugin #130: Suddenly deConz forgot name of Dim Switch

donnib, 2017-08-23:
So i am not sure the problem is related to same thing as this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/112 but my Hue Dim Switches (i have two) decided to forget its their so it went back to its default name it had when it was joined. Idk if this is related somehow that maybe it rejoined and got reset ?

manup, 2017-08-23:
The name in deCONZ nodeview or the name returned in the REST API?

The former is a known problem for various devices and will be fixed soon.

donnib, 2017-08-23:
both went back to Switch X

donnib, 2017-08-23:
I noticed it when Home Assistant were not reacting to the events until i see the names has changed.

manup, 2017-08-23:
> both went back to Switch X

That's not possible (or I don't know how by looking at the code), sensor names are *only* assigned when they are added during setup &mdash; while sensor search is active otherwise just the names previously stored in the database are restored but not modified.

The only thing I can imagine is that the sd-card somehow didn't have the latest database stored correctly or database was corrupted and a previous one was restored, we've seen such issues when power was cut hard (no proper reboot/ shutdown) and with older low quality sd-cards.

manup, 2017-09-10:
Sensor and light names should work with 2.04.75, see commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/186646a5e91864edb14f508d175958fbef512717

closed 2017-09-10 by manup