dresden-elektronik/deconz-rest-plugin #131: deConz stopped working

donnib, 2017-08-23:
Hi,
After some time where everything was working ok then deConz stopped working, in the sense all lights/sensors where not reacting. In webui i could turn on/off lights and nothing happened. A restart of deConz or reboot of rPI did not help. Only a shutdown completely and take the power form the unit then everything went back to normal.

manup, 2017-08-23:
When the shield has gone stuck, most likely queues were filled up.

You can restart the shield via `$ sudo GCFFlasher -r`, deCONZ must be closed before.
But again something in this setup seems to be messed up, currently I have no idea what, maybe the new firmware will help.

donnib, 2017-08-23:
Well it could very well be the IKEA lights which i have so many of or only :(

manup, 2017-08-24:
Version 2.04.68 has also a improvement to prevent the above problem, let's see if it works out.