dresden-elektronik/deconz-rest-plugin #133: Removing and later re-adding a Philips motion sensor doen no longer show in RESP-API

wvuyk, 2017-08-23:
I am on 2.0.04.67 here. I deleted a Philips Motion sensor with the DELETE /api/<apikey>/sensors/<id> commands, for each of the sensors. 
I later decided to re-add the sensor with a search  (/api/<apikey>/sensors).
The sensor shows in the deCONZ interface and seems to communicate, but in the REST API the sensors are never reported again. Am I missing something maybe?

Thanks,

Wim

manup, 2017-08-23:
Yes while the search *is running* you have to *reset the sensor*.
For that press setup button for about 10 seconds, it should reset and rejoin the network.

wvuyk, 2017-08-23:
@manup I have done exactly that, it is showing in deCONZ, but the REST API does not show the sensor anymore. I have been resetting the sensor about 10 times by now, restarted the RaspBee several times. It shows in the deCONZm interface, not in the REST API?

manup, 2017-08-23:
Don't need to restart RaspBee.

* does the Sensor LED blink after you reset it?
* does it blink blue in deCONZ GUI sometimes (means something is received)?
* what's the firmware version of the RaspBee (Help > About deCONZ > Firmware version)?
* what's your REST request and response to search for sensors? Alternativ you can open the network in the web app.

I've now testet it here: delete, reset, then rejoin and the sensor was visible in the REST API very quickly again.

wvuyk, 2017-08-23:
@manup 

Yes the sensor does blink after the reset, I am movng the sensor back and forth between the RaspBee and a Philips Hue bridge. The same procedure I am doing works fine for a DimmeSwith here.
It does blink in the deCONZ inteface. After a reset I do see the sensor in  deCOnz quite often for a while.
The firmware version was upgraded after the upgrade to 2.0.04.67. It is now 0x26130500?
The REST request is like I posted above (IP RaspBee\api\user\sensors) and the response is
 [{"success":{"/sensors":"Searching for new devices","/sensors/duration":60}}]

[image]

manup, 2017-08-23:
>Yes the sensor does blink after the reset, I am movng the sensor back and forth between the RaspBee and a Philips Hue bridge.

That's very tricky, devices remember all sorts of things from their neighbors, short addresses, security counters, routing tables, etc., they also tend to be sensitive to changes. After each reset these numbers will change and may mess up internal state of devices for some time. It learned that the hard way :)

Resets really **shouldn't** be treated as regular functions more as sparingly used last resort functions. For lights it's not so much a problem, but sleeping end devices are quite complicated to handle.

>The same procedure I am doing works fine for a DimmeSwith here.
>It does blink in the deCONZ inteface. After a reset I do see the sensor in deCOnz quite often for a while.

>The firmware version was upgraded after the upgrade to 2.0.04.67. It is now 0x26130500?
Yes that looks good it's the latest version.

The REST request is like I posted above (IP RaspBee\api\user\sensors) and the response is
[{"success":{"/sensors":"Searching for new devices","/sensors/duration":60}}]

Looks good too, since resetting was done a lot, counters may be messed up, please try the following before next sensor reset.

* open `deCONZ > Edit > Network Settings`
* raise `NWK Update ID`  by (+1)
* chose another than the active channel: should be one of 11, 15, 20, 25
* click *Save* and then *Done*
* click *Leave* and after ~3 seconds click *Join*

Now open the network again and try the sensor reset, you may also chose to open the network for more than 60 seconds, click `advanced settings` in the webapp and there select 10 minutes and click `Apply`. The next time sensor search is done it will be active for 10 minutes.

wvuyk, 2017-08-23:
@manup 
I tried these step and now the sensor adds wel.But the other motion sensors (4) are no longer accessable?
I do not think the lights and other motion sesnors made the move?
[image]

manup, 2017-08-23:
Sorry forgot to say: the lights should be power cycled, so that they move to the new channel too. I figure the sensors will follow them shortly.

wvuyk, 2017-08-23:
@manup 
Was starting to run to the lights here, but while I finding a way to power cycle them, things started to work one by one, also lights that did not power cycle. It was just taking some time indeed... not somethng to do when it is dark!

Reason I am trying this is because users of my plugin might want to do just this, move sensors to the RaspBee or ConBee. Testing made me do this several times, something a standard user will not do often I hope . 
Thanks for your help,  I will close this one...

closed 2017-08-23 by wvuyk

manup, 2017-08-23:
Glad it worked. For non developer use cases it should work hopefully :)