dresden-elektronik/deconz-rest-plugin #135: Question about ConBee on Windows

wvuyk, 2017-08-24:
I have a question about the ConBee on Windows. 
Will there be an (beta?) version of deCONZ for the ConBee available sometime soon? Or is there a way I could do this myself? 

Thanks,

Wim

manup, 2017-08-24:
Currently the beta versions are only public for Raspberry Pi and Ubuntu, I can provide the windows version as well, it might take a few days though.

wvuyk, 2017-08-24:
It would be nice, so I could test with some of my users. I think the majority of my users are using Windows for their HomeSeer system (it runs on Linux, Raspberry Pi and Macs too). I guess they want to combine this on the same system.

Thanks in advance!

Wim

wvuyk, 2017-09-04:
@manup Any idea when you have a windows version?

manup, 2017-09-04:
@wvuyk yes sorry, took a while. Here is the current version:
It also includes firmware version 0x26190500 for ConBee.

https://www.dresden-elektronik.de/deconz/win/deCONZ_Setup_Win32_V2_04_74.exe

wvuyk, 2017-09-04:
@manup Thanks! We'll have a go with it.

Wim

wvuyk, 2017-09-04:
@manup I have installed the update over the old 2.0.4.35 version. This was configured as a router to my raspbee gateway(tested working before the update). After the update (including firmware), the ConBee was no longer joined to the gateway. And whatever I try, joining the network as a router does not work anymore for me. I tried on deCONZ level, on WebApp level, it will not work this way. 

Am I missing something?

Wim

wvuyk, 2017-09-05:
I am still tryig an playing with this. When I join the network, the ConBee is added to the network of the router. It stays red though instead of becoming yelow as a router. The conbee Windows does not recognise this it seems and gives up connecting?

manup, 2017-09-05:
The problem is in the newer firmware, due all the added IKEA related updates the functionality to join as router doesn't work anymore. I will check it.. however you can use older ConBee firmware which should work as router.

wvuyk, 2017-09-05:
I understand. Did not check the coordinator here yet, but some of the users of my plugin will try this soon. Will wait for them for now