dresden-elektronik/deconz-rest-plugin #136: 2.4.68 Ikea TRADFRI dimmer not shown

vamanea, 2017-08-24:
I've tried to use a TRADFRI dimmer(as seen here: http://www.ikea.com/us/en/catalog/products/00349934/ ) . It seems to join the network just fine, I can see it in the UI(0x1276) but it does not show up in the web interface. Do you have any suggestions how to get it working? All the other IKEA TRADFRI seem to work just fine.

manup, 2017-08-24:
The dimmer is not integrated yet, but it will be available soon, so the lights can be added to its group via web app.

Hector47, 2017-08-24:
For me, fowling the same process as adding en Philips Dimmer switch (adding it in deconz then paring it with one light) created the needed group in the web app (14575 and 22940):
[image]

emiglio, 2017-08-28:
Hi. When will the Dimmer be available as a group? Really looking forward to that feature. 
Great work by the way!!

manup, 2017-09-10:
The IKEA TRADFRI dimmer has now experimental support in deCONZ version 2.04.76.

* supports button events through websocket
* can be used in rules
* to pair lights to the dimmer, hold the dimmer close to the light and press the pairing button for > 10s
* **Note** adding lights to the dimmer group will likely not work, for some reason the dimmer unicasts to the lights and doesn't use group casts like the IKEA remote control

See https://github.com/dresden-elektronik/deconz-rest-plugin/commit/0fe844cb35343013ed3a01d6b644031f66aedb1b

mariusmotea, 2017-09-10:
How i can install this release? When i try to compile the master branch i get the following error:

```
pi@raspberrypi:~/deconz-rest-plugin $ qmake QMAKE_CFLAGS_ISYSTEM=
pi@raspberrypi:~/deconz-rest-plugin $ make -j2
make -f Makefile.Release
make[1]: Entering directory '/home/pi/deconz-rest-plugin'
g++ -c -pipe -Wno-attributes -Wall -Wno-attributes -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DARCH_ARM -DARCH_ARMV7 -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.04.76\" -DGW_API_VERSION=\"1.0.3\" -DGIT_COMMMIT=\"f1a15d0521d1efd2fc4a7d6b44ca033fcd2f6f8e\" -DGW_MIN_RPI_FW_VERSION=0x26190500 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"deCONZ-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I/usr/include -I../.. -I../../common -I/usr/include/arm-linux-gnueabihf/qt5 -I/usr/include/arm-linux-gnueabihf/qt5/QtWidgets -I/usr/include/arm-linux-gnueabihf/qt5/QtGui -I/usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -I/usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -I/usr/include/arm-linux-gnueabihf/qt5/QtNetwork -I/usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/authentification.o authentification.cpp
g++ -c -pipe -Wno-attributes -Wall -Wno-attributes -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DARCH_ARM -DARCH_ARMV7 -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.04.76\" -DGW_API_VERSION=\"1.0.3\" -DGIT_COMMMIT=\"f1a15d0521d1efd2fc4a7d6b44ca033fcd2f6f8e\" -DGW_MIN_RPI_FW_VERSION=0x26190500 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"deCONZ-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I/usr/include -I../.. -I../../common -I/usr/include/arm-linux-gnueabihf/qt5 -I/usr/include/arm-linux-gnueabihf/qt5/QtWidgets -I/usr/include/arm-linux-gnueabihf/qt5/QtGui -I/usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -I/usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -I/usr/include/arm-linux-gnueabihf/qt5/QtNetwork -I/usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/atmel_wsndemo_sensor.o atmel_wsndemo_sensor.cpp
authentification.cpp: In member function �bool DeRestPluginPrivate::allowedToCreateApikey(const ApiRequest&, ApiResponse&, QVariantMap&)�:
authentification.cpp:93:49: error: �ParamSecurityMaterial0� is not a member of �deCONZ�
         QByteArray sec0 = apsCtrl->getParameter(deCONZ::ParamSecurityMaterial0);
                                                 ^~~~~~
Makefile.Release:411: recipe for target 'release/authentification.o' failed
make[1]: *** [release/authentification.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pi/deconz-rest-plugin'
Makefile:38: recipe for target 'release' failed
make: *** [release] Error 2

```

manup, 2017-09-10:
Forgot to mention, you need to update the dev package as well

http://www.dresden-elektronik.de/rpi/deconz-dev/deconz-dev-2.04.76.deb

mariusmotea, 2017-09-10:
Thanks, is working nice. Any idea why buttonevent 1000 is when you rotate fast in both directions? I believe it must turn off the light if rotated to left and set max bri when rotated to right, but with current values is not possible to detect if there was a rotation made to turn off the light or to obtain maximum brightness fast. Currently i use  buttonevent 1000 to turn off the light when bri is 1 and ignore in other cases. This is working ok but if i rotate the button fast to obtain maximum brightness it will not work, so i need to stop the rotation to not send buttonevent 1000 anymore and then rotate in same direction slowly.

manup, 2017-09-10:
Yep the events are very rough and quite often seem to be not logical,
but that's what the dimmer is sending, not sure yet if the API can improve here.

mariusmotea, 2017-09-10:
I believe i will leave event 1000 turn on the light if is off and vice versa. This will be strange if light is on and you want more brightness fast because will turn off the light while rotating fast to the right.

Thanks again for fixing this.

donnib, 2017-09-11:
Is the Dimmer any better than the IKEA Remote ? I find the IKEA Remote to be crap in terms of stability as mentioned many times. I actually spoke to ikea technical department and they are well aware of that fact :(

manup, 2017-09-11:
Depends, when used with just a few lights, it works quite well, but I figure when Pairing it with more than 5 lights it will go crazy :)

In my setup the IKEA remote works very reliable but since the mix of lights is very different from your large IKEA network the problem might not be the remote itself but somethere on the path in the mesh network. Hard to tell without sniffing the issue.

mariusmotea, 2017-09-11:
I don't have issues with IKEA Remotes, for me is working as expected, but the Dimmer has the issue that i already spoke. If Dimmer had different codes when is rotated fast then it will be more practical.

donnib, 2017-09-11:
@mariusmotea merci, eu am numai probleme

I only have issues with them, they work when they want, i have 47 IKEA nodes and out of that 6 remotes and they don't always send events, i was told by ikea that the ones in the shop now should be little better but i haven't bought one to try.

ebaauw, 2017-09-11:
> In my setup the IKEA remote works very reliable but since the mix of lights is very different from your large IKEA network the problem might not be the remote itself but somethere on the path in the mesh network. Hard to tell without sniffing the issue.

I have a feeling sometimes it simply doesn't send anything, even though it feels like you pressed a button.

> the remotes in the stores now should be a bit improved

Because of the new firmware or because of improved hardware?  We did see some changes with the  new firmware (releasing the Previous or Next button now sends the custom command multiple times).

> New batteries should make the problem go away even though the batteries are not really empty.

Interesting.  So better replace the battery after the firmware upgrade, even though it's still at 47%?

donnib, 2017-09-11:
> Depends, when used with just a few lights, it works quite well, but I figure when pairing it with more than 5 lights it will go crazy :)
>
> In my setup the IKEA remote works very reliable but since [...]

So here is what i do, i pair the remote with deConz then i create a manual group in deConz then i use my HASS component and act ona event from the remote to turn on the deConz group so essentially i don't pair the remote with anything else than just deConz.

It's confirmed by the source from IKEA that the remote just doesn't send the Zigbee packets so i would guess that would be easily seen in a sniffer. What my source tells me is that i should go to the store and replace it since the one in the stores now are improoved however i have not been able to get an answer if i can in some reliable way see if it's the new HW or the old one. My remote says HW version 1 in deConz. @manup @mariusmotea does yours also state the same version ?

manup, 2017-09-11:
[image]

HW version 1, I'm still on the old firmware the remote is a few months old now.

donnib, 2017-09-12:
@manup any place to get hold of an Conbee used e.g cheaper ? I would like to sniff this issue to understand better what's going on.

wvuyk, 2017-09-18:
Wow, this Ikea dimmer sends a load of updates per second? I am counting up to 8 value switches 1001/1003's when turning anti clockwise.  I am wondering how usefull that is?

mariusmotea, 2017-09-18:
Not sure if is the same case, but before to update the firmware with Tradfri Gateway and to replace the battery, it was sending more requests, especially when it was slowly rotated near the "triggering" point.

wvuyk, 2017-09-19:
Thanks, will have the user check his firmware here. Hoping the update will give an easier controllable effect.