dresden-elektronik/deconz-rest-plugin #142: 3rd party app only find 4 devices

Krocko, 2017-08-25:
Until now i have 6 devices in the Raspbee network. 4 Philips Hue bulbs and 2 Osram Plugs. I can control this devices with the deconz web app. But the 3rd party app (Amazon Alexa) only finds the 4 bulbs and not the 2 plugs. Is there a solution for this?

manup, 2017-08-25:
I guess the Alexa app may not support non dimmable on/off only lights yet.

Krocko, 2017-08-25:
If i connect this plugs with the Hue bridge, Alexa find this devices.

manup, 2017-08-25:
Then it would be interesting if the device type of the plug is different for deCONZ compared to the Hue API. Do you use the Philips hue Alexa skill or build-in Alexa hue support?

manup, 2017-08-25:
deCONZ only supports the build-in Alexa Hue functionality, it can't be used with the Philips Hue skill. Can you provide the Hue bridge REST API response of `/lights/<id>` for the OSRAM plug?

Krocko, 2017-08-25:
```
{
  "etag": "e791a5e8891b998ca5c40e91b6651500",
  "hascolor": false,
  "manufacturername": "OSRAM",
  "modelid": "Plug 01",
  "name": "Abzug",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": "V1.04.12",
  "type": "On/Off plug-in unit",
  "uniqueid": "84:18:26:00:00:0b:60:a9-03"
}
```
```
{
  "etag": "6b32792840d5ab7e0b624f54d50c96fc",
  "hascolor": false,
  "manufacturername": "OSRAM",
  "modelid": "Plug 01",
  "name": "Mikrowelle",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": "V1.04.12",
  "type": "On/Off plug-in unit",
  "uniqueid": "7c:b0:3e:aa:00:a3:90:aa-03"
}
```

ebaauw, 2017-08-25:
> Then it would be interesting if the device type of the plug is different for deCONZ compared to the Hue API.

Not for my OSRAM plug, it's an `On/Off plug-in unit` on both.  Mine does have a `modelid` of `Plug - LIGHTIFY` and an older `swversion`of `V1.03.21`.

manup, 2017-08-25:
If they both are the same, it seems that the build-in Hue function in Amazon Alexa may not support it, but only the Philips Hue skill?

Krocko, 2017-08-26:
Output from the Hue Bridge:
```
"16": {
    "state": {
      "on": false,
      "alert": "select",
      "reachable": true
    },
    "swupdate": {
      "state": "notupdatable",
      "lastinstall": null
    },
    "type": "On/Off plug-in unit",
    "name": "Abzug",
    "modelid": "Plug 01",
    "manufacturername": "OSRAM",
    "uniqueid": "84:18:26:00:00:0b:60:a9-03",
    "swversion": "V1.04.12"
  },
  "17": {
    "state": {
      "on": false,
      "alert": "none",
      "reachable": true
    },
    "swupdate": {
      "state": "notupdatable",
      "lastinstall": null
    },
    "type": "On/Off plug-in unit",
    "name": "Mikrowelle",
    "modelid": "Plug 01",
    "manufacturername": "OSRAM",
    "uniqueid": "7c:b0:3e:aa:00:a3:90:aa-03",
    "swversion": "V1.04.12"
  }
```

Krocko, 2017-08-26:
@manup: any chance to get the Osram Plug working with Alexa?

manup, 2017-08-26:
Maybe, I have an idea but can test only next week when I have access to an echo dot.

Krocko, 2017-08-26:
Thank you. Would be really great.

Krocko, 2017-08-31:
Good Morning @manup. Have you any news?

Krocko, 2017-09-03:
> Maybe, I have an idea but can test only next week when I have access to an echo dot.

Anything i can test for you?

manup, 2017-09-03:
Hey Krocko, sorry for the late answer I have the dot now but haven't had time to test due some other bugs which need to be fixed first :/

The idea is to present the OSRAM plug as dimmable light to Echo so that the build-in hue client of Alexa does support the device.

Krocko, 2017-09-03:
No Problem.
Me and my family use the Alexa voice assistant extensivly. 
This is the only reason i have not disposed the Hue bridge by now.

Krocko, 2017-09-24:
Hello @manup. Hand you Time to work on this?

manup, 2017-09-24:
Hi @Krocko  just came back from one week vacation and worked on few reliability issues, so no news on the OSRAM plug so far. Meanwhile mine stopped working for unknown reasons, I'll ordered the new revision and will try to get it to work with Amazon Echo when it arrives (weekend projects).

Stay tuned :)

Krocko, 2017-09-25:
Ok. Thanks in advance.

dotmaster, 2017-11-20:
Hi @Krocko,
can you pint me to how you made Alexa discover the devices? I have been scanning my network with Echo Dot (v2), but it doesn´t find anything.
Homebridge-hue is working fine instead.

Krocko, 2017-11-24:
@manup with deCONZ 2.04.89 all worked perfect. Today i installed 2.04.91 and all devices are offline and Alexa don‘t find an device anymore. I‘m back to 2.04.89 and all worked again. Anything you brake in 2.04.91?

dotmaster, 2017-11-24:
@krocko Ok I did that without luck.
Probably the problem you describe.
@manup AFAIK Alexa does discover by SSDP 1900. Is this implemented in deconZ or can you explain me how the discovery should work.

Krocko, 2017-11-24:
@dotmaster can you test it with deCONZ 2.04.89?
I think the ssdp changes in 2.04.90 will brake something.

Krocko, 2017-11-25:
@manup: this commit 
https://github.com/dresden-elektronik/deconz-rest-plugin/commit/690a70e8332b121a8027620bf9cc753bf99f8cdd
brakes the Alexa function completly. I have compiled latest commits and changed the upnp.ccp with this from 2.04.89 and Alexa worked.

manup, 2017-11-25:
>@manup: the commit 690a70e brakes the Alexa function completly. I have compiled latest commits and changed the upnp.ccp with this from 2.04.89 and Alexa worked.

That commit seems to be ok but https://github.com/dresden-elektronik/deconz-rest-plugin/commit/91fe6b5bacae25ce1c0976c4d28353ed50c3c3cd might be the reason for the issues, I'm looking into it.

manup, 2017-11-25:
Did some testing looks like the problem was: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/288

Please try again with 2.04.93 which will be uploaded later today.

Krocko, 2017-11-25:
You are right. I tested with latest commits and it is working.

dotmaster, 2017-11-26:
@Krocko Hi, I had 2.04.89 running. I don´t knwo exxactly what was the issue, probably I had the dev version of the rest api runnign. I reinstalled deconz and now it works. Thanks!

closed 2017-11-26 by Krocko

ooii, 2018-08-27:
Hi all,
I cannot manage to find any of my devices with Alexa. Should I enable the Philips Hue Skill or another one? 
Thanks.

minscof, 2019-08-04:
Same problem here, could you give more details for the procedure. I didn't install the hue skill, I activated the "authenticate app" button on the Phoscon app, then I asked alexa to discover new equipments, but alexa didn't find them.

I have an Alexa Echo not an Echo Plus, is it why it didn't work ?

ooii, 2019-08-04:
How many device belong to your hue bridge?