dresden-elektronik/deconz-rest-plugin #143: New /sensors resources created for known Hue dimmer switches

ebaauw, 2017-08-26:
When I open the network after restarting deCONZ, before all the sensors have been discovered, the REST API creates new a sensor resource when it discovers a Hue dimmer switch.  Also, when I soft reset the Hue dimmer switch (pressing all four buttons) while the network is open, a new sensor resource is created.  This only happens for the Hue dimmer switch.

After exiting deCONZ, the old resource is still in the database.  If I just delete the newly created resource from the database and start deCONZ, the old resource is fully functional again.

It would seem that the new resource is created too early, before the Hue dimmer has been fully read.  Consequently the check whether it already exists fails.  Looking at the database, the `swversion` of the newly created resource is empty and the `fingerprint` differs from the existing sensor.  It does have the same `uniqueid`.  
The `fingerprint` is `{"d":2096,"ep":1,"in":[0,1,64512],"out":[6,8,5],"p":49246}` for the existing sensor resources and `{"d":2096,"ep":2,"in":[0],"out":[6,8,5],"p":49246}` for the newly created resources.

manup, 2017-08-26:
I've found the related spot in code I need to test and think more for a generic solution. It seems that the problem is that not all endpoints are fetched/restored from cache at the time there the unwanted double is created.

snozzlebert, 2017-09-10:
I tried to add a new sensor (Xiaomi Cube) and opened the network (2.04.75). After that, my Hue dimmer switches don't work anymore.
For my two dimmer switches new entries were added to the database/rest api. Don't know if it is related to restarting deCONZ, because it was already running for a while.

The name of the original entry was changed to Dimmer Switch:
```
    "1": {
        "config": {
            "battery": 100,
            "group": "6107",
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "d1fd9284dcc794d06007dd54146ab7aa",
        "manufacturername": "Philips",
        "mode": 1,
        "modelid": "RWL021",
        "name": "Dimmer Switch",
        "state": {
            "buttonevent": 4002,
            "lastupdated": "2017-09-09T12:02:59"
        },
        "swversion": "5.45.1.16265",
        "type": "ZHASwitch",
        "uniqueid": "00:17:88:01:10:xx:xx:26-02-fc00"
    },
```
Fingerprint: ```{"d":2080,"ep":1,"in":[0,1,64512],"out":[6,8],"p":49246}```

The new entry has no group and a different ```ep``` (2 instead of 1):
```
"11": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true
        },
        "ep": 2,
        "etag": "d1fd9284dcc794d06007dd54146ab7daa",
        "manufacturername": "Philips",
        "mode": 1,
        "modelid": "RWL021",
        "name": "Dimmer Switch",
        "state": {
            "buttonevent": 0,
            "lastupdated": "1969-12-31T23:00:00"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:17:88:01:10:xx:xx:26-02-fc00"
    }
```

Fingerprint: ```{"d":2080,"ep":2,"in":[0,1,64512],"out":[6,8],"p":49246}```

The original entries (the one with ```"ep": 1```) don't get updates, so my rules linked to these sensor don't fire.

Don't know wat ```ep``` exactly means, is this related to ZLL (1) and ZHA (2)? Did the change in #137 somehow result in a fingerprint with ep 2?

For one dimmer switch I removed the original entry and for one the new entry from the database (not on purpose). I had to change some rules to the new id of one of the dimmer switches and everything worked again. 
After opening the network again, a new entry was added for one dimmer switch (the original one with ep 1). This is really annoying. Should I switch to the entries with ```"ep": 2``` instead of ```"ep": 1```, because they won't be recreated when I open the network? Is there a difference between 1 and 2?

ebaauw, 2017-09-10:
If you delete the newly created resources (simply through the REST API) and restart deCONZ, the Hue dimmer switches should work as before.

manup, 2017-09-10:
I think the problem is related to changed fingerprint detection for the hue dimmer. So in this case there shouldn't be created a new dimmer but just the fingerprint needs to be updated.

Hope to fix this in the next days

ebaauw, 2017-10-08:
I think it has been solved.

closed 2017-10-08 by ebaauw

donnib, 2017-11-17:
I just noticed this today that for one of my HUE Switches i have 2 entries in the rest api, should i manually do something ? I am running 2.04.89

manup, 2017-11-17:
Can you please provide the REST output of these two?

A sensor should only be created if not already existing in the API, so somehow the first one didn't prevent that happening.

donnib, 2017-11-17:
Here it is :

````
{
    "config": {
        "battery": 100,
        "group": "",
        "on": true,
        "reachable": true
    },
    "ep": 2,
    "etag": "88c499255ecd16f19013dbbd944008fb",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "Stue knaptryk",
    "state": {
        "buttonevent": 4002,
        "lastupdated": "2017-11-16T20:43:53"
    },
    "swversion": "5.45.1.17846",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:02:e2:68:15-02-fc00"
}
````
and 
````
{
    "config": {
        "battery": 100,
        "group": "9482",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "88c499255ecd16f19013dbbd944008fb",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "Stue knaptryk",
    "state": {
        "buttonevent": 4002,
        "lastupdated": "2017-08-20T18:59:23"
    },
    "swversion": "5.45.1.17846",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:02:e2:68:15-02-fc00"
}
````

manup, 2017-11-17:
Very strange they even have the same unique id, I'll add a simple check to prevent that.

Till then reloading deCONZ should be enough to get rid of the second one, since it should check to only load the sensor if not already present.

snozzlebert, 2017-11-17:
Probably a different fingerprint (ep 1 vs ep 2) in the sensors table of the sqlite database.

manup, 2017-11-17:
Exactly, there was a change to fingerprint endpoint a while back, the last commit should fix the issue.