dresden-elektronik/deconz-rest-plugin #144: problem with schedules

dghh, 2017-08-26:
I´ve tried to work with sensors, schedules & lights.
I´ve created, based upon motion, a schedule to switch on the light for 60 sec. It works as expected. 
Created schedule:
"163":{"autodelete":true,"command":{"address":"/api/1BCB984DEE/groups/8/action","body":{"on":false},"method":"PUT"},"description":"group_8_scene_4","etag":"aa4ef5300e91ad22e8d8708956d05dbc","name":"group_8_scene_4 163","starttime":"2017-08-26T08:34:37","status":"enabled","time":"PT00:00:60"}

Within this 60 sec, a new motion is detected, triggering the same schedule again.
Created schedule:
"164":{"autodelete":true,"command":{"address":"/api/1BCB984DEE/groups/8/action","body":{"on":false},"method":"PUT"},"description":"group_8_scene_4","etag":"0bf8b03706b3fb883d84b61159afb308","name":"group_8_scene_4 164","starttime":"2017-08-26T08:34:55","status":"enabled","time":"PT00:00:60"}

1. I expected that the second triggering of the motion sensor only should extend (or renew the 60 sec) the time of the first schedule, but a new schedule (with the number of the schedule added to the name) is triggered.
2. When the first schedule expires the light goes to off and the second schedule has no effect anymore.
3. How can i handle this "extend" for another 60 sec when again motion within the first 60 sec is detected?

I´ve thought to delete the first schedule and setup a new one ? But without switching the light off ? Would that work ?

Deleting / Identifying a specific schedule is another problem. They are not numbered sequentially. They didn´t start with "1" and then the next is "2". and so on. As you can see they start somewhere....
With this it is difficult to search for a specific schedule with a specific name. (no "for-next loop" possible which starts at 0/1 and ends with the max number of current schedules)

ebaauw, 2017-08-26:
Don't create schedules based on sensors events.  Use rules instead.  For example, to turn off a light, 1 minute after motion was last detected, use the following conditions:
```
"conditions": [
  {
    "address": "/sensors/xx/state/presence",
    "operator": "eq",
    "value": "true"
  },
  {
    "address": "/sensors/xx/state/lastupdated",
    "operator": "ddx"
    "value": "PT00:01:00"
  }
]
```

dghh, 2017-08-26:
Thanks, but i´ve never worked so far with rules and i think that schedules are more flexible for my setup.
Nevertheless i will try it with your proposal, but i also need to have a solution for schedules and beside this need to know why schedules behave so unexpected ? Or is it a bug ?

Hector47, 2017-08-27:
@ebaauw Do you know where I can find more exemples of rules? The documentation is very limited.
Thanks

Krocko, 2017-08-27:
@ebaauw: How can i add a condition that the light only goes on if the light level is < 1,2 lux?

```
"conditions": [
  {
    "address": "/sensors/xx/state/presence",
    "operator": "eq",
    "value": "true"
  },
  {
    "address": "/sensors/xx/state/lastupdated",
    "operator": "ddx",
    "value": "PT00:01:00"
  }
]
```

And how i add a light to this rule?

ebaauw, 2017-08-27:
> Do you know where I can find more exemples of rules? The documentation is very limited.
Thanks

You might search through the issues [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues?utf8=✓&q=is%3Aissue%20rule%20operator) or with [homebridge-hue](https://github.com/ebaauw/homebridge-hue/issues?utf8=✓&q=is%3Aissue%20rule%20operator).  Also checkout the [Hue developers forum](https://developers.meethue.com/forum).

> How can i add a condition that the light only goes on if the light level is < 1,2 lux?

Something like this (`/sensors/xx` is the ZHAPresence sensor, `/sensors/yy` the ZHALightLevel):
```json
{
  "name": "My favourite rule",
  "conditions": [
    {
      "address": "/sensors/xx/state/presence",
      "operator": "eq",
      "value": "true"
    },
    {
      "address": "/sensors/xx/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/sensors/yy/state/lux",
      "operator": "lt",
      "value": "2"
    }
  ],
  "actions": [
    {
      "address": "/lights/zz/state",
      "method": "PUT",
      "body": {
        "on": true
      }
    }
  ]
}
```
`state.lux` only supports integer values - you might want to use `state.lightlevel` instead.  Personally, I prefer `state.dark` and `state.daylight` with the right calibration through `config.tholddark` and `config.tholdoffset` (see the [Hue API documentation](https://developers.meethue.com/documentation/supported-sensors) - scroll down to 2.7).

Krocko, 2017-08-27:
Thank you very much. With this i can work. 
Can i add a Time based condition as well? E.g 7:00 - 20:00?

ebaauw, 2017-08-27:
Use a condition on `/config/localtime`, see: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/98#issuecomment-324904741.  Note that this describes how deCONZ should trigger rules, not how it does so today.

Krocko, 2017-09-06:
@ebaauw can i add a condition to this rule, that the light goes of after 2 minutes no motion is detected, or must i create a new rule?
```
{
  "name": "My favourite rule",
  "conditions": [
    {
      "address": "/sensors/xx/state/presence",
      "operator": "eq",
      "value": "true"
    },
    {
      "address": "/sensors/xx/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/sensors/yy/state/lux",
      "operator": "lt",
      "value": "2"
    }
  ],
  "actions": [
    {
      "address": "/lights/zz/state",
      "method": "PUT",
      "body": {
        "on": true
      }
    }
  ]
}
```

ebaauw, 2017-09-06:
You must create a new rule, with different conditions (presence eq false, lastupdated ddx T00:02:00) and a different body in the action `{"on": false}`

Krocko, 2017-09-06:
Ok. I have done this. But if i switch on the Light with the App, the Light goes of automatically also.

ebaauw, 2017-09-06:
Then you need more complex rules, using a CLIPGenericFlag or CLIPGenericStatus sensor to record whether the light was switched on by the motion sensor.  You would add an action set this flag in the first rule and add a condition on this flag to the second rule, only to turn off the light automatically when is was turned on by the motion sensor, and an action to clear the flag.

Krocko, 2017-09-06:
Can you please give me an example? This is really complicated. My rules looks like this
```
{
  "10": {
    "actions": [
      {
        "address": "/groups/6/scenes/1/recall",
        "body": {
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/config/localtime",
        "operator": "in",
        "value": "T07:00:00/T20:00:00"
      },
      {
        "address": "/sensors/4/state/lightlevel",
        "operator": "lt",
        "value": "3000"
      },
      {
        "address": "/sensors/5/state/presence",
        "operator": "dx"
      },
      {
        "address": "/sensors/5/state/presence",
        "operator": "eq",
        "value": "true"
      }
    ],
    "created": "2017-09-06T18:38:24",
    "etag": "906ae2b1bbf05f333e46e2fea18b7d38",
    "lasttriggered": "none",
    "name": "Küchenlicht tagsüber an",
    "owner": "90EB5834A7",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 0
  },
  "2": {
    "actions": [
      {
        "address": "/lights/5/state",
        "body": {
          "on": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/2/state/presence",
        "operator": "eq",
        "value": "false"
      },
      {
        "address": "/sensors/2/state/lastupdated",
        "operator": "ddx",
        "value": "PT00:02:00"
      }
    ],
    "created": "2017-09-06T18:16:54",
    "etag": "93935b3c3c5dd19db015ff48404480a1",
    "lasttriggered": "2017-09-06T22:01:48",
    "name": "Flurlicht aus",
    "owner": "90EB5834A7",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 17
  },
  "4": {
    "actions": [
      {
        "address": "/groups/5/scenes/1/recall",
        "body": {
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/config/localtime",
        "operator": "in",
        "value": "T07:00:00/T20:00:00"
      },
      {
        "address": "/sensors/1/state/lightlevel",
        "operator": "lt",
        "value": "3000"
      },
      {
        "address": "/sensors/2/state/presence",
        "operator": "dx"
      },
      {
        "address": "/sensors/2/state/presence",
        "operator": "eq",
        "value": "true"
      }
    ],
    "created": "2017-09-06T18:27:51",
    "etag": "906ae2b1bbf05f333e46e2fea18b7d38",
    "lasttriggered": "none",
    "name": "Flurlicht tagsüber an",
    "owner": "90EB5834A7",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 0
  },
  "5": {
    "actions": [
      {
        "address": "/groups/5/scenes/2/recall",
        "body": {
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/config/localtime",
        "operator": "in",
        "value": "T20:00:00/T07:00:00"
      },
      {
        "address": "/sensors/1/state/lightlevel",
        "operator": "lt",
        "value": "3000"
      },
      {
        "address": "/sensors/2/state/presence",
        "operator": "dx"
      },
      {
        "address": "/sensors/2/state/presence",
        "operator": "eq",
        "value": "true"
      }
    ],
    "created": "2017-09-06T18:28:17",
    "etag": "906ae2b1bbf05f333e46e2fea18b7d38",
    "lasttriggered": "none",
    "name": "Flurlicht nachts an",
    "owner": "90EB5834A7",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 2
  },
  "8": {
    "actions": [
      {
        "address": "/lights/6/state",
        "body": {
          "on": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/5/state/presence",
        "operator": "eq",
        "value": "true"
      },
      {
        "address": "/sensors/5/state/lastupdated",
        "operator": "ddx",
        "value": "PT00:01:00"
      }
    ],
    "created": "2017-09-06T18:34:15",
    "etag": "906ae2b1bbf05f333e46e2fea18b7d38",
    "lasttriggered": "none",
    "name": "Küchenlicht aus",
    "owner": "90EB5834A7",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 0
  },
  "9": {
    "actions": [
      {
        "address": "/groups/6/scenes/2/recall",
        "body": {
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/config/localtime",
        "operator": "in",
        "value": "T20:00:00/T07:00:00"
      },
      {
        "address": "/sensors/4/state/lightlevel",
        "operator": "lt",
        "value": "3000"
      },
      {
        "address": "/sensors/5/state/presence",
        "operator": "dx"
      },
      {
        "address": "/sensors/5/state/presence",
        "operator": "eq",
        "value": "true"
      }
    ],
    "created": "2017-09-06T18:36:52",
    "etag": "906ae2b1bbf05f333e46e2fea18b7d38",
    "lasttriggered": "none",
    "name": "Küchenlicht nachts an",
    "owner": "90EB5834A7",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 2
  }
}
```

ebaauw, 2017-09-06:
> This is really complicated.

Yes, too complicated for me to maintain manually.  I create my rules using scripts, see https://github.com/ebaauw/ph.sh/blob/master/ph_rules.sh.  I'm still working on the documentation, but there are some comments in the scripts.