dresden-elektronik/deconz-rest-plugin #149: Feature request: setting _OnOff Transition Time_ and _On Level_

ebaauw, 2017-08-27:
Some more subtle differences in lights, although not caused by deCONZ.

Philips Hue lights don't support the _OnOff Transition Time_ and _On Level_ attributes of the _Level Control_ cluster.  The Hue lights behave as if these values are 4 and 255, i.e. they turn on or off with a default transition time of 0.4s and they turn on at the previous level (when they were turned off - not: powered off).  My other lights, OSRAM, innr, IKEA, ubisys, all support these attributes.  And, of course, they all use different default values for these attributes (from the Hue lights as well as from each other).  From a user experience, I want all lights in a room to behave the same.

Of course, I can update the attributes of the other lights to match the Hue behaviour through the deCONZ GUI, but it would be so much cooler if the REST API plugin would do this for me.  It would be even more cool if the API would support setting these, e.g. through `config.on.transitiontime` and `config.on.bri` (cf. `config.poweron.bri` and `config.poweron.ct` from #102) or through `config.bri_on` and `config.transitiontime_on` (cf. `config.ct_min` and `config.ct_max` from #123).

ebaauw, 2017-09-17:
Looks like these settings are reset when the light is powered off, so the desired settings should be written on device announcement.

abmantis, 2017-11-04:
This would be great. I'm starting to get into deconz, and noticed that the IKEA bulb has a weird turn on fade in, which takes half a second. It is weird because it seems to: fade to 50% quick, stop, fade to 100%. It is not really smooth. Reducing the OnOff Transition Time to from 5 to 1, makes it much better.

It would be nice if this was configurable on the web ui (maybe on the new beta? :P )

AndreasRavn, 2018-03-22:
Would love this, different transition times is killing me at the moment, and updating single bulbs through the GUI is quite cumbersome.

ebaauw, 2018-11-23:
Please keep this one open.  It will be very relevant this fall, with the expected new firmware for the Hue bulbs finally allowing a power-on setting.

alialrahma, 2019-09-04:
Bump also need this for my Gledopto GL-B-008Z

closed 2023-01-07 by manup