dresden-elektronik/deconz-rest-plugin #150: Private beta of new app

manup, 2017-08-27:
**Phoscon App**
There is a new app available as private beta which can be testet with recent deCONZ versions (ConBee, RaspBee).

To get access just leave a small note here and you will be invited to the private GitHub repository wich contains the install instructions.

Thanks in advance for any feedback.

Hector47, 2017-08-28:
Hello,
I'm interested in the beta app :-)

Thanks

Krocko, 2017-08-28:
I am also interested.

AntiHeld889, 2017-08-28:
I am also interested. :)

ebaauw, 2017-08-28:
Yes, please

buzatu, 2017-08-28:
Hi,
Me too, please :)

NisseDILLIGAF, 2017-08-28:
I'm interested :)

mploetne, 2017-08-28:
I'm also interested.
Thank you
Marko

snozzlebert, 2017-08-28:
Yes, please.

dghh, 2017-08-28:
I´m also very interested, thanks.

wvuyk, 2017-08-28:
Interested too here

grover, 2017-08-28:
I'm interested. Would be good as I'm just starting.

shayanhaider, 2017-08-28:
I'm interested too. Looking forward to trying out the new app.

YoungCereal, 2017-08-28:
More interested me is the app an only deamon no XServer interface ? than check me in ;)

manup, 2017-08-28:
> More interested me is the app an only deamon no XServer interface ? than check me in ;)

Still experimental, but you can start deCONZ without running X11 server.

    $ deCONZ -platform minimal --http-port=8080 --dbg-info=1

To start deCONZ at boot time use the *systemd* service:

    $ systemctl enable deconz

YoungCereal, 2017-08-28:
Naja wir sind ja beide deutsch ;)
nien ich habe ein System das ohne display oder grafischen Output ist und da ich mit dem gesamten Xserver echt zu viel Ressourcen verbrauche ist das nicht wirklich ne Lösung für mich.
Was ist eigentlich der unterschied zur neuen App ?

gruß
Michael

manup, 2017-08-28:
> Naja wir sind ja beide deutsch ;)

stimmt aber hier bei GH bitte englisch bleiben damit jeder was versteht :)

>nien ich habe ein System das ohne display oder grafischen Output ist und da ich mit dem gesamten Xserver echt zu viel Ressourcen verbrauche ist das nicht wirklich ne Lösung für mich.

Only a few megabytes for the X11 libs on sd-card, with todays sd-card sizes and prices this is hardly a real life dealbreaker.

deCONZ is able to run completely headless now, getting rid of X11 entirely will take some time and has no high priority due the the minimal benefits (safe 20-30 MB for X11 libs).

> Was ist eigentlich der unterschied zur neuen App ?

It's a complete overhaul of the old one with expanded device support.

alexsporn, 2017-08-28:
Count me in!

pfleghar, 2017-08-28:
Das ist spannend ... bin sehr interessiert an der Beta!

rtenklooster, 2017-08-28:
Yes, I would like to give it a try!

ebaauw, 2017-08-29:
>Thanks in advance for any feedback.

Be careful what you wish for ;-)

**General Impression**
- I really like where this is going.  The UI is refreshing
- The login screen asks for _Passwort_ (in German).
- App works over VPN!
- Would like to see support for sorting for Groups, Lights, Switches, Sensors on ID, Name, Type, Manufacturer, Model.

**Groups**
- Regular groups are reverse ordered by group ID, groups for switches are ordered on sensor ID.
- Took me a while to realise these are the groups linked to the switches, not the switches themselves.
- Hidden groups are shown anyway.
- Groups linked to not yet supported switches (yes, I’m talking about my ubisys dimmer ;-) are not shown at all.  Maybe show these as regular groups until support for the switch is added?
- Why the reduced functionality (no scenes, schedules, sensor controls) for groups linked to switches?
- I would put the slider for _Brightness_ to the left, and _Hue_ / _Saturation_ or _Temperature_ sliders to the right: easier on the eye when switching between these or between lights of different types.  Also, place the icon to switch between _Color_ and _Temperature_ under the _Color_ or _Temperature_ slider.
- The _Color_ on the slider doesn’t match the light colour.  You really need to do RGB to CIE conversion in the app and use `xy` instead of `hue` and `sat` - see Hue developer’s guidelines.  Alternatively, implement this in the deCONZ REST API - best discuss this in a different issue.
- The _Temperature_ slider doesn’t take into account min/max supported values (needs API support as well, see issue #123).

**Settings**
- App doesn’t allow changing the date/time format.
- Open Gateway is missing.

**Lights**
- My OSRAM plug shows as an outlet icon instead of a bulb icon - nice.  I take it, the icon is from the `type`?  Would be nice to show spots vs bulbs, etc like the Hue app, but this would require whitelisting lights and associated icons.  Maybe different icons for different types of light (Dimmable Light, Color Temeperature Light, etc), or maybe add a column for _Type_.
- Detailed window misses label for _Version_.  Maybe add _Date Code_ and _Type_?
- Lights are sorted on ID, as with the API.  I like that, because I hacked the database so my lights are numbered in the right order.  For less neurotic users, order on Name or custom defined order would make more sense.  Or add ID and allow sorting on columns.

**Switches**
- Love this screen, although I’m deeply disappointed that my Xiaomi smart cube isn’t shown ;-)
- Order is by sensor ID.
- Detailed window: 
  - maybe add _Date Code_, _Last Button Pressed_ (incl. date/time);
  - Would like option to enable/disable the switch (once we have `config.on` working in the API, see issue #90);
  - I expect we’ll be able to click the buttons on the image to set rules?

**Sensors**
- Love this screen!
- Order is by sensor ID.
- Motion symbol (walking man) is not really clear in showing the difference between motion and no motion.  Maybe add colour, or use different symbols (cf. traffic light symbols: stopped man vs walking man).
- Having two decimals for temperature and humidity makes no sense, given the accuracy of the sensors - just round to nearest integer.
- Detailed window, general:
  -  Would like option for to enable/disable the sensor (once we have `config.on` working in the API, see issue #90).
- Detailed screen Hue motion sensor:
  - Icon is too big for iPad - text is shown below instead of next and doesn’t scroll so _Temperature_ and _Refreshed_ cannot be seen.
  - Would like different _Refreshed_ values for Motion, Light level and Temperature.
  - Would like config options for `duration`, `tholddark`, `tholdoffset`, and for `sensitivity`, `ledindication`, and `usertest` (once we have these working in the API, see issue #20).  Maybe add wizzard for calibrating light level and sensitivity, cf. Hue app.

bdickhaus, 2017-08-29:
I am interested.

manup, 2017-08-29:
@ebaauw Thanks for comprehensive review, very good points which we'll address soon there possible..

>General Impression

> * I really like where this is going. The UI is refreshing
> * The login screen asks for Passwort (auf Deutsch).

Will be fixed asap, there are more translations which will be fixed soon. We also consider https://pontoon.mozilla.org/ to support more languages in the future.

> * App works over VPN!

Yes in the end it just connects to the gateway via its IP, supporting *common* subnets and anything WebRTC provides. In comparison to the old web app it doesn't need to run on the same origin (gateway). When WebRTC will be supported in full, access will also work remotely without needing a VPN in a secured peer-to-peer fashion.

> * Would like to see support for sorting for Groups, Lights, Switches, Sensors on ID, Name, Type, Manufacturer, Model.

Absolutely we plan to add sorting &mdash; also via drag and drop &mdash; to all of those, once the basics are solid. For some lists we will also add filtering to quickly find devices and lights by any characteristic.

> **Groups**

> * Regular groups are reverse ordered by group ID, groups for switches are ordered on sensor ID.
> * Took me a while to realise these are the groups linked to the switches, not the switches themselves.

The idea is that switches are just switches to which lights can be assigned. Currently scenes are only supported for normal light groups (3 bulb icon) and switches which support scenes via ZLL Scene Cluster (JUNG, GIRA, Busch-Jaeger, dresden elektronik).

Switches like Philips Hue dimmer or the IKEA Switch don't send *recall scene* commands, which is why the scenes button is not active for them yet. Once we figured out a simple way to configure "custom" commands for buttons in the UI these will follow.

Since lights can be assigned to multiple switches and groups pressing the buttons should also update the lights state in the group.

> * Hidden groups are shown anyway.
> * Groups linked to not yet supported switches (yes, I’m talking about my ubisys dimmer ;-) are not shown at all. Maybe show these as regular groups until support for the switch is added?

> *Why the reduced functionality (no scenes, schedules, sensor controls) for groups linked to switches?

See the commend above, currently this is like a whitelist: Scenes are supported for some switches, for sensor controls we have decided it is easier to configure *sensor controls* for a normal group and in there assign sensors and switches which should act in the specific sensor control.

Imagine a *sensor control* for a long floor with multiple motion sensors and switches.

> * I would put the slider for Brightness to the left, and Hue / Saturation or Temperature sliders to the right: easier on the eye when switching between these or between lights of different types. Also, place the icon to switch between Color and Temperature under the Color or Temperature slider.

Makes sense, we'll adjust that soon.

> * The Color on the slider doesn’t match the light colour. You really need to do RGB to CIE conversion in the app and use xy instead of hue and sat - see Hue developer’s guidelines. Alternatively, implement this in the deCONZ REST API - best discuss this in a different issue.

Yes we'll add more fine tuned color modes respecting the specific capabilities of the devices over time. Hue saturation was chosen as start since most devices support it (albeit it's not very precise).

> * The Temperature slider doesn’t take into account min/max supported values (needs API support as well, see issue #123).

Agree, that's already on my todo list :)

> **Settings**

> * App doesn’t allow changing the date/time format.

Our ultimate goal is to let the app determine the correct date/time format on it's own. Currently this is not very sophisticated and fails in many cases, but it should be possible. The plan is to decide afterwards if a manual changing should be implemented too.

> * Open Gateway is missing.

This will be added soon (to fake the Hue push button pressed).

> **Lights**

> * My OSRAM plug shows as an outlet icon instead of a bulb icon - nice. I take it, the icon is from the type? Would be nice to show spots vs bulbs, etc like the Hue app, but this would require [...]

Yes that's planned, for switches and sensors we already created tables to provide meaningful human readable product names and icons. For lights these will follow.

> * Detailed window misses label for Version. Maybe add Date Code and Type?

Strange, I'm sure there was a label once, should be back soon. Agree the Date Code can be added too. For the type the device tables will be used to provide translated friendly types.

> * Lights are sorted on ID, as with the API. I like that, because I hacked the database so my lights are numbered in the right order. For less neurotic users, order on Name or custom defined order would make more sense. Or add ID and allow sorting on columns.
[... 32 lines cut ...]
This should be fixed now, currently by a simple filled symbol for detected motion.

> * Having two decimals for temperature and humidity makes no sense, given the accuracy of the sensors - just round to nearest integer.

Agree

> * Detailed window, general:
>     * Would like option for to enable/disable the sensor (once we have config.on working in the API, see issue #90).

[... cut ...]
>     * Would like config options for duration, tholddark, tholdoffset, and for sensitivity, ledindication, and usertest (once we have these working in the API, see issue #20). Maybe add wizzard for calibrating light level and sensitivity, cf. Hue app.

The ledindication will arrive first (since many sensors provide it). Albeit it will take a while a wizard for the further configuration should be possible.

Thanks again for the review, this is very helpful. We will improve and update the app regularly like the beta versions.

wizkidorg, 2017-09-01:
I would like to try the new Beta App.

adsmf, 2017-09-01:
Another interested in the beta.

wizkidorg, 2017-09-04:
Any update on the private beta app?

manup, 2017-09-04:
>Any update on the private beta app?

Hi, yes we are currently fixing a few bugs and UX issues based on feedback, during the week you will get access to the app.

holli73, 2017-09-07:
i'm interested as well

felixstorm, 2017-09-08:
I would also like to to try the new beta app.

stefan-kaiser, 2017-09-09:
I like to participate too.
Thanks in advance!

CannyDuck, 2017-09-11:
Please give me a chance to test it. Thanks.

mtornblad, 2017-09-19:
I would like to test it out also!

Yannic-HAW, 2017-09-21:
I would like to be beta tester as well

mrmadcow, 2017-09-21:
Hi, I'd be interested too, please.

abmantis, 2017-09-22:
Hey, I would be glad to test it too, please.

wizkidorg, 2017-10-02:
Any update on the availability of the Private beta?

manup, 2017-10-04:
> Any update on the availability of the Private beta?

Yep, sorry for the delay as usual it takes longer than expected :/ we had to rethink a few ux choices, which I wanted to be solved before inviting more people &mdash; details matter. These changes will be finished within the coming days and I post here when invitations are out.

snillevilla, 2017-10-13:
Hi, just found this. I am a huge fan of your products and am really, really interested to test the app. Please invite!

maikburd, 2017-10-22:
Hello,

I'm verry interested. Can you invite me please?

sandman72, 2017-10-22:
I started to play around with the "old" app this weekend and ran into a few limitations.

I would love to try out the new version. Please invite me!

to0b, 2017-10-23:
Hello everyone,

I'm intrested to test the new Software. Could you please invite me?

markuspf, 2017-10-24:
I would like to try the new app, too.

GatoPharaoh, 2017-10-25:
hi :)
I'm verry interested. Can you invite me please?

manup, 2017-10-28:
Took a while; invitations are out, please make sure to use latest version of deCONZ :)

mysteryan, 2017-10-28:
I need an invite

ronaldvans, 2017-11-03:
I would appreciate an invite as well!

etique57, 2017-11-04:
@manup more than willing to test the new beta app ;)

banjohb, 2017-11-05:
Hey :) Want test the beta too. Please invite *_*

oleg77s, 2017-11-07:
it is interesting to me to test a new app

andrejem, 2017-11-09:
Hey, I also want to try out. thx.

dkmh, 2017-11-10:
If testing is still in progress, please add me in.

woodkazor, 2017-11-10:
Please let me test it! Dankeschön

knappkaj, 2017-11-16:
Hi, I would like to have an access to the BETA Repo. Thank you in advance.
Best regards from Austria.

ziggyre, 2017-11-18:
Very good work! Please invite me. :-)

manup, 2017-11-20:
Invitations are out, please make sure to use latest version of deCONZ.
Ubuntu and Windows versions will be updated during the next days too.

josefku, 2017-11-20:
Hallo, I would like to have an access to the BETA App. Thank you.

basty2, 2017-11-20:
I would like to test the beta App.

ensamresande, 2017-11-20:
Interested!

sparkcz, 2017-11-21:
Please invite me, thanks.

simonporter007, 2017-11-26:
More than willing to give this a go!

manup, 2017-11-26:
invitations sent out, hope to get the app public in the next days.

[10 more replies]

closed 2018-09-17 by manup