dresden-elektronik/deconz-rest-plugin #151: DeConz crashing after Homebridge action

AntiHeld889, 2017-08-29:
Hello at first sorry for my bad english.

Issuses:

I have create an action in Homebridge "leave home" with 21 Devices (bulbs) which working over Deconz and Homebridge-hue Plugin. If I start these action DeConz is crashing and can't reconnect to the network. After I restart the program it shows me Firmware update (but all Firmwares are up to date) when it is finished, Deconz is working again.

ebaauw, 2017-08-29:
> I have create an action in Homebridge "leave home" with 21 Devices (bulbs) which working over Deconz and Homebridge-hue Plugin

I assume you mean the _Leave Home_ scene in HomeKit?

> If I start these action DeConz is crashing

deCONZ shouldn't crash by anything done by homebridge-hue (or any app using the deCONZ REST API).  Does this happen every time you activate the HomeKit scene, or did this happen only once?

If it happens every time, could you:
- Start deCONZ manually, capture its output;
- Run `homebridge -D` manually, capture its output (see **Troubleshooting** in homebridge-hue README);
- Reproduce the crash;
- Post the outputs of deCONZ and homebridge here?

> If I start these action DeConz is crashing and can't reconnect to the network. After I restart the program it shows me Firmware update

I assume you mean the RaspBee won't connect to the _ZigBee_ network?  I've experienced this once - as if the RaspBee was zapped completely and needed to be re-initialised, starting with flashing the firmware.  I haven't been able to reproduce this, however.  I suspect it might be related to the RaspBerry Pi producing too much heat (see issue #147).  What other programmes are you running on your Raspberry Pi?

AntiHeld889, 2017-08-29:
>I assume you mean the Leave Home scene in HomeKit?

Yes I mean the Leave Home scene in HomeKit.

> deCONZ shouldn't crash by anything done by homebridge-hue (or any app using the deCONZ REST API). Does this happen every time you activate the HomeKit scene, or did this happen only once?

Yes every time I tested the scene Deconz lost the connection Homebridge runs normally after that I don't think that RaspBee lost connection by anything done by homebridge-hue. But perhaps because there are too many devices or commands at once?

Homebridge:
`[2017-8-29 14:31:01] [RaspBee] deCONZ-GW: hue bridge request #69: ok
[2017-8-29 14:31:01] [RaspBee] deCONZ-GW: hue bridge request #70: ok
[2017-8-29 14:31:03] [RaspBee] deCONZ-GW: hue bridge request #71: get /config
[2017-8-29 14:31:03] [RaspBee] deCONZ-GW: hue bridge request #71: ok
[2017-8-29 14:31:03] [RaspBee] deCONZ-GW: hue bridge request #72: get /sensors
[2017-8-29 14:31:03] [RaspBee] deCONZ-GW: hue bridge request #72: ok
[2017-8-29 14:31:03] [RaspBee] deCONZ-GW: hue bridge request #73: get /lights
[2017-8-29 14:31:03] [RaspBee] deCONZ-GW: hue bridge request #73: ok
[2017-8-29 14:31:08] [RaspBee] deCONZ-GW: hue bridge request #74: get /config
[2017-8-29 14:31:08] [RaspBee] deCONZ-GW: hue bridge request #74: ok
[2017-8-29 14:31:08] [RaspBee] deCONZ-GW: hue bridge request #75: get /sensors
[2017-8-29 14:31:08] [RaspBee] deCONZ-GW: hue bridge request #75: ok
[2017-8-29 14:31:08] [RaspBee] deCONZ-GW: hue bridge request #76: get /lights
[2017-8-29 14:31:08] [RaspBee] deCONZ-GW: hue bridge request #76: ok
[2017-8-29 14:31:11] [Harmony Hub] debug Refreshing current activity
[2017-8-29 14:31:13] [RaspBee] deCONZ-GW: hue bridge request #77: get /config
[2017-8-29 14:31:13] [RaspBee] deCONZ-GW: hue bridge request #77: ok
[2017-8-29 14:31:13] [RaspBee] deCONZ-GW: hue bridge request #78: get /sensors
[2017-8-29 14:31:13] [RaspBee] deCONZ-GW: hue bridge request #78: ok
[2017-8-29 14:31:13] [RaspBee] deCONZ-GW: hue bridge request #79: get /lights
[2017-8-29 14:31:13] [RaspBee] deCONZ-GW: hue bridge request #79: ok
[2017-8-29 14:31:18] [RaspBee] deCONZ-GW: hue bridge request #80: get /config
[2017-8-29 14:31:18] [RaspBee] deCONZ-GW: hue bridge request #80: ok
[2017-8-29 14:31:18] [RaspBee] deCONZ-GW: hue bridge request #81: get /sensors
[2017-8-29 14:31:18] [RaspBee] deCONZ-GW: hue bridge request #81: ok
[2017-8-29 14:31:18] [RaspBee] deCONZ-GW: hue bridge request #82: get /lights
[2017-8-29 14:31:18] [RaspBee] deCONZ-GW: hue bridge request #82: ok
[2017-8-29 14:31:20] [RaspBee] Komodenspot 1: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Couchspot 2: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [Harmony Hub] Queue activity to -1
[2017-8-29 14:31:20] [Harmony Hub] Set activity to -1
[2017-8-29 14:31:20] [RaspBee] Komodenspot 2: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Komodenspot 3: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Nachttisch 2: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Nachttisch 1: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Kugel: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Couchspot 4: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Couchspot 1: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Gadrobenspot: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Flurspot 4: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Flurspot 1: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [IronMan] Turning IronMan off
[2017-8-29 14:31:20] [RaspBee] Light 2: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Gadrobe: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Flurspot 2: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Flurspot 3: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Couchspot 3: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Deckenlampe: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] Schlafzimmer Deckenlampe: homekit power changed from 1 to 0
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #83: put /lights/14/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #84: put /lights/7/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #85: put /lights/13/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #86: put /lights/12/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #87: put /lights/5/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #88: put /lights/6/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #89: put /lights/4/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #90: put /lights/10/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #91: put /lights/8/state {"on":false}
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #92: put /lights/11/state {"on":false}
[2017-8-29 14:31:20] [Denon AVR] denon avr powered false
[2017-8-29 14:31:20] [RaspBee] deCONZ-GW: hue bridge request #83: ok
[... 189 lines cut ...]
14:31:27:822 aps request id: 63 finished, erase from queue
14:31:27:822 emit artificial APSDE-DATA.confirm id: 63
14:31:27:902 aps request id: 64 finished, erase from queue
[... cut ...]
14:36:04:507 APS-DATA.request id: 54, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0036, ep: 0x00 queue: 0 len: 3
14:36:04:542 aps request id: 54 timeout
14:36:04:542 aps request id: 54 erase from queue
14:36:04:542 emit artificial APSDE-DATA.confirm id: 54
14:36:15:507 APS-DATA.request id: 93, addrmode: 0x03, addr: 0x00212effff014355, profile: 0x0104, cluster: 0x0019, ep: 0x0A queue: 0 len: 5
14:36:15:582 aps request id: 93 finished, erase from queue
14:36:15:582 emit artificial APSDE-DATA.confirm id: 93 `

ebaauw, 2017-08-29:
> But perhaps because there are too many devices or commands at once?

Looking at the homebridge log, you see that HomeKit switches off 19 lights in bulk when activating the scene at 14:31:20.  By design, homebridge-hue throttles the corresponding requests to turn off each light, issuing the first 10 requests to deCONZ immediately, and the remaining 9 only after deCONZ has returned a response for a previous request (so it never has more than 10 outstanding requests with deCONZ).  
You can tweak the number of outstanding requests by setting `parallelRequests` in config.json (see https://github.com/ebaauw/homebridge-hue/wiki/Configuration), but I'm afraid that won't help (as deCONZ returns responses for all 19 requests in the same second anyways).

Somehow, deCONZ sets `state.reachable` to `false` for 20 lights, shortly after this action (presumably because the RaspBee has lost the connection to the ZigBee network?).  homebridge-hue detects this when polling deCONZ at 14:31:28.  I think you're running an older version of homebridge-hue that doesn't yet use the websocket notifications?  This is logged at the beginning of the homebridge output, along with the types of lights you have.

From the deCONZ log, you see it handles the first 5 requests (61-65) to the lights to turn off (cluster 0x0006) at 14:31:20.  After 7 seconds it gives up on the first request 61, and handles the next (66), but no sign of the remaining requests...  
I'm afraid this is an issue with deCONZ, beyond my understanding, but it doesn't seem like it's trying to handle too many requests simultaneously.  Hopefully @manup can shed a light on this.  What version of deCONZ and RaspBee firmware are you running (see _Help_|_About_ in the deCONZ GUI)?

AntiHeld889, 2017-08-29:
I have used the last version of homebridge-hue (0.5.24) now 0.5.25.
I have test it with "parallelRequests": 3  but same Problem

Deconz: 2.04.69
Firmeware: 0x26160500

AntiHeld889, 2017-08-29:
Error found another plugin was still active for a 433mhz transmitter which on pin 17 then caused the error.

Very stupid of me!

closed 2017-08-29 by AntiHeld889