dresden-elektronik/deconz-rest-plugin #153: Xiaomi smart plug

rtenklooster, 2017-08-30:
My [xiaomi smart plug](https://www.gearbest.com/living-appliances/pp_344666.html)
Shows up as a Light. I'm able to turn it on / off / toggle.
The power consumption and mails voltage are not listed in the rest api.
The mains voltage is visible in the attributes field, so is the power consumption (  0x0055 shows the actual usage in watts)
[image]

It would be great if this data would be accessible from the rest-api.
If I need to provide more information, please let me know.

manup, 2017-08-30:
Yes we can add it to the API as `config/mainsvoltage` similar to `config/battery`. Not sure yet if it needs to be another sensor or if the attribute should just be attached to the light.

rtenklooster, 2017-08-30:
> Yes we can add it to the API as config/mainsvoltage similar to config/battery.

That would be great!

Technically it's not a light.  It's a plug, there could be anything connected to those plugs.

ebaauw, 2017-08-30:
> Yes we can add it to the API as config/mainsvoltage similar to config/battery.

Voltage is one thing; I would really be interested in the power consumption.  I think it should be a `state` attribute `consumption`, not a `config` attribute.

> so is the power consumption ( 0x0055 shows the actual usage in watts)

I assume that's attribute 0x0055 of the _Analog Input_ cluster (0x000c)?  Which endpoint, 02, 03, or both?  Any idea what the _Binary Input_ cluster (0x000f) on endpoint 4 is used for? 

On the ubisys D1, power consumption is reported on endpoint 04, cluster 0x0b04 (Electrical Management), attribute 0x050b (Active Power) in W (int16).  It's value ranges from 0 when the connected lights are off to 251 when they're at `bri` 254.  I'm controlling three 80W 240V halogen tubes from the dimmer, so that seems close enough.  
As to be expected from ubisys, this is fully in line with the ZCL spec:
> Represents the single phase or Phase A, current demand of active power delivered or received at the premises, in Watts (W). Positive values indicate power delivered to the premises where negative values indicate power received from the premises.

The OSRAM plug provides the same cluster and attribute, but on the 03 ZLL endpoint that also holds the _On/Off_, _Groups_, and _Scenes_  clusters).  However, it isn't working: it reports 28000 when the plug is off and 400 when it's on, regardless whether something it actually consuming power (or even plugged into the plug).  I'm not sure if this is a physical or firmware-related problem.  I have another OSRAM plug that is broken (after pairing, it keeps on clicking/swiching on and off and doesn't respond to ZigBee commands).  The plug is at V1.03.21.  The lightify app reports the latest firmware for the plug is 01020315 - could be the same version (0x15 = 21).

One of my OSRAM E14 bulbs (the one with the older firmware V1.03.20) also has this cluster and attribute.  It reports 400 when the light is off; and 5800 when it's on at `bri` 254 to 600 at `bri` 1.  My guess would be the value is in mW.  The other E14, on firmware V1.04.12, lacks this cluster (maybe because of non-compliance to ZCL?).

> Not sure yet if it needs to be another sensor or if the attribute should just be attached to the light.

Since consumption is on a different endpoint, it should be a separate (ZHAPower?) sensor for the Xiaomi and the ubisys.  I makes sense to do the same for the OSRAM (if we ever get a plug (firmware?) where this works), with `uniqueid` `84:18:26:00:00:xx:xx:xx-03-0b04` for the sensor vs `84:18:26:00:00:xx:xx:xx-03` for the light resource.

rtenklooster, 2017-08-30:
Endpoint 02 cluster 0x000C attribute 0x0055 shows the current power consumption in watts.
Endpoint 03 cluster 0x000C attribute 0x0055 shows 0.00 at the moment, I will report tomorrow if it's the total power consumption.
Edpoint 64 cluster 0x000f attribute 0x0055 shows false, I have to find out what this means. I know there is a default setting after power loss. This might have to do something with that. After powerloss, turn on or remain off. I will test it tomorrow.

rtenklooster, 2017-08-31:
The Rest API seems to be reporting the wrong attribute anyway.
`    "groups": {},
    "lights": {
        "1": {
            "etag": "dd10f779b78ed6e58003165c9294fa7b",
            "hascolor": false,
            "manufacturername": "Unknown",
            "modelid": "",
            "name": "Light 1",
            "state": {
                "alert": "none",
                "on": false,
                "reachable": true
            },
            "swversion": "",
            "type": "Smart plug",
            "uniqueid": "00:15:8d:00:01:2d:af:2d-01"
        }`
While deconz: 
Endpoint 01 cluster 0x0006 attribute 0x0000 reports true.

Endpoint 03 cluster 0x000C attribute 0x0055 This attribute appears to be counting the consumed energy in kWh (it's reading 0.10 atm.)

ebaauw, 2017-08-31:
> The Rest API seems to be reporting the wrong attribute anyway.
> Endpoint 01 cluster 0x0006 attribute 0x0000 reports true.

Odd.  You might want to double-check by reading the attributes of the 0x0006 (On/Off) cluster from the _Cluster Info_ panel in the deCONZ GUI.  This should refresh the value shown there _and_ the RESP API attribute.  Is the plug actually turned on or off?

> Endpoint 03 cluster 0x000C attribute 0x0055 This attribute appears to be counting the consumed energy in kWh

Cool.  I'm very much tempted to order one...  Does it come with EU plugs, or do you need to use an adapter?  The pictures on the Chinese web shops only show the front side :-(

rtenklooster, 2017-08-31:
The plug is turned on.
I've double checked: Endpoint 01 cluster 0x0006 attribute 0x0000 reports true. The API reports On: false.

You would need an adapter for this plug because it has an AU plug.

ebaauw, 2017-08-31:
> The plug is turned on.
I've double checked: Endpoint 01 cluster 0x0006 attribute 0x0000 reports true. The API reports On: false.

Very odd.  Does `state.on` change value when you switch the plug off and on?  Do you see an event on the websocket?  
If not, my guess would be that you have a "orphaned" resource, which is no longer linked to the node (also inspired by the missing values for `manufacturername` and `modelid` - do these show in the GUI when you read the attributes of cluster 0x0000?).  
I'd try deleting the resource (through the API), deleting the node (from the GUI), and re-pairing.

> You would need an adapter for this plug because it has an AU plug.

That's a pity.

> Technically it's not a light. It's a plug, there could be anything connected to those plugs.

Yes.  From ZigBee perspective it smells like a light, so it's exposed as a light resource (same on the Hue bridge; the OSRAM Lightify app also shows plugs and lights on the same screen).  You can check `type` in the API to know that it's in fact a plug, rather than a light.

rtenklooster, 2017-09-04:
Added the corresponding data to my excel sheet.

abmantis, 2017-11-04:
Is there any progress on this? I have also have a wall plug that reports power, but it is not available in the REST api (I can control it tough).

rtenklooster, 2017-11-30:
@manup Is it a lot of work to add the switch actions to the rest api? 
Currently switching trough deconz is working, Beeïng able to see power consumption would be nice, however adding the normal switch actions to the api would be a great start.

manup, 2017-11-30:
From the above REST output I see the smart plug is shown as light. Beside the attribute is reported as wrong state (Xiaomi might use reporting instead of ZCL read for onoff cluster like for their other devices).

Does REST PUT request not already work?

```json
PUT /lights/1/state

{ "on": true }
```

If you can provide a sniffer log it would be helpful to see if something is send to the device and also if the plug is sending ZCL attribute reports.

rtenklooster, 2017-11-30:
Hmm.. I did a new query.
The rest api doesn't reports the smart plug anymore.
It's only showing my two ikea bulbs.

Switching it from the deconz app doesn't generate websockets events.

manup, 2017-11-30:
Is it visible in deCONZ? Just looked in the code and the smart plug type (0x0051) should be supported. You may also try to power-cycle the plug to force it to send a device announce command.

rtenklooster, 2017-11-30:
I gave it a power cycle, still nothing in the rest-api and neither in deCONZ. 
Will try to add another one tomorrow. Maybe this one hasn't survived my deCONZ upgrades and I might have to remove and re-add the outlet in order to get it to work.

manup, 2017-11-30:
Sounds like a plan :)

rtenklooster, 2017-11-30:
F#ck tomorrow :)
Just added a brand new one.
[image]

deCONZ is showing and able to control. Webapp is empty, rest is still only showing my two ikea bulbs.

manup, 2017-11-30:
Hehe. Ah I see, the reason it isn't showing up in the REST API is that other than the one from the above screenshot this doesn't have the 0x0051 device id but 0xffff which isn't recognized by deCONZ.

Does the Basic cluster show anything useful? If not we may create the light resource by identify the device via endpoint 02 (Mains Power Outlet) and use OnOff cluster from endpoint 01 to control it.

It's a bit of hack but everything with Xiaomi is it seems :)

rtenklooster, 2017-11-30:
[image]

manup, 2017-11-30:
Oh that's not too bad the model id is quite handy here.

rtenklooster, 2017-11-30:
It's getting quite "mesh"i here... 
[image]
Didn't even add the 9 xiaomi smoke detectors and 12 on/off contacts and 2 ikea  bulbs hehe.

rtenklooster, 2017-11-30:
After pressing the read button all info is listed:
[image]

manup, 2017-11-30:
It's good to know that the plug reports the model id on it's own too, that way the resource can be created very quickly after a power-cycle, I'll cook something together for the next version.

rtenklooster, 2017-11-30:
Great! Guess after that, there's just one thing remaining to make my xiaomi gateway obsolete.... 
It has to do something with cooking gone bad... 
[image]

manup, 2017-11-30:
I know I know, it's on the radar :)

rtenklooster, 2017-11-30:
Relaxt :-) No need to hurry. Very happy with the progress so far. Thanks for your support!

christmasjumper, 2017-12-22:
Hi,

Apologies if this is the wrong place to post, a new Raspbee user here. Is there a list of known plugs that work with deconz and end up exposed via the REST Api please? I'm considering the Samsung smarthings plug as I'm in the UK.

Thanks

g3wd, 2018-02-08:
any updates on this issue? cant see anything in the change logs for the last couple of versions. 
Super excited to get the plugs working w/o an extra gateway. 

thanks

abmantis, 2018-03-19:
Still no support? Running 2.05.15 and it is not showing any consumption for Xiaomi Plugs nor other (non-xiaomi) plugs I have here.

ebaauw, 2018-03-19:
Did you re-pair the plugs?  What other plugs do you have?

abmantis, 2018-03-19:
@ebaauw Yes I did repair them. Deconz shows the power clusters, but the rest API does not have the info. The other plugs are not commercially available, and I'm not sure if I should mention them at the moment.

ebaauw, 2018-03-19:
Sensors need to be whitelisted, without any info on them, the REST API won’t create ZHAConsumption or ZHAPower sensor resources.

The Xiaomi smart plug should be supported since v2.05.08.  Make sure to open the network from the web app or from Phoscon and refresh the basic cluster in the GUI.  If still no resource is created, best double-check the plug’s details, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/432#issuecomment-368345179.

abmantis, 2018-03-20:
@ebaauw My bad! They do show up for the Xiaomi plug! I was checking the "/lights" instead of "/sensors", since I was expecting them to show as "/lights/plugid" attributes! :smile: 

Any idea why they also have a presence cluster?

ebaauw, 2018-03-20:
> Any idea why they also have a presence cluster?

Not really.  Do you mean ZigBee _Occupancy Sensing_ cluster or ZHAPresence resource in the REST API?  Could you list the resource?  Does it have the same mac address in the `uniqueid` as the light resource and the ZHAConsumption resource?

> Regarding the other plugs, what info is needed?

Device Type, Model ID, Manufacturer Code, Manufacturer Name, mac address prefix, overview of clusters (especially how they report consumption) and unit they report consumption etc in.  Easiest to post some screenshots from the deCONZ GUI.

abmantis, 2018-05-20:
[image]
Here's a screenshot of the Simple Metering (0702) cluster. Is that enough? 

The device already shows up as a light. Couldn't deconz simply add a sensor for every device that have the Simple Metering cluster, without needing it to be whitelisted?

ebaauw, 2018-05-20:
>  Is that enough?

No, see above.  Please provide screenshots of the _Node Info_ panel and the _Cluster Info_ panel for the _Basic_ cluster.

For the _Simple Metering_, attribute 0x0000 should report the total consumption devilvered (since the last reset).  Would that be 102.292 kWh or 1022.92 kWh?  Same question for 0x0400, that should report the current power drawn.  Would that be 62W or 6.2W?

Out of interest: what plug is that?  Where did you buy them?  Do they come in an EU version?

>  Couldn't deconz simply add a sensor for every device that have the Simple Metering cluster, without needing it to be whitelisted?

The ZigBee standard leaves a lot of room for interpretation, and quite some vendors take a very liberal interpretation, so typically some custom code is needed per device type or brand.  The alternative would be to support only your own brand and standard devices, as e.g. the Hue bridge.

closed 2019-03-30 by stale[bot]