dresden-elektronik/deconz-rest-plugin #154: Feature request: MQTT publishing

rtenklooster, 2017-08-30:
It would be great if deconZ supported MQTT.
Maybe at first only publishing events like motion / switch state changes.

This would allow us to use those events in all kind of controllers / esp modules / loggers etc.
I know websockets allow the same kind of action however MQTT is widely supported where websockets need custom coding on almost every controller / platform I have seen.

For example publishing to 
`/decomZ/Out/Sensor/%sensorId%/State/buttonevent "1002"`
`/decomZ/Out/Sensor/%sensorId%/State/lastupdated "2017-08-30T22:23:31"`

Of course it would be great if all switches and lights etc. could be read and set using MQTT however for the time crucial events originated from deconZ MQTT support would be great!

manup, 2017-08-31:
I have no experience with MQTT yet, but it's quite visible in the tech news lately and seems to be very interesting since it provides a more standard way.

> This would allow us to use those events in all kind of controllers / esp modules / loggers etc.
I know websockets allow the same kind of action however MQTT is widely supported where websockets need custom coding on almost every controller / platform I have seen.

Valid points, the events which are fired via websocket are not restricted to websockets, therefore it should be possible to to support further destinations like MQTT subscribers.

Qt has recently added support for MQTT:
https://blog.qt.io/blog/2017/08/14/introducing-qtmqtt-protocol/

The current focus is on the existing issues and release of the new app, but maybe it will be added at some point. Or as alternative I can provide stubs for MQTT events handlers and volunteers may bring it to life and forward the events to subscribers :)

rtenklooster, 2017-11-16:
Let's close this request. I've created a node-red mqtt bridge. Supporting more sensors / actuators should come first.

closed 2017-11-16 by rtenklooster

afoeder, 2021-03-16:
I'd like to bring this back on the table; for plugin devs it should be almost a no-brainer to bring MQTT support when the websocket is there already. Can we have this?