dresden-elektronik/deconz-rest-plugin #155: Renaming motion sensors does not reflect in deCONZ until restart

wvuyk, 2017-08-31:
If I add a motion sensor (tested IKEA and Philips) and I rename them with the REST API, the REST API reports the correct new name. The interface of deCONZ however does not show the new name until it is restarted.
The deCONZ interface does show the correct new name for changed lights immediatly. 

For the Philips Motion sensor I do rename every (sub) sensor, the name shown is mostly the temperature sensor. But only after a restart the new name is shown. The IKEA is only reporting motion, so only this sensor is renamed, but not immediatly reflected in the deCONZ interface. 
Also did another test just now with the IKEA remote, which is also not adjusting its name in deCONZ interface.

ebaauw, 2017-08-31:
I never understood the naming of nodes in the deCONZ GUI.  You can set the name (_User Descriptor_) in the _Node Info_ panel, but that only works for the RaspBee.  I guess none of my other devices support setting the _User Descriptor_.  I used to be able to set the node names by editing `~/.local/share/dresden-elektronik/deCONZ/session.default` (when deCONZ is not running), but this no longer seems to work.

I think it's because lately the names are actually overwritten with the `name` of the corresponding resource by the REST API plugin, when a node is (re-)discovered (after restarting deCONZ).  In older versions, the node name would change immediately, when setting the resource name.  The plugin stores the names in its database (`~/.local/share/dresden-elektronik/deCONZ/zll.db`).  I prefer the old functionality.

For the Hue motion sensors (and other nodes that have multiple associated resources), it's a roulette wheel which resource's `name` is used to set the node name.  I've hacked the plugin database, so the sensor IDs for all my Hue motion sensor resources are xx1 for the ZHAPresence sensor, xx2 for the ZHALightLevel sensor, and xx3 for the ZHATemperature sensor (where xx is the room number).  Yet some nodes show the ZHAPresence name, some the ZHALightLevel, and some the ZHATemperature.

wvuyk, 2017-09-06:
Checked today with the new version 2.0.04.75. Now indeed all sensors are showing the name of the motion sesnor AND are renamed directly after the rename through the rest API. 
Thanks for this solutions. Guess this issue is closed 👍

closed 2017-09-06 by wvuyk