dresden-elektronik/deconz-rest-plugin #156: Xiaomi wireless switch double click long press events

rtenklooster, 2017-08-31:
[The Xiaomi lumi switch](https://www.gearbest.com/smart-light-bulb/pp_257679.html) is supported, however only on / off events are shown. ( They get cheaper every day.. now only €5,97)
The switch should submit those events:

- Off
- Click
- Long click
- Long click release
- Double click

I have uploaded my log to [pastbin](https://pastebin.com/5BdKRtVf),

It's pretty hard to identify witch event belongs to witch state, because my pi will crash when i close deconz. I will submit another issue for that. I believe this events have something to do with the click events:
`14:32:52:673 APS-DATA.request id: 154, addrmode: 0x02, addr: 0x168a, profile: 0x0000, cluster: 0x0002, ep: 0x00 queue: 5 len: 3`

`14:32:53:548 APS-DATA.request id: 165, addrmode: 0x02, addr: 0x168a, profile: 0x0000, cluster: 0x0002, ep: 0x00 queue: 6 len: 3`

`14:33:02:107 APS-DATA.request id: 206, addrmode: 0x02, addr: 0x16c2, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 4 len: 2`

`14:33:05:467 APS-DATA.request id: 222, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 4 len: 2`

`14:33:23:524 APS-DATA.request id: 58, addrmode: 0x02, addr: 0x168a, profile: 0x0000, cluster: 0x0002, ep: 0x00 queue: 5 len: 3`

ebaauw, 2017-08-31:
The Xiaomi switch actually only sends events for button pressed and button released.  The Xiaomi Aqara switch is even more minimalistic: it only sends events for button released.  These events are to profile 0x0104 (ZHA), cluster 0x0006 (On/Off) and attribute 0x0000 (OnOff).  See #89.

The 0x0002 and 0x00031 clusters are ZDP commands (endpoint 0x00), used for network management.

I think the Mi app (or maybe the Mi gateway?) turns these pressed/released events into clicks, double-clicks, long clicks, etc.

rtenklooster, 2017-08-31:
Thanks Erik, for explaining. 
I'd liked the multiple actions but can live without.

closed 2017-08-31 by rtenklooster