dresden-elektronik/deconz-rest-plugin #157: RPi (2) with deconz crashing after closing deconz

rtenklooster, 2017-08-31:
I downloaded the RPi image, upgraded to the latest deconz version upgraded the deconz api and installled the needed tools as described in the readme.

Everything is working fine, until I close deconz. Then my PI shuts down and reboots.
I have the feeling it has something to do with the watchdog, because sometimes it crashes within a couple of seconds after closing, other times I can close and restart deconz without crash.

I have watched the kernel.log and system.log, no entries there.
Suggestions on how to debug this problem are welcome. I'm a bit scared of corrupting my SD card with those hard restarts.

P.s. It's not a power supply issue, and only happening when I close deconz.

manup, 2017-08-31:
It is indeed the watchdog, our sd-card image has it enabled so when deCONZ is crashed/closed the RPi will reboot (this was added mostly for industrial usage).

To disable the watchdog you can:
`$ sudo systemctl disable watchdog`

Or delete the deCONZ watchdog script.

rtenklooster, 2017-08-31:
Thanks for your reply. I will disable it while debugging. Works actually pretty good. I will definitely enable it when in production.

closed 2017-08-31 by rtenklooster