dresden-elektronik/deconz-rest-plugin #159: Feature request: include device type in websocket payload

rtenklooster, 2017-09-01:
I've created a websockets to MQTT bridge. It allows me to publish messages to my MQTT broker.
This way I'm able to respond on motion sensors and switches without polling.

The websocket client is missing some relevant information. It would be great if the` "type": "ZHAPresence"` would be included in the message payload. It would be even nicer if a user would be able to define witch fields should be included in the message.

Currently I have a workaround:
if buttonevent is null > switch
else > motion

However when adding more device types this is not the way it should be done. Of course it would be possible to query the rest api on the corresponding ID, this would cause a lot of unneeded requests.

manup, 2017-09-01:
Doesn't the event already carry all needed information?

* switches: `state/buttonevent`
* motion sensors: `state/buttonevent`
* light sensors: `state/lightlevel`
* contact sensors: `state/openclose`

etc.

In my opinion the events should only contain *dynamic / changed* data, not the static details of the resource &mdash; these should be queried via REST API. Otherwise we could reason to include the full resource in each event which would be overkill.

rtenklooster, 2017-09-01:
Great. Didn't know the event contained that info. Currently I have only connected motion sensors and switches so i didn't saw the logic.

I agree less is better. Thanks again.

closed 2017-09-01 by rtenklooster

manup, 2017-09-01:
@ebaauw has written a nice node.js tool to log the websockets events,  very handy to see what's happening

https://github.com/ebaauw/dc_eventlog

rtenklooster, 2017-09-01:
@manup Thanks for the link. I already created a nice node-red flow.
[image]

I've updated the filtering on specific events. Seems to work fine now.