dresden-elektronik/deconz-rest-plugin #160: Unexpected JSON response in rule conditions

snozzlebert, 2017-09-01:
The JSON type of the ```value``` property in conditions not always correct. Depending on the called rest endpoint (getFullState / getAllRules) it returns null or a string. Only the response of the rule by id endpoint is as expected (getRule).

Some examples with annotations:

Full state: http://{{host}}/api/{{username}}
[rest_rules.cpp#L180](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/7cc29e06a5272657c4843baa2b0203658334d87e/rest_rules.cpp#L180) missing null check?
```
"conditions": [
    {
        "address": "/sensors/5/state/buttonevent",
        "operator": "eq",
        "value": 16
    },
    {
        "address": "/sensors/5/state/lastupdated",
        "operator": "dx",
        "value": null  	   [Not expected]
    }
]
```

http://{{host}}/api/{{username}}/rules
[rest_rules.cpp#L100](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/ddcace4bd8bf80289be850d8118e4575f74b394d/rest_rules.cpp#L100) remove toString?
```
"conditions": [
    {
        "address": "/sensors/5/state/buttonevent",
        "operator": "eq",
        "value": "16"	  [Expect an int, no quotes]
    },
    {
        "address": "/sensors/5/state/lastupdated",
        "operator": "dx"  [OK, no value null]
    }
]
```

http://{{host}}/api/{{username}}/rules/{{id}}
```
"conditions": [
    {
        "address": "/sensors/5/state/buttonevent",
        "operator": "eq",
        "value": 16
    },
    {
        "address": "/sensors/5/state/lastupdated",
        "operator": "dx"
    }
]
```

http://{{host}}/api/{{username}}
```
"conditions": [
    {
        "address": "/sensors/8/state/open",
        "operator": "eq",
        "value": true		[OK]
    },
    {
        "address": "/sensors/8/state/lastupdated",
        "operator": "dx",
        "value": null		[Not expected]
    }
]
```

http://{{host}}/api/{{username}}/rules
```
"conditions": [
    {
        "address": "/sensors/8/state/open",
        "operator": "eq",
        "value": "true"		[Expect boolean, no quotes]
    },
    {
        "address": "/sensors/8/state/lastupdated",
        "operator": "dx"
    }
]
```

http://{{host}}/api/{{username}}/rules/{{id}}
```
"conditions": [
    {
        "address": "/sensors/8/state/open",
        "operator": "eq",
        "value": true		[OK]
    },
    {
        "address": "/sensors/8/state/lastupdated",
        "operator": "dx"	[OK]
    }
]
```

manup, 2017-09-01:
Good catch I'll currently work on the rules and will check the `null` issues.

The values actually *are* exposed as string and will be cast to their respective data type.

> Official hue description: `value` string 1..64: The resource attribute is compared to this value using the given operator. The value is cast to the data type of the resource attribute (in case of time, casted to a timePattern). If the cast fails or the operator does not support the data type the value is cast to the rule is rejected.

Output from the hue bridge:
```
"conditions": [
            {
                "address": "/config/localtime",
                "operator": "in",
                "value": "T08:00:00/T23:00:00"
            },
            {
                "address": "/sensors/9/state/presence",
                "operator": "eq",
                "value": "true"
            },
            {
                "address": "/sensors/9/state/presence",
                "operator": "dx"
            },
            {
                "address": "/sensors/11/state/status",
                "operator": "lt",
                "value": "1"
            },
            {
                "address": "/sensors/10/state/dark",
                "operator": "eq",
                "value": "true"
            }
        ]
```

manup, 2017-09-01:
The wrong string cast is fiexed now, however I couldn't find why you get null values, the code does check for it.

Which version of deCONZ are you running?

Here is output on my PC:

```
"conditions": [
        {
            "address": "/sensors/21/state/status",
            "operator": "gt",
            "value": "-1"
        },
        {
            "address": "/sensors/20/state/presence",
            "operator": "eq",
            "value": "true"
        },
        {
            "address": "/sensors/20/state/presence",
            "operator": "dx"
        }
    ]
```

manup, 2017-09-01:
Ah I see ruleToMap() has the error ...

ebaauw, 2017-09-01:
You might also check and filter an empty body in an action (typically the case when recalling a scene).  As the Hue syntax for recalling scenes differs, there's no comparing to the Hue bridge here.

grover, 2017-09-04:
I'm still seeing some inconsistencies with rule responses on 2.04.73 - see #169 for a sample response.

ebaauw, 2017-09-04:
The commit with the fix is from after 2.04.73 - you'll have to wait for the next version, or compile the REST API plugin yourself using this commit (or a later one).

manup, 2017-09-10:
Should be resolved in version 2.04.75, I close this for now.

closed 2017-09-10 by manup