dresden-elektronik/deconz-rest-plugin #162: How do I restart raspbeegw without power cycle ?

donnib, 2017-09-02:
My system went dead, all nodes red. I tried rebooting remotely but it can't find any nodes. I guess a reboot of the GW is needed. I remember something with GCC flasher but were is this located?

manup, 2017-09-02:
`$ sudo GCFFlasher_Internal -r`

This will restart the RaspBee shield, deCONZ must not be running while doing this.

Do you have already version 2.04.73 installed? It should improve the red node issue.

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.73-qt5.deb

donnib, 2017-09-02:
Where is GCFlasher located in the original rPi image ?

No not yet as I am on vacation for another week but all my automation died with the RaspbeeGW. I'll upgrade as soon as I get a chance.

manup, 2017-09-02:
You don't need to provide the path it's in /usr/bin

But I had a typo .. lower case i:

`$ sudo GCFFlasher_internal -r`