dresden-elektronik/deconz-rest-plugin #164: deleted sensor resources not purged from memory

ebaauw, 2017-09-03:
Deleted sensor resources still count against the limit of 1000 sensors.  After creating 1000 CLIP sensors, when trying to create another sensor, deCONZ rightly returns error 502: The Sensor List has reached its maximum capacity of 1000 sensors.  However after deleting a sensor resource, deCONZ continues to return this error when trying to create a replacement sensor, until deCONZ is restarted.

The sensor records are marked as deleted in the database.  Even after restarting deCONZ, the records are not purged from the database.

manup, 2017-09-03:
> Deleted sensor resources still count against the limit of 1000 sensors.

I think we can kill or greatly raise the limit, on a Raspberry it doesn't make much sense to have such low limits. As near time fix the limit however should only count not as deleted marked sensors.

> The sensor records are marked as deleted in the database. Even after restarting deCONZ, the records are not purged from the database.

That's on purpose to prevent nasty side effects. The plan is to purge / garbage collect deleted sensors, groups and other resources in future. But only after some time has passed.

ebaauw, 2017-09-03:
> I think we can kill or greatly raise the limit, 

I am actually quite happy with the fact that there is a limit.  As you know, I maintain my gateway configuration through scripts, which also serve as backup.  I want these scripts to be idempotent.  In particular, I want to be able to recreate a sensor with the same ID as before.  On the Hue bridge, I used to fill up all sensors resources with dummy CLIP sensors and just delete the one with the ID I wanted to create next.  After all sensors had been created this way, I'd delete the remaining dummy sensors, see https://github.com/ebaauw/ph.sh/blob/master/ph_sensors.sh.

Of course, in deCONZ, I can "patch" the sensor IDs in the database, but I haven't yet looked at a way of scripting that.  Also, I'd like as much as possible to use the same scripts for deCONZ and the Hue bridge.

> on a Raspberry it doesn't make much sense to have such low limits.

The corresponding limit on the Hue bridge is 64 (actually 63, since the Daylight sensor already takes up one slot).  I think 1000 is a pretty decent limit, and would not call it "low".

closed 2018-06-12 by ebaauw