dresden-elektronik/deconz-rest-plugin #165: Support of Aqara switch WXKG02LM

etique57, 2017-09-03:
Hello,

If I'm not wrong, I've updated to the latest beta following the instruction, which I think is compatible with the Aqara (Xiaomi) switches.

How should we add them in the webinterface? As a device, I guess? But what type should I choose?

Thanks!

manup, 2017-09-03:
Not sure if this double button switch is already detected by the plugin. You can let it join to the network after opening the network in the web app `Menu > Settings > Open Network` and press the reset button on the switch (I assume it has one like the other Xiaomi devices?). It should show up in deCONZ, but it might not be visible in the REST API yet.

etique57, 2017-09-03:
> I assume it has one like the other Xiaomi devices?

Nope, unfortunately I can't find any reset button.

Happy to help in any way if it's not supported, but you'll have to cope with my utter ignorance of many things! (for instance I have no clue what a REST API versus deCONZ is ;))

rtenklooster, 2017-09-03:
@etique57 Do you want to connect it to Home Assistant?

etique57, 2017-09-03:
@rtenklooster I want indeed ;)

Nope, can't find any "reset" button. On the notice it seems to show that I need to push on the 2 buttons at the same time, which I tried. The switch blinks for several seconds and then stops.

But on the web interface I see nothing happening.

etique57, 2017-09-09:
Just for information, the switch is this one:
https://www.gearbest.com/alarm-systems/pp_610095.html

etique57, 2017-09-15:
hello @manup any way I can help in the support of these nice looking switches? :)

manup, 2017-09-15:
Yes the support for the switch will be added, I just need to receive mine for further testing and integration :)

etique57, 2017-09-16:
Excellent, thanks!

deennoo, 2017-09-26:
Reset is do like this : 
Remove it from your xiaomi system
Remove switch batterie
Replace batterie
Press the twice button on the same time during 5 sec until led are blinking

simonporter007, 2017-09-30:
I've got the 2 gang wired version just arrived today. If you hold one of the paddles down for around 3 seconds, the two leds underneath start flashing blue (presumably searching for network). If I have deconz open during this time, they do stop flashing and then turning one paddle on shows a blue led rather than a red led when it wasn't connected. So presumably it is connected to the network? Unfortunately I'm running deconz in headless (--platform minimal) server, so can't check the deCONZ app and the REST API is empty as you've guessed.

Let me know if you need a test, i'm running 2.04.78 ubuntu beta version.

etique57, 2017-10-07:
@manup I can send you my switch if you're still waiting for yours ;)

alexsporn, 2017-10-07:
I have some WXKG03LM switches (single switch) and have the same issue when trying to pair them to deCONZ. I open the network from the webapp and then keep the switch pressed until it blinks 3 times, then I release it. After that only the left light blinks and nothing happens.

manup, 2017-10-08:
Hey all, the switch is waiting on my desk and in the todo queue.

It takes longer though, since currently focus is on improve API, polling and IKEA mesh problems as well as bring the closed beta app to next release.

I'm thinking of using the GitHub projects feature to more transparently share what's in the pipeline and what is planned next, but not sure if it brings more paper work and reduces actual work on the items.

etique57, 2017-10-08:
no problem @manup  sorry I din't mean to be pushy.

It's fully egoist though as I find there's so few very good looking switches... I'm eager to get this one working!

So far the only rather acceptable ones to me were the Osram switches, but they're really not practical. All the rest are either utterly expansive (sorry for the Dresden ones ;) or the B&J) or utterly ugly (hue, tradfri...)

manup, 2017-10-08:
>no problem @manup sorry I din't mean to be pushy.

All good, I hope to get it working soon :)

I agree the german brands including ours are indeed pricy, the benefits are i) that they can be used with existing classical frames like JUNG LS, A and CD designs  ii) they can be integrated into even combined housings like power plug and switch or three switches in a row etc.

Personally I really like the IKEA remote, it's perfect for couch and work table and even survived the WAF test :)

simonporter007, 2017-11-01:
Hey @manup I don't suppose you've had time to share where this is in the pipeline? I see there's been a fair few updates to deconz since I last tried this. Wondered if it has gotten a look in yet?

Keep up the good work!

manup, 2017-11-01:
The switch integration is scheduled for this weekend, weather looks promising (rain) :)

simonporter007, 2017-11-01:
Ahhhhhhhhh sweet. Thanks, I'll be in line to test then!

svenove, 2017-11-18:
Is this a momentary switch? So that I can trigger 4 different events with the two-button version?

manup, 2017-11-18:
Actually it only provides 3 events: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/e395207899d4f6c653a25328bf5419073ea850a9

simonporter007, 2017-11-20:
Hi @manup sorry to be a pain, looks like you've added the code - when's the next debian/ubuntu beta release was going to happen? (don't mind waiting, just wondering if there was a timeline). Looks like rpi is on .90 but [ubuntu ](https://www.dresden-elektronik.de/deconz/ubuntu/beta/) page only has .86

manup, 2017-11-20:
All good :) the Ubuntu/Windows packages will be uploaded during the next 1&ndash;2 days too.

simonporter007, 2017-11-26:
Gave .92 a run today, the wall switch seems to join the network (it goes from red to flashing blue to off) and when activating the switch, it remains constant blue which according to the user guide means its connected to a network an "on". However, nothing seems to show up in the REST GUI and tried deconz-config which also doesn't seem to pick anything up.

Wondering if anyone else had any luck?

manup, 2017-11-27:
Normally it works like this:

* *WebApp > Menu > Settings > Open Network*
* reset the switch by pressing both buttons for 10 seconds
* switch should then (re-)join the network and REST resource will be created

simonporter007, 2017-11-27:
I've tried all I can think of, including your steps above. Is there some extra logging I can enable with the headless setup?

- Open network for 60 seconds
- Hold left or right button down (I don't think it matters in this case, but it doesn't seem like two buttons does anything)
- Light turns itself off, the two mini led's underneath the wall switch go to red and start flashing blue
- The two mini led's turn off
- Turn on the light again and the mini led underneath the switch that was turned on lights up static blue

According to the documentation I've read on xiaomi website, it looks like this should be connected to a network. Nothing appears in the web UI, I can't see any mention of it when calling a GET /sesnors request either. Am I missing something crucial?

simonporter007, 2017-11-30:
Tried with .94, seems the same unfortunately :(

manup, 2017-11-30:
Can you see the device in deCONZ? If so a screenshot from the basic cluster info would be helpful.
And just to confirm you got the model WXKG02LM printed on the back of the switch?

Just tried it, worked.
Here is what I did:
* *WebApp > Menu > Settings > Open Network*
* reset the switch by pressing **both** buttons for ~10 seconds
* the two LEDs start to blink and will flash once the switch joined the network
* sensor was created in the REST API

```json
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "054bf6a25f5d1b7c96d63bebe65b0457",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.sensor_86sw2",
    "name": "lumi.sensor_86sw2",
    "state": {
        "buttonevent": 2002,
        "lastupdated": "2017-11-30T21:38:56"
    },
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:01:63:97:03-01-0006"
}
```
[image]

simonporter007, 2017-11-30:
So I guess this might be my bad :cry: it looks like I got my models mixed up.
[image]
The actual model is: [QBKG03LM](https://www.banggood.com/New-Arrival-Original-Xiaomi-Aqara-Wall-Switch-ZigBee-Version-Smart-Home-Controller-p-1153555.html?cur_warehouse=CN)
I can raise a new issue I guess as this will probably need a new code change?

Sorry!!

manup, 2017-11-30:
No worries :) can you also show a screenshots of the endpoints? I guess the device might be similar to the WXKG02LM switch.

simonporter007, 2017-11-30:
Do you mean these?
[image]
[image]

manup, 2017-11-30:
Yes, well this looks quite different. Ill add the model to supported devices and when we will see some more debug output for the button events.

simonporter007, 2017-11-30:
Thanks, let me know if I can help any.

simonporter007, 2017-12-11:
hi @manup Sorry for the bother, I tried .97 and still no luck. It looks like you may have made a typo?
`{ VENDOR_JENNIC, "lumi.ctrl_neural2", jennicMacPrefix },`
vs
`{ VENDOR_JENNIC, "lumi.ctrl_neutral2", jennicMacPrefix },`
neutral with the "t"?

simonporter007, 2018-04-24:
Fixed with 2.05.53 as per #356

manup, 2018-10-05:
Closing this issue, discussion continues in issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/798

closed 2018-10-05 by manup