dresden-elektronik/deconz-rest-plugin #166: OTA upgrade of IKEA GU 10 bulbs gets stuck

donnib, 2017-09-04:
Hi, i did a a comment here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/96#issuecomment-324965537 about a problem I have with my new GU 10 IKEA bulbs which all get stuck on OTA upgrade. What can I do ?

donnib, 2017-09-04:
[image]

manup, 2017-09-05:
On it, I have a lamp here (non IKEA) with the same problem, might be an issue in the OTA plugin. I'll report back when there are news.

donnib, 2017-09-05:
Cool Nice to know that you can reproduce it. Hope u can find a fix.

manup, 2017-09-05:
I could fix the issue for the non IKEA light, can you please test the new version with your lights.

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.75-qt5.deb

ebaauw, 2017-09-05:
Would this also fix the issue with the firmware update getting stuck on the ubisys dimmer?

manup, 2017-09-05:
I'm afraid it won't, the issue here was a too long OTA reply packet which was rejected from the firmware.

You may try it, I'll put an ubisys OTA test on the list.

donnib, 2017-09-06:
@manup thank you, that nailed it

closed 2017-09-06 by donnib

donnib, 2020-09-28:
@manup I have many GU10 which get's stuck on OTA, what can i do to fix this ?

Mimiix, 2020-09-28:
@donnib Please do not necro post. Open a new issue with your question. 

Use the template!

Smanar, 2020-09-28:
@donnib if your problem is the update reach 99% but don't work, please make a new issue with the firmware you have used, I have perhaps a solution (perhaps same bug but on another manufacture, corrected on m y side)